mu-editor/mu #1941: Crash Report

SandraSandra2, 2021-12-22:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.6
Locale: pl_PL
Platform: Windows 10 10.0.19042 AMD64

```

    self.install_baseline_packages()
  File "C:\Users\HP\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 880, in install_baseline_packages
    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "C:\Users\HP\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 840, in install_from_zipped_wheels
    zip.extractall(unpacked_wheels_dirpath)
  File "C:\Users\HP\AppData\Local\Programs\Mu Editor\Python\lib\zipfile.py", line 1647, in extractall
    self._extract_member(zipinfo, path, pwd)
  File "C:\Users\HP\AppData\Local\Programs\Mu Editor\Python\lib\zipfile.py", line 1702, in _extract_member
    shutil.copyfileobj(source, target)
  File "C:\Users\HP\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 205, in copyfileobj
    buf = fsrc_read(length)
  File "C:\Users\HP\AppData\Local\Programs\Mu Editor\Python\lib\zipfile.py", line 940, in read
    data = self._read1(n)
  File "C:\Users\HP\AppData\Local\Programs\Mu Editor\Python\lib\zipfile.py", line 1030, in _read1
    self._update_crc(data)
  File "C:\Users\HP\AppData\Local\Programs\Mu Editor\Python\lib\zipfile.py", line 958, in _update_crc
    raise BadZipFile("Bad CRC-32 for file %r" % self.name)
zipfile.BadZipFile: Bad CRC-32 for file 'cryptography-35.0.0-cp36-abi3-win_amd64.whl'

```

carlosperate, 2021-12-23:
Thanks for the report @SandraSandra2!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\HP\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2022-03-07:
Hi @SandraSandra2, are you still encountering this issue?
If you are, could you try downloading the latest installer again? Might be a corrupted file.

carlosperate, 2022-05-31:
Closing due to lack of activity.

closed 2022-05-31 by carlosperate