mu-editor/mu #1944: Crash Report

Akuabata01, 2021-12-22:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0b7
Locale: en_GB
Platform: Windows 10 10.0.19042 AMD64

```
.create()
  File "C:\Users\Theresae\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 772, in create
    self.install_baseline_packages()
  File "C:\Users\Theresae\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 885, in install_baseline_packages
    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "C:\Users\Theresae\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 860, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "C:\Users\Theresae\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 270, in install
    return self.run(
  File "C:\Users\Theresae\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "C:\Users\Theresae\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "C:\Users\Theresae\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 177, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process did not terminate normally:

```

Akuabata01, 2021-12-22:
C:\Users\Theresae\AppData\Local\python\mu\Logs\mu.log

carlosperate, 2021-12-23:
Thanks for the report !
Would you be able to drag and drop the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\Theresae\AppData\Local\python\mu\Logs\mu.log`