mu-editor/mu #1946: Crash Report

dsssssssss9, 2021-12-22:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0b7
Locale: en_GB
Platform: Linux 5.15.10-051510-generic #202112171031 SMP Fri Dec 17 10:53:39 UTC 2021 x86_64

```
e/jkd/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/home/jkd/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 772, in create
    self.install_baseline_packages()
  File "/home/jkd/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 885, in install_baseline_packages
    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "/home/jkd/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 860, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "/home/jkd/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 270, in install
    return self.run(
  File "/home/jkd/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "/home/jkd/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "/home/jkd/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 1:
ERROR: cryptography-36.0.1-cp36-abi3-manylinux_2_24_x86_64.whl is not a supported wheel on this platform.

```

dsssssssss9, 2021-12-22:
[mu.log](https://github.com/mu-editor/mu/files/7764337/mu.log)

carlosperate, 2021-12-23:
Thanks for the report and the logs @dsssssssss9!

That's a bit odd, pip downloads the `cryptography-36.0.1-cp36-abi3-manylinux_2_24_x86_64.whl` wheel, but then it errors indicating that is not supported on this platform 🤔

What version of pip have you got? Did you pip install Mu in a virtual environment?

dsssssssss9, 2021-12-23:
Hi

Thanks for taking the time to reply - it is appreciated!!

pip3 --version
pip 21.3.1 from /home/jkd/.local/lib/python3.8/site-packages/pip (python 3.8)

No i ( as i did with the previous 2 versions ) just did the following....

1) uninstall the previous version with pip3  uninstall
2) install the latest version with pip3 install

since trying to install 1.1.0b7 i can no longer reinstall 1.1.0b6 ?

please let me know if i can provide any more help for you on this  - not urgent as i am sure you good folks are snowed under with MU related things at the moment let alone having a happy holiday season with family / friends!!

carlosperate, 2021-12-23:
Thanks @dsssssssss9! 

> since trying to install 1.1.0b7 i can no longer reinstall 1.1.0b6 ?

Outside of this specific issue with the `cryptography` module I can't think of a reason why that wouldn't work, it is beta 6 throwing the same error?

What distro are you using?

My guess is that i has something to do with pip not downloading the right `cryptography` wheel for your platform. However, pip v21.3.1 is the latest version, so not too sure what could be the problem.

Could you run these commands on a terminal and paste the output here?

```
/home/jkd/.local/share/mu/mu_venv-38-20211222-174319/bin/pip --version
```

```
/home/jkd/.local/share/mu/mu_venv-38-20211222-174319/bin/pip install cryptography==36.0.1 --verbose
```

dsssssssss9, 2021-12-23:
sure no problem...
*-*-*-*-*-*-*-*-*-*
$ /home/jkd/.local/share/mu/mu_venv-38-20211222-174319/bin/pip --version

gives....
pip 20.0.2 from /home/jkd/.local/share/mu/mu_venv-38-20211222-174319/lib/python3.8/site-packages/pip (python 3.8)

*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*-*
:~$ /home/jkd/.local/share/mu/mu_venv-38-20211222-174319/bin/pip install cryptography==36.0.1 --verbose

Gives the following ( i have uploaded it as a text file as it is quite long! ).

/home/jkd/Desktop/mu_output1.txt

carlosperate, 2021-12-24:
Thanks @dsssssssss9, that was really useful!

Don't worry, you don't have two pips in your system. The way Mu works is that during the first launch it creates a new virtual environment to install the "user packages" and isolate those from the system and Mu's Python. So this other pip is only contained there.

An important thing to consider in the Mu handles this, is that before it creates the virtual environment it downloads (in this case with the system pip) the wheels it needs to install into the venv.

I believe the problem you are encountering is this one: 
- https://github.com/mu-editor/mu/issues/1826

Basically the system Python, which runs Mu, has the latest pip version 21.3.1, and this version of pip supports new "manylinux" tags (added in 20.3 with format `manylinux_x_y`), so it downloads the `cryptography-36.0.1-cp36-abi3-manylinux_2_24_x86_64.whl` wheel: https://pip.pypa.io/en/stable/news/#id181

Then Mu creates a virtual environment, but the version of pip in this virtual environment is older, 20.0.2. This version of pip does not support this new "manylinux" tag, and that's why it fails to install that wheel.
Your `pip instal` output shows that this venv pip installs `cryptography-36.0.1-cp36-abi3-manylinux_2_17_x86_64.manylinux2014_x86_64.whl`, which has the older `manylinux2014` tag that it's able to recognise.

The part that surprises me is that the `virtualenv` module it's meant to upgrade the pip version in the virtual environemt during its creation, so I'm not quite sure why it is leaving pip v20.0.2 in the venv.

carlosperate, 2021-12-24:
About `virtualenv` installing and up-to-date pip: Looking at the docs it will not update the latest pip on venv creation, but what it has as a "seeder": https://virtualenv.pypa.io/en/latest/user_guide.html#seeders

Not sure I fully understood the next "wheels" section (https://virtualenv.pypa.io/en/latest/user_guide.html#wheels), but it sounds like the seeder wheels might get updated "in the background" after 14 days, but I'm not sure if this cover pip as well. Even if it does, it means that it's not guaranteed that the venv pip is as up-to-date as Mu's pip.

dsssssssss9, 2021-12-24:
Thanks for all the detective work ....

So from what I understand there is no solution for my install problem at this point in time but hopefully there may be soon once some of the underlying magic of all this casts another spell or 2 ?

Will await developments .....

carlosperate, 2021-12-24:
I think there might be two possible workarounds:
1) Get the system `virtualenv` module to update its "seeder" version of pip
    - I'll check the docs to find out how to do this
2) Temporarily downgrade the system pip to 20.0.2, uninstall and reinstall Mu, launch Mu (hopefully this time it downloads the other wheel), and he update the system pip back to v21.3.1

dsssssssss9, 2021-12-24:
> 2. Temporarily downgrade the system pip to 20.0.2, uninstall and reinstall Mu, launch Mu (hopefully this time it downloads the other wheel), and then update the system pip back to v21.3.1

Tried this by first uninstalling current version ....
python -m pip uninstall  pip

Just got he following error.....

/usr/bin/python: No module named pip

carlosperate, 2021-12-24:
Did you use `python3` and/or `pip3` before?

dsssssssss9, 2021-12-25:
> Did you use `python3` and/or `pip3` before?

Hi

whilst waiting for my holiday season lunch to cook i thought i'd go through all our previous steps again just in case...

i did pip & pip3 uninstall to make sure there were no remains of any previous installation - 
discovered there was an old version of 1.0.2 that had must have been installed via sudo apt install so i did sudo apt remove

tried downgrading pip to version 20.0.2 as per your suggestion..
[pip downgrade 25 dec 2021.txt](https://github.com/mu-editor/mu/files/7775922/pip.downgrade.25.dec.2021.txt)

then did .....

pip3 install mu-editor==1.1.0b7

[mu install log 25 dec 2021.txt](https://github.com/mu-editor/mu/files/7775917/mu.install.log.25.dec.2021.txt)

when i ran
[mu 25 dec 2021.log](https://github.com/mu-editor/mu/files/7775924/mu.25.dec.2021.log)
 mu i got the following log file......

not sure this gets us ( you ) any closer to a solution but maybe ?

Regards

dsssssssss9, 2021-12-29:
Ok have read through some of the other issues & saw @carlosperate mention installing via PIPX,,,, in #1954 

So uninstalled 1.1.0b7  vai pip3

installed pipx via sudo apt install pipx

ran pipx install mu-editor==1.1.0b7

Success!!! Mu version 1.1.0b7 now seems to work  - will do some further testing over the course of the next couple of days & if this is the case i will close this issue

carlosperate, 2021-12-30:
> ran pipx install mu-editor==1.1.0b7
> 
> Success!!! Mu version 1.1.0b7 now seems to work - will do some further testing over the course of the next couple of days & if this is the case i will close this issue

Oh that's great, glad that works!

> when i ran mu i got the following log file......
> 
> [mu 25 dec 2021.log](https://github.com/mu-editor/mu/files/7775924/mu.25.dec.2021.log)
> 
> not sure this gets us ( you ) any closer to a solution but maybe ?

Looking at the log it looks like we had the same issue again. I forgot that Mu had already created a zip file with the downloaded wheels the last time it run, so it didn't download the wheels with the older pip that was just installed.

Could you ensure the system pip is still at v20, then delete this file `/home/jkd/.local/lib/python3.8/site-packages/mu/wheels/1.1.0b7.zip` and then try to run Mu from the system Python again (`python3 -m mu`)? I think this way it will download the wheels again with the old pip and should work.