mu-editor/mu #1948: This Popped up on First Start - But then it worked --- Crash Report

glennhefley, 2021-12-23:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0b7
Locale: en_US
Platform: Windows 10 10.0.19043 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Owner\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Owner\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Owner\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\Owner\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\Owner\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 349, in run
    editor.setup(setup_modes(editor, editor_window))
  File "C:\Users\Owner\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 809, in setup
    os.makedirs(wd)
  File "C:\Users\Owner\AppData\Local\Programs\Mu Editor\Python\lib\os.py", line 223, in makedirs
    mkdir(name, mode)
FileExistsError: [WinError 183] Cannot create a file when that file already exists: 'C:\\Users\\Owner\\mu_code'

```

glennhefley, 2021-12-23:
Not sure what else to do. I'll put some "good" code in and see if it runs.

glennhefley, 2021-12-23:
Seems to work. So, thanks? and there's your log report... have  a good new year.... ()bye.

carlosperate, 2021-12-23:
Thanks for the report @glennhefley!

I assume the crash happened the first time you were launching Mu, during the splash screen?

Would you be able to attach the Mu log to this GitHub Issue as well? It'll contain really useful information for debugging, that should help make sure this doesn't happen again. File should be located in: `C:\Users\Owner\AppData\Local\python\mu\Logs\mu.log`

%LOCALAPPDATA%\python\mu\Logs

~/.cache/mu/
~/Library/Logs/mu

carlosperate, 2021-12-29:
Hi @glennhefley, I was wondering if you'd be able to send us the log? It would be really useful to us too ensure this error can be fixed in future releases.

glennhefley, 2021-12-31:
sorry... I forgot about this and I've uninstalled.

carlosperate, 2022-01-03:
No worries, thanks for the report!

closed 2022-01-03 by carlosperate