mu-editor/mu #1949: Crash Report

ondragottwald, 2021-12-23:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0b7
Locale: cs_CZ
Platform: Linux 5.11.0-43-generic #47~20.04.2-Ubuntu SMP Mon Dec 13 11:06:56 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/home/ondra/.local/lib/python3.8/site-packages/mu/app.py", line 159, in run
    raise ex
  File "/home/ondra/.local/lib/python3.8/site-packages/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/ondra/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/home/ondra/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 771, in create
    self.create_venv()
  File "/home/ondra/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 803, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using /usr/bin/python3 at /home/ondra/.local/share/mu/mu_venv-38-20211223-140715
STDERR: /usr/bin/python3: No module named virtualenv

```

carlosperate, 2021-12-23:
Thanks for the report @ondragottwald!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `~/.cache/mu/`

lujo6911, 2021-12-23:
Same error, my logfile:

[fedora-laptop mu-1.1.0-beta.7]$ ./run.py 
QSocketNotifier: Can only be used with threads started with QThread
Traceback (most recent call last):
  File "/home/lujo/Mu/mu-1.1.0-beta.7/mu/app.py", line 159, in run
    raise ex
  File "/home/lujo/Mu/mu-1.1.0-beta.7/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/lujo/Mu/mu-1.1.0-beta.7/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/home/lujo/Mu/mu-1.1.0-beta.7/mu/virtual_environment.py", line 771, in create
    self.create_venv()
  File "/home/lujo/Mu/mu-1.1.0-beta.7/mu/virtual_environment.py", line 803, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using /usr/bin/python3 at /home/lujo/.local/share/mu/mu_venv-39-20211223-180215
STDERR: /usr/bin/python3: No module named virtualenv

(python3:16455): GLib-CRITICAL **: 18:02:22.963: g_main_context_pop_thread_default: assertion 'g_queue_peek_head (stack) == context' failed
QObject::~QObject: Timers cannot be stopped from another thread
Opening in existing browser session.
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
[16561:16561:0100/000000.276521:ERROR:sandbox_linux.cc(376)] InitializeSandbox() called with multiple threads in process gpu-process.

[fedora-laptop log]$ cat mu.log 
2021-12-23 18:02:15,379 - root:269(run) INFO: 

-----------------

Starting Mu 1.1.0b7
2021-12-23 18:02:15,379 - root:270(run) INFO: uname_result(system='Linux', node='fedora-laptop', release='5.15.5-100.fc34.x86_64', version='#1 SMP Fri Nov 26 00:52:21 UTC 2021', machine='x86_64')
2021-12-23 18:02:15,382 - root:271(run) INFO: Platform: Linux-5.15.5-100.fc34.x86_64-x86_64-with-glibc2.33
2021-12-23 18:02:15,382 - root:272(run) INFO: Python path: ['/home/lujo/Mu/mu-1.1.0-beta.7', '/usr/lib64/python39.zip', '/usr/lib64/python3.9', '/usr/lib64/python3.9/lib-dynload', '/home/lujo/.local/lib/python3.9/site-packages', '/usr/local/lib/python3.9/site-packages', '/usr/lib64/python3.9/site-packages', '/usr/lib/python3.9/site-packages', '/usr/lib/python3.9/site-packages/IPython/extensions']
2021-12-23 18:02:15,382 - root:273(run) INFO: Language code: en_US
2021-12-23 18:02:15,511 - mu.virtual_environment:585(ensure_and_create) INFO: Added log handler.
2021-12-23 18:02:15,511 - mu.virtual_environment:596(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-12-23 18:02:15,512 - mu.virtual_environment:616(ensure_and_create) WARNING: Venv created by Mu version 1.1.0.beta.6; Current Mu is version 1.1.0b7
2021-12-23 18:02:15,512 - mu.virtual_environment:922(installed_packages) INFO: Discovering installed third party modules in venv.
2021-12-23 18:02:15,512 - mu.virtual_environment:114(run_blocking) INFO: About to run blocking /home/lujo/.local/share/mu/mu_venv-39-20211223-174821/bin/pip with args ['list', '--disable-pip-version-check'] and envvars {}
2021-12-23 18:02:15,844 - mu.virtual_environment:164(wait) DEBUG: Finished: True; exitStatus 0; exitCode 0
2021-12-23 18:02:15,845 - mu.virtual_environment:249(run) DEBUG: Process output: Package           Version
----------------- -------
backcall          0.2.0
bitstring         3.1.9
cffi              1.15.0
click             8.0.3
cryptography      35.0.0
decorator         5.1.0
ecdsa             0.17.0
entrypoints       0.3
esptool           3.1
Flask             1.1.2
ipykernel         5.5.6
ipython           7.28.0
ipython-genutils  0.2.0
itsdangerous      2.0.1
jedi              0.18.0
Jinja2            3.0.2
jupyter-client    7.0.6
jupyter-core      4.8.1
MarkupSafe        2.0.1
matplotlib-inline 0.1.3
nest-asyncio      1.5.1
numpy             1.21.2
parso             0.8.2
pexpect           4.8.0
pgzero            1.2.1
pickleshare       0.7.5
pip               21.3
prompt-toolkit    3.0.20
[... 61 lines cut ...]
STDERR: /usr/bin/python3: No module named virtualenv

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/lujo/Mu/mu-1.1.0-beta.7/mu/virtual_environment.py", line 534, in quarantine_venv
    os.rename(self.path, error_dirpath)
FileNotFoundError: [Errno 2] No such file or directory: '/home/lujo/.local/share/mu/mu_venv-39-20211223-180215' -> '/home/lujo/.local/share/mu/mu_venv-39-20211223-180215.FAILED'
2021-12-23 18:02:22,928 - root:171(excepthook) ERROR: Unrecoverable error
Traceback (most recent call last):
  File "/home/lujo/Mu/mu-1.1.0-beta.7/mu/app.py", line 159, in run
    raise ex
  File "/home/lujo/Mu/mu-1.1.0-beta.7/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/lujo/Mu/mu-1.1.0-beta.7/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/home/lujo/Mu/mu-1.1.0-beta.7/mu/virtual_environment.py", line 771, in create
    self.create_venv()
  File "/home/lujo/Mu/mu-1.1.0-beta.7/mu/virtual_environment.py", line 803, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using /usr/bin/python3 at /home/lujo/.local/share/mu/mu_venv-39-20211223-180215
STDERR: /usr/bin/python3: No module named virtualenv
2021-12-23 18:02:22,961 - mu.settings:169(save) DEBUG: Saving to /home/lujo/.local/share/mu/session.json
2021-12-23 18:02:22,961 - mu.settings:169(save) DEBUG: Saving to /home/lujo/.local/share/mu/venv.json
[lujo@fedora-laptop log]$

carlosperate, 2021-12-23:
Did the `pip install mu-editor` step finish successfully? The error is because Python cannot load the virtualenv module, but it should have been installed as part of Mu:
https://github.com/mu-editor/mu/blob/a0ce7bce247429c2877c692a648cdd5681373650/setup.py#L65

Could try pip installing again and copy/pasting the terminal output here?

lujo, 2021-12-23:
No, tried to install with pip on dnf upgraded Fedora 33 and Fedora 34:

[ocalhost ~]$ python3 -m pip install mu-editor==1.1.0b7
Defaulting to user installation because normal site-packages is not writeable
ERROR: Could not find a version that satisfies the requirement mu-editor==1.1.0b7 (from versions: 1.0.0b12, 1.0.0b13, 1.0.0b14, 1.0.0b15, 1.0.0b16, 1.0.0b17, 1.0.0rc1, 1.0.0, 1.0.1, 1.0.2, 1.0.3)
ERROR: No matching distribution found for mu-editor==1.1.0b7
[ocalhost ~]$ 

So downloaded tar file from github:
https://github.com/mu-editor/mu/archive/refs/tags/1.1.0-beta.7.tar.gz
Did gunzip, tar xvf, cd to mu-1.1.0-beta.7, ran ./run.py. With the log file
I posted as result. Did not run setup.py.

Because, this worked for 1.1.0-beta.6.

virtualenv already was installed (for beta.6):
[localhost]$ pip3 list | grep virtualenv
virtualenv                        20.10.0

Reinstalled  virtualenv
[localhost]$ sudo python3 -m pip uninstall virtualenv
[localhost]$ sudo python3 -m pip install virtualenv

Same virtualenv version.
Same error log file when ./run.py.

ondragottwald, 2021-12-23:
> 

2021-12-23 14:01:30,036 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.2
2021-12-23 14:01:30,037 - root:113(run) INFO: uname_result(system='Linux', node='ondra-ESPRIMO-P', release='5.11.0-43-generic', version='#47~20.04.2-Ubuntu SMP Mon Dec 13 11:06:56 UTC 2021', machine='x86_64', processor='x86_64')
2021-12-23 14:01:30,037 - root:114(run) INFO: Python path: ['/usr/share/mu-editor', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/home/ondra/.local/lib/python3.8/site-packages', '/usr/local/lib/python3.8/dist-packages', '/usr/lib/python3/dist-packages', '/home/ondra/.local/lib/python3.8/site-packages/IPython/extensions']
2021-12-23 14:01:30,037 - root:115(run) INFO: Language code: cs_CZ
2021-12-23 14:01:30,633 - mu.logic:543(__init__) INFO: Setting up editor.
2021-12-23 14:01:30,634 - mu.logic:560(__init__) DEBUG: Creating directory: /home/ondra/.local/share/mu
2021-12-23 14:01:30,635 - mu.logic:315(get_admin_file_path) DEBUG: Creating admin file: /home/ondra/.local/share/mu/settings.json
2021-12-23 14:01:30,635 - mu.logic:562(__init__) INFO: Settings path: /home/ondra/.local/share/mu/settings.json
2021-12-23 14:01:30,635 - mu.logic:315(get_admin_file_path) DEBUG: Creating admin file: /home/ondra/.local/share/mu/session.json
2021-12-23 14:01:30,635 - mu.logic:563(__init__) INFO: Session path: /home/ondra/.local/share/mu/session.json
2021-12-23 14:01:30,636 - mu.logic:564(__init__) INFO: Log directory: /home/ondra/.cache/mu/log
2021-12-23 14:01:30,636 - mu.logic:565(__init__) INFO: Data directory: /home/ondra/.local/share/mu
2021-12-23 14:01:30,704 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2021-12-23 14:01:30,704 - mu.logic:582(setup) DEBUG: Creating directory: /home/ondra/mu_code
2021-12-23 14:01:30,705 - mu.logic:590(setup) DEBUG: Creating directory: /home/ondra/mu_code/images
2021-12-23 14:01:30,706 - mu.logic:597(setup) DEBUG: Creating directory: /home/ondra/mu_code/fonts
2021-12-23 14:01:30,706 - mu.logic:600(setup) DEBUG: Creating directory: /home/ondra/mu_code/sounds
2021-12-23 14:01:30,707 - mu.logic:603(setup) DEBUG: Creating directory: /home/ondra/mu_code/music
2021-12-23 14:01:30,978 - mu.logic:1147(change_mode) INFO: Workspace directory: /home/ondra/mu_code
2021-12-23 14:01:30,980 - mu.logic:625(restore_session) INFO: Restoring session from: /home/ondra/.local/share/mu/session.json
2021-12-23 14:01:30,981 - mu.logic:626(restore_session) DEBUG: {}
2021-12-23 14:01:30,982 - mu.logic:1104(select_mode) INFO: Showing available modes: ['python', 'adafruit', 'microbit', 'debugger', 'pygamezero']
2021-12-23 14:01:41,947 - mu.logic:1110(select_mode) INFO: New mode selected: adafruit
2021-12-23 14:01:42,030 - mu.logic:1147(change_mode) INFO: Workspace directory: /media/ondra/CIRCUITPY
2021-12-23 14:01:42,139 - mu.logic:678(restore_session) INFO: Starting with blank file.
2021-12-23 14:01:42,203 - mu.logic:1147(change_mode) INFO: Workspace directory: /media/ondra/CIRCUITPY
2021-12-23 14:01:53,205 - mu.logic:1104(select_mode) INFO: Showing available modes: ['python', 'adafruit', 'microbit', 'debugger', 'pygamezero']
2021-12-23 14:01:58,495 - mu.modes.base:227(find_device) INFO: Found device on port: ttyACM0
2021-12-23 14:01:58,495 - mu.modes.base:228(find_device) INFO: Serial number: E66038B7137E5639
2021-12-23 14:01:58,497 - mu.modes.base:281(add_repl) ERROR: Cannot connect to device on port /dev/ttyACM0
2021-12-23 14:01:58,524 - mu.interface.main:723(show_message) DEBUG: Cannot connect to device on port /dev/ttyACM0
2021-12-23 14:01:58,525 - mu.interface.main:724(show_message) DEBUG: Click on the device's reset button, wait a few seconds and then try again.
2021-12-23 14:02:08,342 - mu.modes.base:259(toggle_repl) INFO: Toggle REPL on.
2021-12-23 14:02:16,164 - mu.modes.base:227(find_device) INFO: Found device on port: ttyACM0
2021-12-23 14:02:16,164 - mu.modes.base:228(find_device) INFO: Serial number: E66038B7137E5639
2021-12-23 14:02:16,171 - mu.modes.base:278(add_repl) INFO: Started REPL on port: /dev/ttyACM0
2021-12-23 14:02:16,171 - mu.modes.base:259(toggle_repl) INFO: Toggle REPL on.
2021-12-23 14:02:59,049 - mu.logic:1104(select_mode) INFO: Showing available modes: ['python', 'adafruit', 'microbit', 'debugger', 'pygamezero']
2021-12-23 14:03:06,137 - mu.logic:809(get_dialog_directory) INFO: Using path for file dialog: /media/ondra/CIRCUITPY
2021-12-23 14:03:14,432 - mu.interface.main:251(get_load_path) DEBUG: Getting load path: /media/ondra/CIRCUITPY/code.py
2021-12-23 14:03:14,432 - mu.logic:713(_load) INFO: Loading script from: /media/ondra/CIRCUITPY/code.py
2021-12-23 14:03:14,433 - mu.logic:274(read_and_decode) DEBUG: Trying to decode with utf-8
2021-12-23 14:03:14,434 - mu.logic:277(read_and_decode) INFO: Decoded with utf-8
2021-12-23 14:03:14,435 - mu.logic:290(read_and_decode) DEBUG: Detected newline '\n'
2021-12-23 14:03:14,435 - mu.logic:787(_load) DEBUG: """Example for Pico. Turns on the built-in LED."""
import board
import digitalio

led = digitalio.DigitalInOut(board.LED)
led.direction = digitalio.Direction.OUTPUT

while True:
    led.value = True

2021-12-23 14:03:20,120 - mu.modes.base:227(find_device) INFO: Found device on port: ttyACM0
2021-12-23 14:03:20,120 - mu.modes.base:228(find_device) INFO: Serial number: E66038B7137E5639
2021-12-23 14:03:20,148 - mu.modes.base:317(add_plotter) INFO: Started plotter
[... cut ...]
  File "/home/ondra/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 803, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using /usr/bin/python3 at /home/ondra/.local/share/mu/mu_venv-38-20211223-141039
STDERR: /usr/bin/python3: No module named virtualenv
2021-12-23 14:10:47,814 - mu.settings:169(save) DEBUG: Saving to /home/ondra/.local/share/mu/session.json
2021-12-23 14:10:47,819 - mu.settings:169(save) DEBUG: Saving to /home/ondra/.local/share/mu/venv.json

ondragottwald, 2021-12-23:
_I tried uninstalling and reinstalling it. Here is the result._

`ondra@ondra-ESPRIMO-P:~$ pip install mu-editor
Collecting mu-editor
  Using cached mu_editor-1.0.3-py3-none-any.whl (2.0 MB)
Collecting gpiozero>=1.4.1
  Downloading gpiozero-1.6.2-py2.py3-none-any.whl (148 kB)
     |████████████████████████████████| 148 kB 3.1 MB/s 
Collecting pyflakes==2.0.0
  Using cached pyflakes-2.0.0-py2.py3-none-any.whl (53 kB)
Collecting pycodestyle==2.4.0
  Downloading pycodestyle-2.4.0-py2.py3-none-any.whl (62 kB)
     |████████████████████████████████| 62 kB 172 kB/s 
Collecting PyQtChart==5.14.0
  Using cached PyQtChart-5.14.0-5.14.2-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (1.2 MB)
Collecting qscintilla==2.11.4
  Downloading QScintilla-2.11.4-1-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (1.9 MB)
     |████████████████████████████████| 1.9 MB 6.9 MB/s 
Requirement already satisfied: nudatus>=0.0.3 in ./.local/lib/python3.8/site-packages (from mu-editor) (0.0.5)
Collecting pyqt5==5.14.1
  Downloading PyQt5-5.14.1-5.14.1-cp35.cp36.cp37.cp38-abi3-manylinux2014_x86_64.whl (63.5 MB)
     |████████████████████████████████| 63.5 MB 13 kB/s 
Requirement already satisfied: appdirs>=1.4.3 in ./.local/lib/python3.8/site-packages (from mu-editor) (1.4.4)
Requirement already satisfied: guizero>=1.1.0 in /usr/lib/python3/dist-packages (from mu-editor) (1.1.0)
Collecting pigpio>=1.40.post1
  Downloading pigpio-1.78-py2.py3-none-any.whl (39 kB)
Collecting qtconsole==4.3.1
  Downloading qtconsole-4.3.1-py2.py3-none-any.whl (108 kB)
     |████████████████████████████████| 108 kB 6.4 MB/s 
Collecting pyserial==3.4
  Using cached pyserial-3.4-py2.py3-none-any.whl (193 kB)
Requirement already satisfied: semver>=2.8.0 in ./.local/lib/python3.8/site-packages (from mu-editor) (2.13.0)
Requirement already satisfied: Pillow>=5.2.0 in /usr/lib/python3/dist-packages (from mu-editor) (7.0.0)
Requirement already satisfied: pgzero==1.2 in /usr/lib/python3/dist-packages (from mu-editor) (1.2)
Collecting matplotlib==2.2.2
  Using cached matplotlib-2.2.2.tar.gz (37.3 MB)
    ERROR: Command errored out with exit status 1:
     command: /usr/bin/python3 -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-rmmlm0i3/matplotlib/setup.py'"'"'; __file__='"'"'/tmp/pip-install-rmmlm0i3/matplotlib/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base /tmp/pip-install-rmmlm0i3/matplotlib/pip-egg-info
         cwd: /tmp/pip-install-rmmlm0i3/matplotlib/
    Complete output (63 lines):
    ============================================================================
    Edit setup.cfg to change the build options
    
    BUILDING MATPLOTLIB
                matplotlib: yes [2.2.2]
                    python: yes [3.8.10 (default, Nov 26 2021, 20:14:08)  [GCC
                            9.3.0]]
                  platform: yes [linux]
    
    REQUIRED DEPENDENCIES AND EXTENSIONS
                     numpy: yes [version 1.17.4]
          install_requires: yes [handled by setuptools]
                    libagg: yes [pkg-config information for 'libagg' could not
                            be found. Using local copy.]
                  freetype: no  [The C/C++ header for freetype2 (ft2build.h)
                            could not be found.  You may need to install the
                            development package.]
                       png: no  [pkg-config information for 'libpng' could not
                            be found.]
                     qhull: yes [pkg-config information for 'libqhull' could not
                            be found. Using local copy.]
    
    OPTIONAL SUBPACKAGES
               sample_data: yes [installing]
                  toolkits: yes [installing]
                     tests: no  [skipping due to configuration]
            toolkits_tests: no  [skipping due to configuration]
    
    OPTIONAL BACKEND EXTENSIONS
                    macosx: no  [Mac OS-X only]
                    qt5agg: yes [installing, Qt: 5.13.2, PyQt: 5.13.2; PySide2
                            not found]
                    qt4agg: no  [PySide not found; PyQt4 not found]
                   gtk3agg: yes [installing, version 3.20.24]
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-install-rmmlm0i3/matplotlib/setup.py", line 197, in <module>
        msg = pkg.install_help_msg()
      File "/tmp/pip-install-rmmlm0i3/matplotlib/setupext.py", line 592, in install_help_msg
        release = platform.linux_distribution()[0].lower()
    AttributeError: module 'platform' has no attribute 'linux_distribution'
                 gtk3cairo: yes [installing, version 3.20.24]
                    gtkagg: no  [Requires pygtk]
                     tkagg: yes [installing; run-time loading from Python Tcl /
                            Tk]
                     wxagg: no  [requires wxPython]
                       gtk: no  [Requires pygtk]
                       agg: yes [installing]
                     cairo: yes [installing, pycairo version 1.16.2]
                 windowing: no  [Microsoft Windows only]
    
    OPTIONAL LATEX DEPENDENCIES
                    dvipng: no
               ghostscript: yes [version 9.50]
                     latex: no
                   pdftops: yes [version 0.86.1]
    
    OPTIONAL PACKAGE DATA
                      dlls: no  [skipping due to configuration]
    
    ============================================================================
                            * The following required packages can not be built:
                            * freetype, png
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.
ondra@ondra-ESPRIMO-P:~$ 

`

carlosperate, 2021-12-24:
@lujo @lujo6911 I've separated your log and testing into a different issue, as otherwise having two parallel conversations gets confusing: https://github.com/mu-editor/mu/issues/1951

carlosperate, 2021-12-24:
@ondragottwald I assume you didn't use a virtual environment to install Mu, is that right?

Could you run this on a terminal and paste the output?

```
/usr/bin/python3 -m pip list --verbose
```

Also, the log showed you've had issues with Mu v1.1.0b7, but in your last comment `pip install mu-editor` will install v1.0.3 as that is the last "stable" release.

Could you uninstall Mu, and then try to install again with `pip install mu-editor==1.1.0b7` and paste the output here? Thanks!

ondragottwald, 2021-12-25:
@carlosperate Yes, you guessed right. I did not use a virtual environment to install Mu.
~pokus~
Yes. I installed Mu v1.1.0b7 for the first time, but it ended up with an "error window". The second installation was to fix the first. But it didn't work out.

Here is the output of `/usr/bin/python3 -m pip list --verbose`

ondra@ondra-ESPRIMO-P:~$ /usr/bin/python3 -m pip list --verbose
Package                           Version              Location                                       Installer
--------------------------------- -------------------- ---------------------------------------------- ---------
adafruit-board-toolkit            1.1.0                /home/ondra/.local/lib/python3.8/site-packages pip      
appdirs                           1.4.4                /home/ondra/.local/lib/python3.8/site-packages pip      
apturl                            0.5.2                /usr/lib/python3/dist-packages                          
backcall                          0.2.0                /home/ondra/.local/lib/python3.8/site-packages pip      
backports.entry-points-selectable 1.1.1                /home/ondra/.local/lib/python3.8/site-packages pip      
black                             21.12b0              /home/ondra/.local/lib/python3.8/site-packages pip      
blinker                           1.4                  /usr/lib/python3/dist-packages                          
Brlapi                            0.7.0                /usr/lib/python3/dist-packages                          
cajarename                        19.7.15              /usr/lib/python3/dist-packages                          
certifi                           2019.11.28           /usr/lib/python3/dist-packages                          
chardet                           3.0.4                /usr/lib/python3/dist-packages                          
click                             8.0.3                /home/ondra/.local/lib/python3.8/site-packages pip      
colorama                          0.4.3                /usr/lib/python3/dist-packages                          
command-not-found                 0.3                  /usr/lib/python3/dist-packages                          
configobj                         5.0.6                /usr/lib/python3/dist-packages                          
cryptography                      2.8                  /usr/lib/python3/dist-packages                          
cupshelpers                       1.0                  /usr/lib/python3/dist-packages                          
dbus-python                       1.2.16               /usr/lib/python3/dist-packages                          
decorator                         5.1.0                /home/ondra/.local/lib/python3.8/site-packages pip      
defer                             1.0.6                /usr/lib/python3/dist-packages                          
distlib                           0.3.4                /home/ondra/.local/lib/python3.8/site-packages pip      
distro                            1.4.0                /usr/lib/python3/dist-packages                          
distro-info                       0.23ubuntu1          /usr/lib/python3/dist-packages                          
entrypoints                       0.3                  /usr/lib/python3/dist-packages                          
filelock                          3.4.0                /home/ondra/.local/lib/python3.8/site-packages pip      
flake8                            4.0.1                /home/ondra/.local/lib/python3.8/site-packages pip      
folder-color-caja                 0.0.86               /usr/lib/python3/dist-packages                          
folder-color-common               0.0.86               /usr/lib/python3/dist-packages                          
gpg                               1.13.1-unknown       /usr/lib/python3/dist-packages                          
httplib2                          0.14.0               /usr/lib/python3/dist-packages                          
idna                              2.8                  /usr/lib/python3/dist-packages                          
ipykernel                         5.5.6                /home/ondra/.local/lib/python3.8/site-packages pip      
ipython                           7.30.1               /home/ondra/.local/lib/python3.8/site-packages pip      
ipython-genutils                  0.2.0                /home/ondra/.local/lib/python3.8/site-packages pip      
jedi                              0.18.1               /home/ondra/.local/lib/python3.8/site-packages pip      
jupyter-client                    6.1.12               /home/ondra/.local/lib/python3.8/site-packages pip      
jupyter-core                      4.9.1                /home/ondra/.local/lib/python3.8/site-packages pip      
keyring                           18.0.1               /usr/lib/python3/dist-packages                          
language-selector                 0.1                  /usr/lib/python3/dist-packages                          
launchpadlib                      1.10.13              /usr/lib/python3/dist-packages                          
lazr.restfulclient                0.14.2               /usr/lib/python3/dist-packages                          
lazr.uri                          1.0.3                /usr/lib/python3/dist-packages                          
louis                             3.12.0               /usr/lib/python3/dist-packages                          
[... cut ...]
Requirement already satisfied: parso<0.9.0,>=0.8.0 in ./.local/lib/python3.8/site-packages (from jedi>=0.16->ipython>=5.0.0->ipykernel<6,>=4.1->mu-editor==1.1.0b7) (0.8.3)
Requirement already satisfied: wcwidth in ./.local/lib/python3.8/site-packages (from prompt-toolkit!=3.0.0,!=3.0.1,<3.1.0,>=2.0.0->ipython>=5.0.0->ipykernel<6,>=4.1->mu-editor==1.1.0b7) (0.2.5)
Requirement already satisfied: pyparsing!=3.0.5,>=2.0.2 in ./.local/lib/python3.8/site-packages (from packaging->qtpy->qtconsole==4.7.7->mu-editor==1.1.0b7) (3.0.6)
Installing collected packages: mu-editor
Successfully installed mu-editor-1.1.0b7
ondra@ondra-ESPRIMO-P:~$

ondragottwald, 2021-12-25:
@carlosperate 
I tryed `mu-editor`

ondra@ondra-ESPRIMO-P:~$ mu-editor
Traceback (most recent call last):
  File "/home/ondra/.local/lib/python3.8/site-packages/mu/app.py", line 159, in run
    raise ex
  File "/home/ondra/.local/lib/python3.8/site-packages/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/ondra/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/home/ondra/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 771, in create
    self.create_venv()
  File "/home/ondra/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 803, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using /usr/bin/python3 at /home/ondra/.local/share/mu/mu_venv-38-20211225-135240
STDERR: /usr/bin/python3: No module named virtualenv
QObject::~QObject: Timers cannot be stopped from another thread
ondra@ondra-ESPRIMO-P:~$

carlosperate, 2021-12-27:
Thanks for all the testing and additional information @ondragottwald @lujo @lujo6911, it has been really useful to identify the issue!
I've collected my findings in #1954.

The easiest workaround for this would be to install Mu inside a virtual environment.

Alternatively I would personally recommend to use pipx, as it's great for installing Python-based applications like Mu: https://pypa.github.io/pipx/
pipx will automatically install Python applications in their own isolated virtual environment, and add the executable to the PATH so that it can be easily launched like any other app.

ondragottwald, 2021-12-27:
@carlosperate Thanks for your time. It looks like MU-EDITOR works for me.
I had to install virtualenv `sudo apt install python3.8-venv` and then pipx `sudo apt install pipx` and finally mu-editor `pipx install mu-editor==1.1.0b7`

carlosperate, 2021-12-27:
Awesome! Thanks for the update @ondragottwald.

Since we are tracking the general issue via https://github.com/mu-editor/mu/issues/1954 I'll close this one, thanks again for the report and all the testing!

closed 2021-12-27 by carlosperate