mu-editor/mu #1950: Crash Report

spaceddoggy, 2021-12-24:
Failed to load on Acer Chromebook

carlosperate, 2021-12-24:
Thanks for the report @spaceddoggy!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in this Linux directory: `~/.cache/mu/`

carlosperate, 2021-12-24:
Thanks for adding the Mu log to the top comment. From the log, this is the interesting part:

```
Collecting cryptography>=2.1.4 (from esptool==3.*)
  Using cached https://files.pythonhosted.org/packages/f9/4b/1cf8e281f7ae4046a59e5e39dd7471d46db9f61bb564fddbff9084c4334f/cryptography-36.0.1.tar.gz
  Saved /tmp/tmp6goh47ut/cryptography-36.0.1.tar.gz
  Installing build dependencies: started
  Installing build dependencies: finished with status 'error'
  Complete output from command /usr/bin/python3 -m pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-u6yy1sac --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --find-links /tmp/tmp6goh47ut -- setuptools>=40.6.0 wheel "cffi>=1.12; platform_python_implementation != 'PyPy'" setuptools-rust>=0.11.4:
  Looking in links: /tmp/tmp6goh47ut
  Collecting setuptools>=40.6.0
  Collecting wheel
    Downloading https://files.pythonhosted.org/packages/27/d6/003e593296a85fd6ed616ed962795b2f87709c3eee2bca4f6d0fe55c6d00/wheel-0.37.1-py2.py3-none-any.whl
  Collecting cffi>=1.12
    Using cached https://files.pythonhosted.org/packages/00/9e/92de7e1217ccc3d5f352ba21e52398372525765b2e0c4530e6eb2ba9282a/cffi-1.15.0.tar.gz
  Collecting setuptools-rust>=0.11.4
    Using cached https://files.pythonhosted.org/packages/66/ca/66bdf8f326977098eff28c314c8f825bc28d6986944c590e40ad0f74c5f0/setuptools_rust-1.1.2-py3-none-any.whl
  Collecting pycparser (from cffi>=1.12)
    Using cached https://files.pythonhosted.org/packages/62/d5/5f610ebe421e85889f2e55e33b7f9a6795bd982198517d912eb1c76e1a53/pycparser-2.21-py2.py3-none-any.whl
  Collecting semantic-version<3,>=2.8.2 (from setuptools-rust>=0.11.4)
    Using cached https://files.pythonhosted.org/packages/a5/15/00ef3b7888a10363b7c402350eda3acf395ff05bebae312d1296e528516a/semantic_version-2.8.5-py2.py3-none-any.whl
  Collecting typing-extensions>=3.7.4.3 (from setuptools-rust>=0.11.4)
  Building wheels for collected packages: cffi
    Running setup.py bdist_wheel for cffi: started
    Running setup.py bdist_wheel for cffi: finished with status 'error'
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-ql7b8oa_/cffi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-z1b6z9v5 --python-tag cp37:
    running bdist_wheel
    running build
    running build_py
    creating build
    creating build/lib.linux-x86_64-3.7
    creating build/lib.linux-x86_64-3.7/cffi
    copying cffi/__init__.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/api.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/backend_ctypes.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/cffi_opcode.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/commontypes.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/cparser.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/error.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/ffiplatform.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/lock.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/model.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/pkgconfig.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/recompiler.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/setuptools_ext.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/vengine_cpy.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/vengine_gen.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/verifier.py -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/_cffi_include.h -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/parse_c_type.h -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/_embedding.h -> build/lib.linux-x86_64-3.7/cffi
    copying cffi/_cffi_errors.h -> build/lib.linux-x86_64-3.7/cffi
    running build_ext
    building '_cffi_backend' extension
    creating build/temp.linux-x86_64-3.7
    creating build/temp.linux-x86_64-3.7/c
    x86_64-linux-gnu-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -g -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -fPIC -DUSE__THREAD -DHAVE_SYNC_SYNCHRONIZE -I/usr/include/ffi -I/usr/include/libffi -I/usr/include/python3.7m -c c/_cffi_backend.c -o build/temp.linux-x86_64-3.7/c/_cffi_backend.o
    c/_cffi_backend.c:15:10: fatal error: ffi.h: No such file or directory
     #include <ffi.h>
              ^~~~~~~
    compilation terminated.
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
    ----------------------------------------
    Failed building wheel for cffi
    Running setup.py clean for cffi
  Failed to build cffi
  Installing collected packages: setuptools, wheel, pycparser, cffi, semantic-version, typing-extensions, setuptools-rust
    Running setup.py install for cffi: started
      Running setup.py install for cffi: finished with status 'error'
      Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-ql7b8oa_/cffi/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-record-wuteqyg5/install-record.txt --single-version-externally-managed --prefix /tmp/pip-build-env-u6yy1sac --compile:
[... 35 lines cut ...]
      error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
[... cut ...]
  File "/home/rowleycolin/.local/lib/python3.7/site-packages/mu/virtual_environment.py", line 878, in install_baseline_packages
    wheels.download(zipped_wheels_filepath, logger)
  File "/home/rowleycolin/.local/lib/python3.7/site-packages/mu/wheels/__init__.py", line 192, in download
    pip_download(temp_dirpath, logger)
  File "/home/rowleycolin/.local/lib/python3.7/site-packages/mu/wheels/__init__.py", line 130, in pip_download
    "Pip was unable to download %s" % pip_identifier
mu.wheels.WheelsDownloadError: Pip was unable to download esptool==3.*
```

carlosperate, 2021-12-24:
@spaceddoggy I assume you've installed Mu using the system Python and pip and not in a virtual environment, is that correct?

What version of pip did you use to install Mu?
```
/usr/bin/python3 -m pip --version
```

spaceddoggy, 2021-12-24:
i installed it under the linux operating system on chromebook this may be a
virtual environment but i dont know enough about it to be sure
Regards
Colin R

☏ 07851225971
☏ 01684 310628

spaceddoggy, 2021-12-24:
pip 18.1

Regards
Colin R

☏ 07851225971
☏ 01684 310628

carlosperate, 2021-12-24:
Right, thanks @spaceddoggy!

If you haven't manually created a virtual environment then it was installed in the system Python. That's not wrong, but it's usually better to install things in virtual environments to avoid package versions clashing.
I personally really like pipx, as it takes care of isolating Python apps into their own virtual environment and make then accessible in the PATH like a normal app installation: https://pypa.github.io/pipx/

I think this issue is similar to this: https://github.com/mu-editor/mu/issues/1441
If you prefer not to use pipx, you can update your pip version and try again:

```
python3 -m pip install --upgrade pip
```

```
python3 -m pip uninstall mu-editor
```

```
python3 -m pip install mu-editor==1.1.0b7
```

spaceddoggy, 2021-12-24:
Thank you. Have a merry Christmas and a happy new year

Best Wishes
Colin R

spaceddoggy, 2021-12-24:
Traceback (most recent call last):
  File "/home/rowleycolin/.local/lib/python3.7/site-packages/mu/app.py",
line 159, in run
    raise ex
  File "/home/rowleycolin/.local/lib/python3.7/site-packages/mu/app.py",
line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/rowleycolin/.local/lib/python3.7/site-packages/mu/virtual_environment.py",
line 603, in ensure_and_create
    self.create()
  File "/home/rowleycolin/.local/lib/python3.7/site-packages/mu/virtual_environment.py",
line 771, in create
    self.create_venv()
  File "/home/rowleycolin/.local/lib/python3.7/site-packages/mu/virtual_environment.py",
line 805, in create_venv
    % (sys.executable, self.path, compact(output))
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create
a virtual environment using /usr/bin/python3 at
/home/rowleycolin/.local/share/mu/mu_venv-37-20211224-182536

STDERR: /usr/bin/python3: No module named virtualenv at least i get a
different error now!

Regards
Colin R

carlosperate, 2021-12-24:
Oh, interesting, thanks @spaceddoggy !

Could you uninstall Mu again, then install it once more and copy paste the full output from `python3 -m pip install mu-editor==1.1.0b7`?
I'm hoping this steps have some issues, as the `virtualenv` module should have been installed in this step.

spaceddoggy, 2021-12-25:
Don't worry too much I may just revert to my PC instead
2-24 07:48:32,929 - root:269(run) INFO:

-----------------

Starting Mu 1.1.0.beta.6
2021-12-24 07:48:32,935 - root:270(run) INFO: uname_result(system='Linux',
node='penguin', release='5.4.151-16908-gff376e5d5ee1', version='#1 SMP
PREEMPT Sat Dec 11 19:28:06 PST 2021', machine='x86_64', processor='')
2021-12-24 07:48:32,939 - root:271(run) INFO: Platform:
Linux-5.4.151-16908-gff376e5d5ee1-x86_64-with-debian-10.11
2021-12-24 07:48:32,939 - root:272(run) INFO: Python path:
['/home/rowleycolin/.local/bin', '/usr/lib/python37.zip',
'/usr/lib/python3.7', '/usr/lib/python3.7/lib-dynload',
'/home/rowleycolin/.local/lib/python3.7/site-packages',
'/usr/local/lib/python3.7/dist-packages', '/usr/lib/python3/dist-packages',
'/home/rowleycolin/.local/lib/python3.7/site-packages/IPython/extensions']
2021-12-24 07:48:32,940 - root:273(run) INFO: Language code: en_US
2021-12-24 07:48:32,941 - mu.settings:220(load) WARNING: No settings file
found at /home/rowleycolin/.local/share/mu/settings.json; skipping
2021-12-24 07:48:34,077 - mu.virtual_environment:585(ensure_and_create)
INFO: Added log handler.
2021-12-24 07:48:34,445 - mu.virtual_environment:596(ensure_and_create)
DEBUG: Checking virtual environment; attempt #1.
2021-12-24 07:48:34,446 - mu.virtual_environment:619(ensure_and_create)
WARNING: Venv created by Mu version -no-version-; Current Mu is version
1.1.0.beta.6
2021-12-24 07:48:34,452 - mu.virtual_environment:917(installed_packages)
INFO: Discovering installed third party modules in venv.
2021-12-24 07:48:34,458 - mu.virtual_environment:118(run_blocking) INFO:
About to run blocking
/home/rowleycolin/.local/share/mu/mu_venv-37-20211221-171622/bin/pip with
args ['list', '--disable-pip-version-check'] and envvars {}
2021-12-24 07:48:37,764 - mu.virtual_environment:168(wait) DEBUG: Finished:
True; exitStatus 0; exitCode 0
2021-12-24 07:48:37,765 - mu.virtual_environment:249(run) DEBUG: Process
output: Package    Version
---------- -------
pip        21.3.1
setuptools 58.3.0
wheel      0.37.0
2021-12-24 07:48:37,766 - mu.virtual_environment:926(installed_packages)
INFO: ['pip', 'setuptools', 'wheel']
2021-12-24 07:48:37,771 - mu.virtual_environment:540(quarantine_venv) INFO:
Quarantined /home/rowleycolin/.local/share/mu/mu_venv-37-20211221-171622 as
/home/rowleycolin/.local/share/mu/mu_venv-37-20211221-171622.SUPERSEDED
2021-12-24 07:48:37,779 - mu.virtual_environment:460(relocate) DEBUG:
Relocating to /home/rowleycolin/.local/share/mu/mu_venv-37-20211224-074837
2021-12-24 07:48:37,780 - mu.virtual_environment:477(relocate) DEBUG:
Virtual environment set up mu_venv-37-20211224-074837 at
/home/rowleycolin/.local/share/mu/mu_venv-37-20211224-074837
2021-12-24 07:48:37,781 - mu.virtual_environment:777(create_venv) INFO:
Creating virtualenv:
/home/rowleycolin/.local/share/mu/mu_venv-37-20211224-074837
2021-12-24 07:48:37,782 - mu.virtual_environment:778(create_venv) INFO:
Virtualenv name: mu_venv-37-20211224-074837
2021-12-24 07:48:37,783 - mu.virtual_environment:429(run_subprocess) INFO:
Running ('/usr/bin/python3', '-m', 'virtualenv', '-p', '/usr/bin/python3',
'-q', '/home/rowleycolin/.local/share/mu/mu_venv-37-20211224-074837') with
kwargs {'env': {'SHELL': '/bin/bash', 'QT_ACCESSIBILITY': '1',
'XDG_CONFIG_HOME': '/home/rowleycolin/.config', 'XCURSOR_SIZE': '24',
'DISPLAY_LOW_DENSITY': ':1', 'PWD': '/home/rowleycolin', 'LOGNAME':
'rowleycolin', 'QT_QPA_PLATFORMTHEME': 'gtk2',
'WAYLAND_DISPLAY_LOW_DENSITY': 'wayland-1', 'XCURSOR_SIZE_LOW_DENSITY':
'12', 'SOMMELIER_VERSION': '0.20', 'HOME': '/home/rowleycolin', 'LANG':
'en_US.UTF-8', 'LS_COLORS':
'rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=30;41:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:',
'WAYLAND_DISPLAY': 'wayland-0', 'INVOCATION_ID':
'cc80dc5dc3674de59aa682be29a71c5c', 'MANAGERPID': '105',
'SOMMELIER_DRM_DEVICE': '/dev/dri/renderD128', 'TERM': 'xterm-256color',
[... 950 lines cut ...]
2021-12-24 15:38:23,658 - mu.virtual_environment:778(create_venv) INFO:
Virtualenv name: mu_venv-37-20211224-153823
[... cut ...]
'QT_AUTO_SCREEN_SCALE_FACTOR': '1', 'JOURNAL_STREAM': '9:18779',
'XCURSOR_THEME': 'Adwaita', 'XDG_DATA_DIRS':
'/home/rowleycolin/.local/share:/home/rowleycolin/.local/share/flatpak/exports/share:/var/lib/flatpak/exports/share:/usr/local/share:/usr/share',
'BROWSER': '/usr/bin/garcon-url-handler', 'PATH':
'/home/rowleycolin/.local/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games',
'DBUS_SESSION_BUS_ADDRESS': 'unix:path=/run/user/1000/bus', 'LD_ARGV0_REL':
'../bin/vshd', '_': '/home/rowleycolin/.local/bin/mu-editor

carlosperate, 2021-12-27:
Hi @spaceddoggy,

Okay, so I was able to identified the last issue you had as well as this one: https://github.com/mu-editor/mu/issues/1954

The easiest workaround for this would be to install Mu inside a virtual environment.

Alternatively I would personally recommend to use pipx, as it's great for installing Python-based applications like Mu: https://pypa.github.io/pipx/
pipx will automatically install Python applications in their own isolated virtual environment, and add the executable to the PATH so that it can be easily launched like any other app.

spaceddoggy, 2022-01-01:
Thank you so much for your help. I installed pipx and reinstalled mu-editor
and it seems to work fine now.
Best wishes for a happy, healthy and interesting New Year.
Regards
Colin R

☏ 07851225971
☏ 01684 310628

carlosperate, 2022-01-01:
That's great!
As we are tracking the issue in https://github.com/mu-editor/mu/issues/1954 I'll close this one.
Thanks for the report and the debugging!

closed 2022-01-01 by carlosperate