mu-editor/mu #1953: Crash Report - Mu crashed on first execution after installing

Inkblotr, 2021-12-26:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0b7
Locale: en_US
Platform: Linux 5.4.0-91-generic #102-Ubuntu SMP Fri Nov 5 16:31:28 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/home/michael/.local/lib/python3.8/site-packages/mu/app.py", line 159, in run
    raise ex
  File "/home/michael/.local/lib/python3.8/site-packages/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/michael/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/home/michael/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 771, in create
    self.create_venv()
  File "/home/michael/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 803, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using /usr/bin/python3 at /home/michael/.local/share/mu/mu_venv-38-20211225-205307
STDERR: /usr/bin/python3: No module named virtualenv

```
Hardware is a Dell Latitude E6230, 6GB RAM, 500GB SSD, not docked, Ubuntu Budgie 20.04.03, fresh install, laptop to be my learning environment for Python programming.  Crash report attached.
[mu.log](https://github.com/mu-editor/mu/files/7776492/mu.log)

Please let me know if I can provide more information to help fix the problem.

Inkblotr, 2021-12-26:
On a hunch I issued a "pip list" and redirected the list to a file.  I don't know if that would be helpful or not...
[pip-list.txt](https://github.com/mu-editor/mu/files/7776530/pip-list.txt)

carlosperate, 2021-12-27:
Thanks for the report and log @Inkblotr, it has been really useful to identify the issue!
I've collected my findings in https://github.com/mu-editor/mu/issues/1954.

The easiest workaround for this would be to install Mu inside a virtual environment.

I personally would alternatively recommend to use pipx, as it's great for installing Python-based applications: https://pypa.github.io/pipx/
pipx will automatically install Python applications in their own isolated virtual environment, and add the executable to the PATH so that it can be easily launched like any other app.