mu-editor/mu #1954: Issue launching virtualenv module isolated from user's environment

carlosperate, 2021-12-26:


carlosperate, 2021-12-26:
Possible solutions:
- Detect if Mu is running from an installer and only apply the flag in that environment
- Try to import virtualenv (or mu) with the -I flag and if it fails launch the virtualenv command without it
- Document that if a user in Debian-based distros doesn't want to user venvs they might need to pip install with the `--system` flag
    - This is highly discouraged, as it can mess with packages that the OS needs 
- others?

tmontes, 2022-03-21:
@carlosperate, just released `pup` 1.0.0a15 that defaults to launching with the `-I` flag.

rickcox, 2022-05-12:
SUGGESTION - You would see alot less newbies (like Me!) running into this if you were
to update the Mu install instructions at:
         https://codewith.mu/en/howto/1.1/install_with_python
to recommend installing in a virtual environment.
Ref: (Dup Issue #2144)

carlosperate, 2022-05-16:
Thanks for the feedback @rickcox!

Technically speaking, the instructions already indicate this:
[image]

But it is not very visible and I agree that having this emphasised would likely help users.