mu-editor/mu #1955: Crash Report

AKAMEDIASYSTEM, 2021-12-27:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0b7
Locale: en_US
Platform: Windows 10 10.0.19043 AMD64

```
s\mu\logic.py", line 721, in check_usb
    detected = mode.find_devices(with_logging=False)
  File "C:\Users\nfeeh\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\base.py", line 423, in find_devices
    device = self.compatible_board(port)
  File "C:\Users\nfeeh\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\circuitpython.py", line 295, in compatible_board
    for comport in circuitpython_serial.repl_comports():
  File "C:\Users\nfeeh\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\circuitpython_serial.py", line 69, in repl_comports
    for port in comports()
  File "C:\Users\nfeeh\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\circuitpython_serial.py", line 49, in comports
    ports = _list_ports_windows.comports()
  File "C:\Users\nfeeh\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\_list_ports_windows.py", line 458, in comports
    return list(iterate_comports())
  File "C:\Users\nfeeh\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\_list_ports_windows.py", line 342, in iterate_comports
    info = list_ports_common.ListPortInfo(port_name_buffer.value, skip_link_detection=True)
TypeError: __init__() got an unexpected keyword argument 'skip_link_detection'

```

AKAMEDIASYSTEM, 2021-12-27:
-----------------

Starting Mu 1.1.0b7
2021-12-26 20:25:05,607 - root:270(run) INFO: uname_result(system='Windows', node='aka-thinkpad', release='10', version='10.0.19043', machine='AMD64', processor='Intel64 Family 6 Model 142 Stepping 10, GenuineIntel')
2021-12-26 20:25:05,652 - root:271(run) INFO: Platform: Windows-10-10.0.19043-SP0
2021-12-26 20:25:05,652 - root:272(run) INFO: Python path: ['C:\\Users\\nfeeh\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\nfeeh\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\nfeeh\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\nfeeh\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\nfeeh\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\nfeeh\\AppData\\Roaming\\Python\\Python38\\site-packages', 'C:\\Users\\nfeeh\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\nfeeh\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-57.4.0-py3.8.egg', 'C:\\Users\\nfeeh\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-21.1.3-py3.8.egg', 'C:\\Users\\nfeeh\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\nfeeh\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\nfeeh\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\nfeeh\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2021-12-26 20:25:05,652 - root:273(run) INFO: Language code: en_US
2021-12-26 20:25:05,652 - mu.settings:220(load) WARNING: No settings file found at C:\Users\nfeeh\AppData\Local\python\mu\settings.json; skipping
2021-12-26 20:25:05,727 - mu.virtual_environment:585(ensure_and_create) INFO: Added log handler.
2021-12-26 20:25:05,728 - mu.virtual_environment:596(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.2021-12-26 20:25:05,728 - mu.virtual_environment:685(ensure_path) INFO: Virtual Environment found at: C:\Users\nfeeh\AppData\Local\python\mu\mu_venv-38-20211226-194952
2021-12-26 20:25:06,010 - mu.virtual_environment:698(ensure_interpreter) INFO: Interpreter found at: C:\Users\nfeeh\AppData\Local\python\mu\mu_venv-38-20211226-194952\scripts\python.exe
2021-12-26 20:25:06,011 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\nfeeh\\AppData\\Local\\python\\mu\\mu_venv-38-20211226-194952\\scripts\\python.exe', '-c', 'import sys; print("%s%s" % sys.version_info[:2])') with kwargs {'shell': True}
2021-12-26 20:25:06,146 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 38
2021-12-26 20:25:06,146 - mu.virtual_environment:730(ensure_interpreter_version) INFO: Both interpreters at version 38
2021-12-26 20:25:06,146 - mu.virtual_environment:759(ensure_pip) INFO: Pip found at: C:\Users\nfeeh\AppData\Local\python\mu\mu_venv-38-20211226-194952\scripts\pip.exe
2021-12-26 20:25:06,146 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: pgzero
2021-12-26 20:25:06,146 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\nfeeh\\AppData\\Local\\python\\mu\\mu_venv-38-20211226-194952\\scripts\\python.exe', '-c', 'import pgzero') with kwargs {'shell': True}
2021-12-26 20:25:06,254 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output:
2021-12-26 20:25:06,254 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: flask
2021-12-26 20:25:06,254 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\nfeeh\\AppData\\Local\\python\\mu\\mu_venv-38-20211226-194952\\scripts\\python.exe', '-c', 'import flask') with kwargs {'shell': True}
2021-12-26 20:25:06,711 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output:
2021-12-26 20:25:06,711 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: ipykernel
2021-12-26 20:25:06,711 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\nfeeh\\AppData\\Local\\python\\mu\\mu_venv-38-20211226-194952\\scripts\\python.exe', '-c', 'import ipykernel') with kwargs {'shell': True}
2021-12-26 20:25:07,376 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output:
2021-12-26 20:25:07,377 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: ipython_genutils
2021-12-26 20:25:07,377 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\nfeeh\\AppData\\Local\\python\\mu\\mu_venv-38-20211226-194952\\scripts\\python.exe', '-c', 'import ipython_genutils') with kwargs {'shell': True}
2021-12-26 20:25:07,483 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output:
2021-12-26 20:25:07,483 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: esptool
2021-12-26 20:25:07,484 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\nfeeh\\AppData\\Local\\python\\mu\\mu_venv-38-20211226-194952\\scripts\\python.exe', '-c', 'import esptool') with kwargs {'shell': True}
2021-12-26 20:25:07,631 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output:
2021-12-26 20:25:07,631 - mu.virtual_environment:628(ensure_and_create) INFO: Valid virtual environment found at C:\Users\nfeeh\AppData\Local\python\mu\mu_venv-38-20211226-194952
2021-12-26 20:25:07,631 - mu.settings:169(save) DEBUG: Saving to C:\Users\nfeeh\AppData\Local\python\mu\venv.json
[... cut ...]
  File "C:\Users\nfeeh\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\adafruit_board_toolkit\_list_ports_windows.py", line 342, in iterate_comports
    info = list_ports_common.ListPortInfo(port_name_buffer.value, skip_link_detection=True)
TypeError: __init__() got an unexpected keyword argument 'skip_link_detection'
2021-12-26 20:25:08,847 - mu.settings:169(save) DEBUG: Saving to C:\Users\nfeeh\AppData\Local\python\mu\session.json
2021-12-26 20:25:08,847 - mu.settings:169(save) DEBUG: Saving to C:\Users\nfeeh\AppData\Local\python\mu\venv.json

carlosperate, 2021-12-27:
Thanks for the report and the log @AKAMEDIASYSTEM!

Could you run this in a command prompt and copy paste the result here? That'd be really useful to help us identify the problem:

```
"%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -m pip list --verbose
```

AKAMEDIASYSTEM, 2021-12-27:
PS C:\Users\nfeeh> python -m pip list --verbose
Package          Version      Location                                                                Installer
---------------- ------------ ----------------------------------------------------------------------- ---------
adafruit-nrfutil 0.5.3.post12 c:\users\nfeeh\appdata\roaming\python\python38\site-packages
altgraph         0.17         c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages pip
antlib           1.1b0        c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages pip
certifi          2019.11.28   c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages pip
chardet          3.0.4        c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages pip
click            7.1.2        c:\users\nfeeh\appdata\roaming\python\python38\site-packages            pip
crcmod           1.7          c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages
ecdsa            0.15         c:\users\nfeeh\appdata\roaming\python\python38\site-packages            pip
future           0.18.2       c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages
idna             2.9          c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages pip
imutils          0.5.3        c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages
intelhex         2.2.1        c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages pip
ipaddress        1.0.23       c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages pip
libusb1          1.8          c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages
nrfutil          6.1.0        c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages
numpy            1.18.1       c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages pip
pc-ble-driver-py 0.14.2       c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages pip
pefile           2019.4.18    c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages
piccata          2.0.1        c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages pip
Pillow           7.0.0        c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages pip
pip              20.2.2       c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages pip
protobuf         3.12.2       c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages pip
PyInstaller      3.6          c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages pip
pyserial         3.4          c:\users\nfeeh\appdata\roaming\python\python38\site-packages            pip
pyspinel         1.0.1        c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages pip
pywin32-ctypes   0.2.0        c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages pip
PyYAML           5.3.1        c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages pip
pyzmq            19.0.0       c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages pip
requests         2.23.0       c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages pip
setuptools       41.2.0       c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages pip
six              1.15.0       c:\users\nfeeh\appdata\roaming\python\python38\site-packages            pip
tornado          6.0.4        c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages pip
tqdm             4.48.0       c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages pip
urllib3          1.25.8       c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages pip
wrapt            1.12.1       c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages
zmq              0.0.0        c:\users\nfeeh\appdata\local\programs\python\python38\lib\site-packages
WARNING: You are using pip version 20.2.2; however, version 21.3.1 is available.
You should consider upgrading via the 'C:\Users\nfeeh\AppData\Local\Programs\Python\Python38\python.exe -m pip install --upgrade pip' command.

AKAMEDIASYSTEM, 2021-12-27:
(I upgraded pip per the cli recommendation and still cannot start mu)

carlosperate, 2021-12-27:
Hi @AKAMEDIASYSTEM,

Thanks for sending the system python pip ouput, however we need that command from the Python interpreter installed with Mu (Mu brings its own isolated Python).

Could you run the command I posted earlier and paste he output here? Thanks!
```
"%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -m pip list --verbose
```

AKAMEDIASYSTEM, 2021-12-27:
Sure, that line gave an error so I thought I was misreading your instructions.

PS C:\Users\nfeeh> "%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -m pip list --verbose
At line:1 char:55
+ "%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -m pip list --v ...
+                                                       ~~
Unexpected token '-m' in expression or statement.
At line:1 char:58
+ "%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -m pip list --v ...
+                                                          ~~~
Unexpected token 'pip' in expression or statement.
    + CategoryInfo          : ParserError: (:) [], ParentContainsErrorRecordException
    + FullyQualifiedErrorId : UnexpectedToken

carlosperate, 2021-12-27:
Ah, I see! Thanks for trying again, I think the issue is that the command was executed in PowerShell, could you try again in the "normal" command prompt? (`cmd.exe`)

AKAMEDIASYSTEM, 2021-12-27:
oh sorry! I forget cmd.exe exists...

When I run that exact command, I see a pop-up window titled "Unsupported 16-bit application" with some text saying basically program cannot run because of incompatability with 64-bit Windows", and then the command line has this error:
'''
This version of C:\Users\nfeeh\AppData\Local\Programs\Mu Editor\Python\python.exe is not compatible with the version of Windows you're running. Check your computer's system information and then contact the software publisher.
'''

carlosperate, 2021-12-27:
Oh, weird, what version of Windows are you using?

AKAMEDIASYSTEM, 2021-12-27:
Windows 10 Pro, Version 21H1 (OS Build 19043.1348)

carlosperate, 2021-12-27:
That's really weird, the Python executable is x86_64 compatible, so not sure why it would throw that error, I think this is the first time we've seen this.
Not only that, but in order to have raise the originally error that python.exe **must have run** before, so not sure why it is giving you this error now 🤷‍♂️
Did something change from the the time Mu was installed until you've run that command? If you try to launch Mu again, does it throw the same original error or does it throw this new "incompatible application" one?

Could you try downloading beta7 again, uninstalling Mu and installing it again? Then trying `"%LOCALAPPDATA%\Programs\Mu Editor\Python\python.exe" -m pip list --verbose` once more.

carlosperate, 2022-05-10:
Thanks for all your help and testing @AKAMEDIASYSTEM, and sorry it's taken a while, but one of the latest fixes we've added could resolve the issues that you were having

Would you be able to test the windows installer from this page? https://github.com/mu-editor/mu/actions/runs/2299649753

You should be able to download it from the Artifacts section:
[image]

It would be really helpful for us if you could give this a try to see if it makes a difference. Thanks!