mu-editor/mu #1961: s$ l��< �l�<�l�c|����;�c�c��gn�dog���c8�lrl;lx�o�

KorkhovPavel, 2021-12-28:
### What were you trying to do?

Run script

### What steps did you take to trigger the issue?

Run script:
from machine import Pin
import time

led_pin = Pin(2,Pin.OUT)

while True:
    led_pin.on()
    time.sleep(1)
    led_pin.off()

    time.sleep(1)

### What did you expect to happen?

blinking

### What actually happened?

nothing

in the console: 
s$ l��< �l�<�l�c|����;�c�c��gn�dog���c8�lrl;lx�o�

### Operating System Version

Ubuntu 20.04.3 LTS

### Mu Version

mu-editor==1.1.0b7

### Other Info

installed as indicated here:
https://codewith.mu/en/howto/1.1/install_with_python

thanks for the help

### Editor Log

2021-12-29 00:03:54,885 - root:269(run) INFO: 

-----------------

Starting Mu 1.1.0b7
2021-12-29 00:03:54,885 - root:270(run) INFO: uname_result(system='Linux', node='psg-HP-Laptop-14s-dq1xxx', release='5.11.0-43-generic', version='#47~20.04.2-Ubuntu SMP Mon Dec 13 11:06:56 UTC 2021', machine='x86_64', processor='x86_64')
2021-12-29 00:03:54,892 - root:271(run) INFO: Platform: Linux-5.11.0-43-generic-x86_64-with-glibc2.29
2021-12-29 00:03:54,892 - root:272(run) INFO: Python path: ['/home/psg/PycharmProjects/Wem0s_d1_mini/venv/bin', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/home/psg/PycharmProjects/Wem0s_d1_mini/venv/lib/python3.8/site-packages', '/home/psg/PycharmProjects/Wem0s_d1_mini/venv/lib/python3.8/site-packages/IPython/extensions']
2021-12-29 00:03:54,892 - root:273(run) INFO: Language code: ru_RU
2021-12-29 00:03:55,035 - mu.virtual_environment:585(ensure_and_create) INFO: Added log handler.
2021-12-29 00:03:55,036 - mu.virtual_environment:596(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2021-12-29 00:03:55,036 - mu.virtual_environment:685(ensure_path) INFO: Virtual Environment found at: /home/psg/.local/share/mu/mu_venv-38-20211228-155212
2021-12-29 00:03:55,036 - mu.virtual_environment:698(ensure_interpreter) INFO: Interpreter found at: /home/psg/.local/share/mu/mu_venv-38-20211228-155212/bin/python
2021-12-29 00:03:55,036 - mu.virtual_environment:429(run_subprocess) INFO: Running ('/home/psg/.local/share/mu/mu_venv-38-20211228-155212/bin/python', '-c', 'import sys; print("%s%s" % sys.version_info[:2])') with kwargs {'shell': False}
2021-12-29 00:03:55,062 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 38
2021-12-29 00:03:55,062 - mu.virtual_environment:730(ensure_interpreter_version) INFO: Both interpreters at version 38
2021-12-29 00:03:55,062 - mu.virtual_environment:759(ensure_pip) INFO: Pip found at: /home/psg/.local/share/mu/mu_venv-38-20211228-155212/bin/pip
2021-12-29 00:03:55,062 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: pgzero
2021-12-29 00:03:55,062 - mu.virtual_environment:429(run_subprocess) INFO: Running ('/home/psg/.local/share/mu/mu_venv-38-20211228-155212/bin/python', '-c', 'import pgzero') with kwargs {'shell': False}
2021-12-29 00:03:55,085 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2021-12-29 00:03:55,086 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: flask
2021-12-29 00:03:55,086 - mu.virtual_environment:429(run_subprocess) INFO: Running ('/home/psg/.local/share/mu/mu_venv-38-20211228-155212/bin/python', '-c', 'import flask') with kwargs {'shell': False}
2021-12-29 00:03:55,220 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
[... 1284 lines cut ...]
matplotlib-inline 0.1.3
nest-asyncio      1.5.4
numpy             1.21.5
parso             0.8.3
pexpect           4.8.0
pgzero            1.2.1
pickleshare       0.7.5
pip               21.3.1
prompt-toolkit    3.0.24
ptyprocess        0.7.0
pycparser         2.21
pygame            2.1.2
Pygments          2.10.0
pyserial          3.5
python-dateutil   2.8.2
pyzmq             22.3.0
reedsolo          1.5.4
setuptools        60.1.0
six               1.16.0
tornado           6.1
traitlets         5.1.1
wcwidth           0.2.5
Werkzeug          2.0.2
wheel             0.37.1
2021-12-29 00:54:51,566 - mu.virtual_environment:931(installed_packages) INFO: []

carlosperate, 2021-12-28:
Thanks for the report @KorkhovPavel! 

> ### What actually happened?
> 
> nothing
> 
> in the console: s$�l��<��l�<�����l��c|������;�c��c��gn�dog����c�8�lrl;lx�o���

Is that in the Ubuntu terminal? As an output from the Mu application?

What Mu Mode are you using and which board did you use?

KorkhovPavel, 2021-12-29:
It's in Mu app

Mode: ESP MicroPython
Write MicroPython on ESP8266 / ESP32 boards.

device wemos d1 mini

carlosperate, 2021-12-29:
Thanks @KorkhovPavel, 

> It's in Mu app

Where in the Mu app does it show that, could you capture a screenshoot?

KorkhovPavel, 2021-12-29:
As if the connection is lost with the device, but the photo shows the device is connected.
Maybe something is blocking?

python version 3.8
firmware version micropythona esp8266-20210902-v1.17.bin (tried different versions)
Thank you

[image]
[image]
[image]

KorkhovPavel, 2021-12-29:
the device is exactly this
https://creatorplanet.com/index.php?route=product/product&path=1&product_id=665

carlosperate, 2021-12-29:
Looks like the issue might be with getting MicroPython running correctly on the device.

I don't know much about the ESP devices, but see if you can find the right MicroPython build and if the REPL works correctly before trying to flash any code.

KorkhovPavel, 2021-12-29:
Can you tell me the name of the board on which micropython works well?
I will try all the firmwares and write to you about the result.

Thank you

carlosperate, 2021-12-29:
The MicroPython website has a list of compatible devices: https://micropython.org/download/

One thing to try before changing boards might be to flash (what you think might be the correct) MicroPython and then try to connect to the REPL via a serial console application instead of Mu, that way you could try to connect at 9600 baud (Mu by default tries to connect at 115200 baud).
The output we see in Mu looks like it's receiving data at the baud rate different than 115200.

KorkhovPavel, 2022-01-07:
Good day.
I tried all the firmware versions, in Win10 and Ubuntu operating systems and tried to change the speed, all without result.
I bought another board and everything worked, so something is wrong with it.

I apologize for the disturbance, the matter was not in the program.
Thank you.

carlosperate, 2022-01-07:
No worries, I glad you found the issue and are able to programme the other board correctly.

I'll close this issue, but don't hesitate to open a new one if you encounter any problem with Mu, it's really useful for us to help identify possible problems in the editor 👍

closed 2022-01-07 by carlosperate