mu-editor/mu #1962: Crash Report

paymerich, 2021-12-29:
[**REMEMBER]([url](url)) TO ATTACH YOUR LOG FILE**

Version: 1.1.0b7
Locale: en_US
Platform: Windows 10 10.0.19043 AMD64

```
Traceback (most recent call last):
  File "C:\Users\pablo\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\pablo\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\pablo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\pablo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\pablo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 349, in run
    editor.setup(setup_modes(editor, editor_window))
  File "C:\Users\pablo\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 817, in setup
    os.makedirs(images_path)
  File "C:\Users\pablo\AppData\Local\Programs\Mu Editor\Python\lib\os.py", line 223, in makedirs
    mkdir(name, mode)
FileNotFoundError: [WinError 2] The system cannot find the file specified: 'G:\\images'

``
[mu.log](https://github.com/mu-editor/mu/files/7789513/mu.log)
`

paymerich, 2021-12-29:
I have done full add /removal of all older versions still errors out. also deleted APPDATA directories

carlosperate, 2021-12-29:
It looks like it's trying to configure the mu files directory as `G:\`, is that something you've manually configured in the settings/session json file?
It's G: a normal partition on your PC, or is it a removable drive like a micro:bit, CircuitPython, or something like that?

paymerich, 2021-12-29:
Mu crashes almost immediately upon start; it never gets to the UI.   On my system  G: is GoogleDrive.
I am doing a normal C: drive install of Mu btw.

paymerich, 2021-12-29:
oops

carlosperate, 2021-12-30:
Did use Mu before? It looks like it's trying to initialise the mu folder in the G:\ drive and I think it'd only do that if it's been manually configured in the `settings.json` or `session.json`.

Could you add your C:\Users\pablo\AppData\Local\python\mu\session.json file to this GitHub issue?

paymerich, 2021-12-30:
{
  "theme": "night",
  "mode": "adafruit",
  "paths": [],
  **"workspace": "G:\\",**
  "microbit_runtime_hex": null
}

I was not aware of this file but by changing the workspace location allowed it to startup..   Thanks  for your help.

closed 2021-12-30 by paymerich

carlosperate, 2021-12-30:
Glad it was resolved, thanks for the report and the debugging!