mu-editor/mu #1964: Crash Report

lutailai2, 2021-12-30:
[mu.log](https://github.com/mu-editor/mu/files/7794405/mu.log)
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0b7
Locale: en_US
Platform: Windows 10 10.0.19042 AMD64

```
Traceback (most recent call last):
  File "C:\Users\lutai\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\lutai\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\lutai\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\lutai\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\lutai\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 349, in run
    editor.setup(setup_modes(editor, editor_window))
  File "C:\Users\lutai\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 809, in setup
    os.makedirs(wd)
  File "C:\Users\lutai\AppData\Local\Programs\Mu Editor\Python\lib\os.py", line 223, in makedirs
    mkdir(name, mode)
FileExistsError: [WinError 183] Cannot create a file when that file already exists: 'C:\\Users\\lutai\\mu_code'

```

carlosperate, 2021-12-30:
Hi @lutailai2, thanks for the report and the log!

I believe the issue you've encountered is this one: https://github.com/mu-editor/mu/issues/1755

Could you try opening Mu again, but ensuring it is only executed once? 
I believe the problem arises when the Mu application is opened more than once when running for the first time. It might take a some time for the first launch to complete, but after that it should be much quicker.

lutailai2, 2021-12-30:
Got it. Working fine now. It happened during install so idk why -- but resolved nonetheless...

carlosperate, 2021-12-30:
I'm glad it worked now!
Since the issue is being tracked in https://github.com/mu-editor/mu/issues/1755 I'll close this crash report.
Thanks again for sending it and helping us debug the issue!

closed 2021-12-30 by carlosperate