mu-editor/mu #1965: Crash Report

jhanarato, 2022-01-01:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0b7
Locale: en_AU
Platform: Linux 5.4.0-91-generic #102~18.04.1-Ubuntu SMP Thu Nov 11 14:46:36 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/home/jr/Code/mu/mu-venv/lib/python3.6/site-packages/mu/app.py", line 159, in run
    raise ex
  File "/home/jr/Code/mu/mu-venv/lib/python3.6/site-packages/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/jr/Code/mu/mu-venv/lib/python3.6/site-packages/mu/virtual_environment.py", line 621, in ensure_and_create
    self.recreate()
  File "/home/jr/Code/mu/mu-venv/lib/python3.6/site-packages/mu/virtual_environment.py", line 566, in recreate
    self.create()
  File "/home/jr/Code/mu/mu-venv/lib/python3.6/site-packages/mu/virtual_environment.py", line 772, in create
    self.install_baseline_packages()
  File "/home/jr/Code/mu/mu-venv/lib/python3.6/site-packages/mu/virtual_environment.py", line 883, in install_baseline_packages
    wheels.download(zipped_wheels_filepath, logger)
  File "/home/jr/Code/mu/mu-venv/lib/python3.6/site-packages/mu/wheels/__init__.py", line 174, in download
    pip_download(temp_dirpath, logger)
  File "/home/jr/Code/mu/mu-venv/lib/python3.6/site-packages/mu/wheels/__init__.py", line 112, in pip_download
    "Pip was unable to download %s" % pip_identifier
mu.wheels.WheelsDownloadError: Pip was unable to download esptool==3.*

```

jhanarato, 2022-01-01:
Creating a venv with python 3.8 solves the problem.

carlosperate, 2022-01-01:
Thanks for the report @jhanarato!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to identify the underlying problem. The file should be located in: `~/.cache/mu/`

I think maybe the thing that resolved it could have been a newer version of pip on the venv. Do you know what pip version you had in the Python 3.6 venv vs the 3.8 one?

carlosperate, 2022-01-20:
Hi @jhanarato, I was wondering if you still had this issue?

closed 2022-05-31 by carlosperate