mu-editor/mu #1969: Crash Report

elfisho, 2022-01-03:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0b7
Locale: en_US
Platform: Linux 5.11.0-43-generic #47~20.04.2-Ubuntu SMP Mon Dec 13 11:06:56 UTC 2021 x86_64

```
Traceback (most recent call last):
  File "/home/donfi/.local/lib/python3.8/site-packages/mu/app.py", line 159, in run
    raise ex
  File "/home/donfi/.local/lib/python3.8/site-packages/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/donfi/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/home/donfi/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 771, in create
    self.create_venv()
  File "/home/donfi/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 803, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using /usr/bin/python3 at /home/donfi/.local/share/mu/mu_venv-38-20220103-155953
STDERR: /usr/bin/python3: No module named virtualenv

```

elfisho, 2022-01-03:
After Reboot i received that report and mu does not start.

carlosperate, 2022-01-03:
Thanks for the report @elfisho! I believe this issue you might be experiencing is this one: https://github.com/mu-editor/mu/issues/1954

Could you install the python3-virtualenv apt package (assuming Debina/Ubuntuu) and try again?

Shoes01, 2022-01-13:
I had the same problem, and installing the python3-virtualenv apt package solved it for me!

```Traceback (most recent call last):
  File "/home/jason/.local/lib/python3.8/site-packages/mu/app.py", line 159, in run
    raise ex
  File "/home/jason/.local/lib/python3.8/site-packages/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/jason/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/home/jason/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 771, in create
    self.create_venv()
  File "/home/jason/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 803, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using /usr/bin/python3 at /home/jason/.local/share/mu/mu_venv-38-20220112-211334
STDERR: /usr/bin/python3: No module named virtualenv```

carlosperate, 2022-01-20:
I'm glad that worked @Shoes01, thanks for the update!

@elfisho I was wondering if you were able to resolve the issue with this workaround?

elfisho, 2022-01-21:
sry i forgot to write
jep the workaround solved the issue
thx for the support :)

carlosperate, 2022-01-21:
Awesome, thanks for the update!
As we are tracking the issue as part of #1954 I'll close this one to keep things tidy. 👍

closed 2022-01-21 by carlosperate