mu-editor/mu #1972: Crash Report

Lalannep, 2022-01-04:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0b7
Locale: en_CA
Platform: Windows 10 10.0.19043 AMD64

```
\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\via_app_data.py", line 57, in _install
    installer.install(creator.interpreter.version_info)
  File "C:\Users\lalan\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\pip_install\base.py", line 39, in install
    self._sync(filename, into)
  File "C:\Users\lalan\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\pip_install\copy.py", line 13, in _sync
    copy(src, dst)
  File "C:\Users\lalan\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\util\path\_sync.py", line 53, in copy
    method(norm(src), norm(dest))
  File "C:\Users\lalan\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\util\path\_sync.py", line 64, in copytree
    shutil.copy(src_f, dest_f)
  File "C:\Users\lalan\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 418, in copy
    copyfile(src, dst, follow_symlinks=follow_symlinks)
  File "C:\Users\lalan\AppData\Local\Programs\Mu Editor\Python\lib\shutil.py", line 264, in copyfile
    with open(src, 'rb') as fsrc, open(dst, 'wb') as fdst:
FileNotFoundError: [Errno 2] No such file or directory: 'C:\\Users\\lalan\\AppData\\Local\\python\\mu\\mu_venv-38-20220104-141811\\Lib\\site-packages\\pip\\_internal\\commands\\debug.py'

```

carlosperate, 2022-01-04:
Thanks for the report @Lalannep!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\lalan\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2022-03-07:
Hi @Lalannep, could you try the latest Mu release? It's posible this issue has been fixed already in https://github.com/mu-editor/mu/releases/tag/v1.1.1

carlosperate, 2022-05-31:
Hi @Lalannep, I was wondering if you had a chance to try the latest v1.1.1 release?
https://codewith.mu/en/download