mu-editor/mu #1974: Cannot install on Debian 11

Sciss, 2022-01-05:
### What were you trying to do?

Install following https://codewith.mu/en/howto/1.1/install_with_python

### What steps did you take to trigger the issue?

`sudo apt install python3-pip`; `pip3 install mu-editor==1.1.0b7`

### What did you expect to happen?

installation

### What actually happened?

```
ERROR: Could not find a version that satisfies the requirement mu-editor==1.1.0b7
ERROR: No matching distribution found for mu-editor==1.1.0b7
```

Try  `pip3 install mu-editor` ; attempts to install 1.0.3 ; mathplotlib is missing dependencies which must be installed via apt. Then pygame is missing dependencies which must be installed via apt. Then pygame fails to compile due to "incompatibility with Python 3.9" (needs 3.8 LOL).

All that only to find you I can do `sudo apt install mu-editor`. You would _think_ that works. But whoever is responsible for the debian packaging apparently didn't try to run it:

```
$ mu-editor
Traceback (most recent call last):
  File "/usr/bin/mu-editor", line 33, in <module>
    sys.exit(load_entry_point('mu-editor==1.0.3', 'console_scripts', 'mu-editor')())
  File "/usr/bin/mu-editor", line 25, in importlib_load_entry_point
    return next(matches).load()
  File "/usr/lib/python3.9/importlib/metadata.py", line 77, in load
    module = import_module(match.group('module'))
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 790, in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/usr/share/mu-editor/mu/app.py", line 34, in <module>
    from mu.interface import Window
  File "/usr/share/mu-editor/mu/interface/__init__.py", line 1, in <module>
    from .main import Window
  File "/usr/share/mu-editor/mu/interface/main.py", line 38, in <module>
    from mu.interface.editor import EditorPane
  File "/usr/share/mu-editor/mu/interface/editor.py", line 25, in <module>
    from PyQt5.Qsci import QsciScintilla, QsciLexerPython, QsciAPIs
ModuleNotFoundError: No module named 'PyQt5.Qsci'
```

### Operating System Version

Debian 11 (stable)

### Mu Version

1.1.0b7 (failed), 1.0.3 (failed), 1.0.3+dfsg-2 .deb (failed)

carlosperate, 2022-01-05:
Hi @Sciss thanks for the report!

Yeah, unfortunately the debian package is very old, so I'm not surprised it doesn't work any more (it did at the time though).
As v1.0.3 in itself is quite old, the dependencies it pinned at the time are likely not going to work for relatively new versions of Python either, so it will try to build them from source and that usually requires quite a few packages installed. Some users have been able to build it and run it, but is not something I would recommend unless you have experience chasing down dependencies for old versions of packages.

This dependency hell is somewhat mitigadted in the latest beta releases, but at the moment Mu is only compatible with Python 3.5 to 3.8 (this is once again due to limitations with some of the dependencies we need), so if you are able to install it in an older version of Python that would be the easiest workaround.

knowledgejunkie, 2022-01-31:
> All that only to find you I can do `sudo apt install mu-editor`. You would _think_ that works. But whoever is responsible for the debian packaging apparently didn't try to run it:

@Sciss,

I've tested this morning on a fresh VirtualBox VM of Debian 11 "Bullseye" (using Python 3.9.2) and installed mu-editor via apt: it ran without issue. I also tested on Debian testing/sid, with Python 3.9.10, and it also started without issue.

Please provide the version of the python3-pyqt5.qsci package you have installed. On my test VM I see version 2.11.6+dfsg-2.

I can see no bugs similar to yours reported against mu-editor since the release of Debian 11 "Bullseye" in August or Ubuntu 21.10 in October. The mu-editor 1.0.3 package in Debian was last updated less than a year ago.