mu-editor/mu #1975: Totally new to this. Tried to download MU Editor but got this code after it crashed.

ckendren7, 2022-01-06:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0.beta.3
Locale: en_US
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Kendren Cox\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 442, in ensure_and_create
    self.ensure()
  File "C:\Users\Kendren Cox\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 467, in ensure
    self.ensure_path()
  File "C:\Users\Kendren Cox\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 480, in ensure_path
    raise VirtualEnvironmentError(message)
mu.virtual_environment.VirtualEnvironmentError: C:\Users\Kendren Cox\AppData\Local\python\mu\mu_venv-38-20220105-194040 does not exist.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Kendren Cox\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 300, in installed
    next(iterlines)
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Kendren Cox\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 162, in run
    raise ex
  File "C:\Users\Kendren Cox\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 149, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Kendren Cox\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 449, in ensure_and_create
    self.create()
  File "C:\Users\Kendren Cox\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 598, in create
    self.register_baseline_packages()
  File "C:\Users\Kendren Cox\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 665, in register_baseline_packages
    packages = list(self.pip.installed())
  File "C:\Users\Kendren Cox\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 305, in installed
    raise VirtualEnvironmentError("Unable to parse installed packages")
mu.virtual_environment.VirtualEnvironmentError: Unable to parse installed packages

```

carlosperate, 2022-01-06:
Thanks for the report @ckendren7, looks like you've installed an old version of Mu.
Could you uninstall it and then install the latest releasee? Hopefully that version should have resolved this issue: 
- https://github.com/mu-editor/mu/releases/tag/1.1.0-beta.7
- https://codewith.mu/en/download

carlosperate, 2022-03-07:
Hi @ckendren7, did you try a newer version of Mu? Are you still experiencing this issue?

closed 2022-06-06 by carlosperate