mu-editor/mu #1976: Crash Report

MuhammadArdhaniKusumoputrao, 2022-01-06:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0b7
Locale: en_ID
Platform: Windows 10 10.0.19042 AMD64

```
e_and_create
    self.create()
  File "C:\Users\USER\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 772, in create
    self.install_baseline_packages()
  File "C:\Users\USER\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 885, in install_baseline_packages
    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "C:\Users\USER\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 860, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "C:\Users\USER\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 270, in install
    return self.run(
  File "C:\Users\USER\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "C:\Users\USER\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "C:\Users\USER\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 259:

```

carlosperate, 2022-01-06:
Thanks for the report @MuhammadArdhaniKusumoputrao!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\USER\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2022-05-31:
Closing due to lack of activity.

closed 2022-05-31 by carlosperate