mu-editor/mu #1982: 'nothing happens' bug generated probably by non-English user name

Hanmeim, 2022-01-18:
### What steps did you take to trigger the issue?

after installing the mu(version 1.10beta5) my student told me he can't get the software started. nothing happens after double-clicking the icon. I noticed that he used non-English user name(Chinese characters) so i moved the whole folder to a new path(English only) and it worked.

### Operating System Version

win10

### Mu Version

1.10beta5

Hanmeim, 2022-01-18:
suggestions: let the user choose where to install

carlosperate, 2022-01-18:
Thanks for the report @Hanmeim!
Do you have access to the log from the computer and user account that had the problem? It would be really useful to inspect them, they should be saved in `%LOCALAPPDATA%\python\mu\Logs`

Hanmeim, 2022-01-27:
> Thanks for the report @Hanmeim! Do you have access to the log from the computer and user account that had the problem? It would be really useful to inspect them, they should be saved in `%LOCALAPPDATA%\python\mu\Logs`

I'm afraid they're not available, since the problems only occurred on my students' personal computers. I'll see if I can repeat the bug on different computers.