mu-editor/mu #1983: Crash Report

amos1969, 2022-01-18:
Version: 1.1.0b7
Locale: en_GB
Platform: Windows 10 10.0.22000 AMD64
[mu.log](https://github.com/mu-editor/mu/files/7888425/mu.log)

This looks similar to issues I was having on Windows 11 (which this machine is running contrary to what is showing above), with a fresh install of Python and Pycharm. The default install location for Python appears to have moved to one inside the User's directory structure, and Pycharm fails to create new Virtual Environments properly when the User name has a space in it.

Not sure if that helps (or is a red herring).

Dave

```
all_from_zipped_wheels(zipped_wheels_filepath)
  File "C:\Users\Dave Ames\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 860, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "C:\Users\Dave Ames\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 270, in install
    return self.run(
  File "C:\Users\Dave Ames\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "C:\Users\Dave Ames\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "C:\Users\Dave Ames\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 101:
Unable to create process using 'C:\Users\Dave Ames\AppData\Local\Programs\Mu Editor\Python\python.exe "C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-104319\scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index C:\Users\DAVEAM~1\AppData\Local\Temp\tmp80d8052c\backcall-0.2.0-py2.py3-none-any.whl'

```

amos1969, 2022-01-18:
Sadly installing earlier versions (1.1.0b6 and 1.1.0b5) gives me the same errors, when I try and run them. I uninstalled 1.1.05beta which was working to install the newer version.

carlosperate, 2022-01-18:
Hi @amos1969,

First of all, thanks for opening the issue with the crash report and providing additional information!

We were just talking about this issue yesterday in a maintainers call, as we've struggle to replicate this problem, even though it affect several users. If you are available to do some debugging with us it would be incredibly helpful to try to get to the bottom of this one.

I've been grouping similar crash reports in this issue thread: https://github.com/mu-editor/mu/issues/1926

The main common factor is that all users experiencing this problem has spaces in their Windows username, but apart from that we haven't gotten very far identifying this problem, and as you say it might be a red herring.

We've tried replicating this creating new user accounts on Windows with spaces in their name, but unable to replicate it, and some people have been able to "fix" this issue when Mu was installed in a new account without spaces. But this could be due to the fact that a new user account doesn't have other programmes or OS environmental factors affecting Mu.

In your case as you mention this is in a fresh system it offers a great opportunity to debug it without having a lot of other factors getting in the way.

Could you try to run these commands to see if they make a difference?

I suspect this one would crash the same way, but it's good to confirm it first:
```
"C:\\Users\\Dave Ames\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe" -m mu
```

Then I'm hoping this can make a difference, fingers crossed:
```
"C:\\Users\\Dave Ames\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe" -I -m mu
```
After this second command, could you attach the latest log? It'd be good to check, even if it still crashes, if it has made any difference.

amos1969, 2022-01-18:
Hi @carlosperate

I was able to get a working version onto the machine by creating a new local user, with a single word username, that didn't use a Microsoft account, so it's almost certainly something to do with the user/path, rather than the wider environment.

I ran both commands and Mu failed to launch each time with similar errors to previously. I ran:

`& 'C:\Users\Dave Ames\AppData\Local\Programs\Mu Editor\Python\pythonw.exe' -I -m mu`

for the second one, which is what Powershell converted it to.

Here's the log: [mu.log](https://github.com/mu-editor/mu/files/7890876/mu.log)

carlosperate, 2022-01-18:
In case the temp folder is still present, can you try running this command to see what the command line output would be?

```
"C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834\scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index C:\Users\DAVEAM~1\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl'
```

carlosperate, 2022-01-18:
If the temp folder Mu creates is still there (so the last command didn't raise an error saying it cannot find the `backcall-0.2.0-py2.py3-none-any.whl`) file, it would also be interesting to see if this variation of the command makes a difference:
```
"C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834\scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index "C:\Users\Dave Ames\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl"
```

carlosperate, 2022-01-18:
And if the temp folder has been removed automatically, could you try recreating it?

The `backcall-0.2.0-py2.py3-none-any.whl` wheel file can be download from here: https://pypi.org/project/backcall/0.2.0/#files
So if the temp folder is re-created and the wheel file placed there, then the previous commands should be able to find the file in this path: `C:\Users\Dave Ames\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl`

We could use other paths under the user account, but for the first debugging attempts here I'd like to recreate the steps Mu runs as close as possible, as this has been hard to replicate so far.

Thank you again for your help, it is immensely appreciated!

amos1969, 2022-01-18:
First suggestion above using the closest folder I could get (it has .FAILED after the mu_venv-somenumber part) and then a second attempt to do the same thing with an almost identical folder name but slightly different number, give me this result:

```
PS C:\Users\Dave Ames> & "C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834.FAILED\Scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index C:\Users\DAVEAM~1\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl
Fatal error in launcher: Unable to create process using '"C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834\Scripts\python.exe"  "C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834.FAILED\Scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index C:\Users\DAVEAM~1\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl': The system cannot find the file specified.

PS C:\Users\Dave Ames> & "C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163734\Scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index C:\Users\DAVEAM~1\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl
Unable to create process using 'C:\Users\Dave Ames\AppData\Local\Programs\Mu Editor\Python\python.exe "C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163734\Scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index C:\Users\DAVEAM~1\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl'
PS C:\Users\Dave Ames>
```

amos1969, 2022-01-18:
The second option that you mentioned gives me:

```
PS C:\Users\Dave Ames> & "C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834\scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index "C:\Users\Dave Ames\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl"
& : The term 'C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834\scripts\pip.exe' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path
was included, verify that the path is correct and try again.
At line:1 char:3
+ & "C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163 ...
+   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (C:\Users\Dave A...scripts\pip.exe:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

PS C:\Users\Dave Ames> & "C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163734\scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index "C:\Users\Dave Ames\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl"
Unable to create process using 'C:\Users\Dave Ames\AppData\Local\Programs\Mu Editor\Python\python.exe "C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163734\scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index "C:\Users\Dave Ames\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl"'
```

I'm guessing the first response is what Powershell does when the file isn't found.

carlosperate, 2022-01-18:
Thanks a lot @amos1969!

Are you able to manually check if the temp folder still exists?
Is the `C:\Users\Dave Ames\AppData\Local\Temp\tmpw4vm7kxq` folder still there? And if it is, does it have the `backcall-0.2.0-py2.py3-none-any.whl` file inside it?

The main reason I want to check with this path exactly is in case the `Dave Ames` -> `DAVEAM~1` conversion that Python/pip is doing on the path could be part of the problem.

If the temp folder is not there, could you create the folder and add the file? The `backcall-0.2.0-py2.py3-none-any.whl` wheel can be manually downloaded from https://pypi.org/project/backcall/0.2.0/#files.
Technically we could do this in any other folder inside your user path, but we might as well use the Temp directory, as that is where Mu is currently failing.

About the `...\mu\mu_venv-38-20220118-163834\...` directory not existing that's because Mu has marked it as a failed virtual environment, thanks for noticing!
So, as you have identified, we would need to use `mu_venv-38-20220118-163834.FAILED` instead (for as long as it exists there, but Mu currently doesn't delete these, so it should stay as `'C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834.FAILED\`)

So, just to confirm, ensuring that the `C:\Users\Dave Ames\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl` file exists, could you run these commands?

```
"C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834.FAILED\Scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index C:\Users\DAVEAM~1\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl
```

```
"C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834.FAILED\Scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index "C:\Users\Dave Ames\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl"
```

I'm really not familiar with PowerShell, sorry, so if you could run them in a normal command prompt (`cmd.exe`) that would make sure I might not have accidentally malformed the command.
I believe the way Python shells outs these commands would be with with cmd.exe (or equivalent) by default, so we could be getting different results when running these tests in PowerShell.

amos1969, 2022-01-19:
The folder in the temp directory didn't exist so I've created it and added the file to it. I'm not really familiar with PowerShell either as I usually work on a Linux machine. I've swapped to the normal command prompt. 

The first command above gives me:

```
C:\Users\Dave Ames>"C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834.FAILED\Scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index C:\Users\DAVEAM~1\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl
Fatal error in launcher: Unable to create process using '"C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834\Scripts\python.exe"  "C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834.FAILED\Scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index C:\Users\DAVEAM~1\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl': The system cannot find the file specified.
```

The second command gives me:

```
C:\Users\Dave Ames>"C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834.FAILED\Scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index "C:\Users\Dave Ames\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl"
Fatal error in launcher: Unable to create process using '"C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834\Scripts\python.exe"  "C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834.FAILED\Scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index "C:\Users\Dave Ames\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl"': The system cannot find the file specified.
```

The error appears to be from the first part (trying to run pip) rather than the rest. As an experiment, I tried upgrading pip and got the same error:

```
C:\Users\Dave Ames>"C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834.FAILED\Scripts\pip.exe" install --upgrade pip
Fatal error in launcher: Unable to create process using '"C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834\Scripts\python.exe"  "C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834.FAILED\Scripts\pip.exe" install --upgrade pip': The system cannot find the file specified.
```
The folders and pip.exe all exist.

tjguk, 2022-01-19:
Thanks @amos1969; that's very useful. As @carlosperate mentioned, we've been trying to nail this issue for ages, unable to reproduce it. (We even have a specific test for it which never fails!). The extra info you've provided can only help us move it forward.

amos1969, 2022-01-19:
Have just tried activating the virtual environment in that folder, and although it appears to activate, it's actually using the system python:

```
C:\Users\Dave Ames\AppData\Local\python\mu>mu_venv-38-20220118-163834.FAILED\Scripts\activate.bat

(mu_venv-38-20220118-163834) C:\Users\Dave Ames\AppData\Local\python\mu>python
Python 3.10.2 (tags/v3.10.2:a58ebcc, Jan 17 2022, 14:12:15) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>> exit()

(mu_venv-38-20220118-163834) C:\Users\Dave Ames\AppData\Local\python\mu>pip -V
pip 21.2.4 from C:\Users\Dave Ames\AppData\Local\Programs\Python\Python310\lib\site-packages\pip (python 3.10)

(mu_venv-38-20220118-163834) C:\Users\Dave Ames\AppData\Local\python\mu>
```

The pyvenv.cfg file in that folder is pointing to Python 3.8:

```
home = C:\Users\Dave Ames\AppData\Local\Programs\Mu Editor\Python
implementation = CPython
version_info = 3.8.11.final.0
virtualenv = 20.10.0
include-system-site-packages = false
base-prefix = C:\Users\Dave Ames\AppData\Local\Programs\Mu Editor\Python
base-exec-prefix = C:\Users\Dave Ames\AppData\Local\Programs\Mu Editor\Python
base-executable = C:\Users\Dave Ames\AppData\Local\Programs\Mu Editor\Python\pythonw.exe
```

I can run the Python executables that pyvenv.cfg is pointing to:

```
(mu_venv-38-20220118-163834) C:\Users\Dave Ames\AppData\Local\python\mu>"C:\Users\Dave Ames\AppData\Local\Programs\Mu Editor\Python\python.exe" -V
Python 3.8.11
```

carlosperate, 2022-01-19:
Ah! One thing I didn't remember before was that the virtual environment is probably configured to use the old folder name, so `mu_venv-38-20220118-163834.FAILED` would need to be renamed back to `mu_venv-38-20220118-163834`.

It's quite likely that when the virtualenv is created, it uses full paths internally, so `pip.exe` tries to execute the Python interpreter from the original folder name `...\mu_venv-38-20220118-163834\Scripts\python.exe` and because Mu had renamed that virtualenv folder to `.FAILED` it cannot find it any more.

Okay so making sure the following conditions are met:
- `C:\Users\Dave Ames\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl` file exists
- Folder `mu_venv-38-20220118-163834.FAILED` folder has been renamed to `mu_venv-38-20220118-163834` (so full path is `"C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834\`)
- Using cmd.exe

```
"C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834\Scripts\pip.exe" list --verbose
```

```
"C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834\Scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index C:\Users\DAVEAM~1\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl
```

```
"C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834\Scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index "C:\Users\Dave Ames\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl"
```

Thank you again for all your help, this debugging is really helpful!

amos1969, 2022-01-19:
First command:

```
C:\Users\Dave Ames>"C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834\Scripts\pip.exe" list --verbose
Unable to create process using 'C:\Users\Dave Ames\AppData\Local\Programs\Mu Editor\Python\python.exe "C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834\Scripts\pip.exe" list --verbose'
```

Second command:

```
C:\Users\Dave Ames>"C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834\Scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index C:\Users\DAVEAM~1\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl
Unable to create process using 'C:\Users\Dave Ames\AppData\Local\Programs\Mu Editor\Python\python.exe "C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834\Scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index C:\Users\DAVEAM~1\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl'
```

Third command:

```
C:\Users\Dave Ames>"C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834\Scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index "C:\Users\Dave Ames\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl"
Unable to create process using 'C:\Users\Dave Ames\AppData\Local\Programs\Mu Editor\Python\python.exe "C:\Users\Dave Ames\AppData\Local\python\mu\mu_venv-38-20220118-163834\Scripts\pip.exe" install --disable-pip-version-check --no-deps --no-index "C:\Users\Dave Ames\AppData\Local\Temp\tmpw4vm7kxq\backcall-0.2.0-py2.py3-none-any.whl"'
```

No problem with the debugging. We have online workshops running at February Half Term, and I can guarantee this will be a problem I'll end up having to deal with then if it doesn't get fixed. I'm currently reviewing all of our install instructions so it's kind of for work anyhow.

carlosperate, 2022-01-19:
Thanks @amos1969!

Weird that pip still cannot launch the Python interpreter!

I believe you've already tried this in your previous tests activating the virtualenv, but can you confirm that the  `C:\Users\Dave Ames\AppData\Local\Programs\Mu Editor\Python\python.exe` file exists?

And this does run, correct?
```
"C:\Users\Dave Ames\AppData\Local\Programs\Mu Editor\Python\python.exe" --version
```

Could you zip the `mu_venv-38-20220118-163834` folder and attach it to this ticket?

carlosperate, 2022-01-19:
Also, could you run this command?

```
"C:\Users\Dave Ames\AppData\Local\Programs\Mu Editor\Python\python.exe" -m pip list --verbose
```

amos1969, 2022-01-19:
Yep. I think I tried it, but just to confirm:

```
C:\Users\Dave Ames>"C:\Users\Dave Ames\AppData\Local\Programs\Mu Editor\Python\python.exe"
Python 3.8.11 (default, Jul 24 2021, 23:33:17) [MSC v.1929 64 bit (AMD64)] on win32
Type "help", "copyright", "credits" or "license" for more information.
>>>
```

Here's the zip of the folder: [mu_venv-38-20220118-163834.zip](https://github.com/mu-editor/mu/files/7898090/mu_venv-38-20220118-163834.zip)

I suspect that the issue is something not directly related to the way Mu is interacting with the system. I tried earlier to create a virtual env using the system Python and it fails (but doesn't get as far as populating the venv folder).  I get:

```
C:\Users\Dave Ames\Checking>python -m venv venv
Error: Command '['C:\\Users\\Dave Ames\\Checking\\venv\\Scripts\\python.exe', '-Im', 'ensurepip', '--upgrade', '--default-pip']' returned non-zero exit status 101.
```

Googling that brings up various other people encountering the same issue. 

Pycharm also fails when trying to create a Virtual Environment with a message about an Invalid SDK. Digging into that it seems to be related to having a space in the User's folder name.

The solution to both of these is apparently to reinstall Python but for All Users instead of just the current user.

amos1969, 2022-01-19:
Oops. Missed the second post @carlosperate 

Have copied the command and output to a file.

[debugging.txt](https://github.com/mu-editor/mu/files/7898218/debugging.txt)