mu-editor/mu #1997: Crash Report

LukeASmith, 2022-01-21:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0b7
Locale: en_GB
Platform: Windows 10 10.0.18362 AMD64

```
"C:\Program Files (x86)\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "C:\Program Files (x86)\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 772, in create
    self.install_baseline_packages()
  File "C:\Program Files (x86)\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 885, in install_baseline_packages
    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "C:\Program Files (x86)\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 860, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "C:\Program Files (x86)\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 270, in install
    return self.run(
  File "C:\Program Files (x86)\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "C:\Program Files (x86)\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "C:\Program Files (x86)\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 177, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process did not terminate normally:

``
[mu LOG.log](https://github.com/mu-editor/mu/files/7912717/mu.LOG.log)
`

carlosperate, 2022-01-21:
Hi @LukeASmith,

Thanks for the report and the log!

It looks like one of the steps needed for setting up Mu is timing out. Is it possible something might be blocking writes to disk on these accounts? Maybe something like an antivirus with real time protection?

If so, would it be possible to disable it and try again? To see if that is the culprit?

LukeASmith, 2022-01-24:
Hi @carlosperate ,

Thanks for the quick reply.
I have passed this information onto our Network Manager. We will let you know if we encounter any issues.

Many thanks, Luke