mu-editor/mu #2000: UnicodeDecodeError: 'gbk' codec can't decode byte 0x80 in position 64: illegal multibyte sequence

shaohaiyang, 2022-01-22:
### What were you trying to do?

bug fixed

### What steps did you take to trigger the issue?

renamed filename with unicode or gbk.

### What did you expect to happen?

open filename with encoding utf-8

### What actually happened?

'''
    with open(path,encoding="utf-8") as f:
'''

### Operating System Version

windows10

### Mu Version

1.1.0b7

### Other Info

    with open(path,encoding="utf-8") as f:

### Editor Log

  File "C:\Users\Administrator\AppData\Local\python\mu\mu_venv-38-20220121-141938\lib\site-packages\pgzero\runner.py", line 77, in main
    src = f.read()
UnicodeDecodeError: 'gbk' codec can't decode byte 0x80 in position 64: illegal multibyte sequence

carlosperate, 2022-01-24:
Hi @shaohaiyang,

```
File "C:\Users\Administrator\AppData\Local\python\mu\mu_venv-38-20220121-141938\lib\site-packages\pgzero\runner.py", line 77, in main
src = f.read()
UnicodeDecodeError: 'gbk' codec can't decode byte 0x80 in position 64: illegal multibyte sequence
```

Where did you get that error? and what steps did you take to trigger it?
Could you attach a copy of the file?
If the file you are trying to run is not 'utf-8' encoded you need to specify the right encoding as a shebang: https://stackoverflow.com/a/729016

carlosperate, 2022-05-31:
Hi @shaohaiyang, were you able to resolve this issue? Was it the code file encoding?

ldu2, 2024-01-26:
I got a similar error as @shaohaiyang shared. Here's how I got the error. I have a python script simply prints out some text. But instead of printing it out in the windows prompt. I redirect rhe output to a .txt file `python fetch_online_text.py > sample.txt`. Then when I try to read the .txt file with python again. I get that error.

carlosperate, 2024-01-27:
I would guess that the encoding of the `sample.txt` file depends on your terminal, so the first step would be to figure out your terminal text encoding, and then on your python script when trying to open the `sample.txt` file you can pass the encoding type using the `encoding` arguments in the `open` function:
https://docs.python.org/3/library/functions.html#open

ldu2, 2024-01-27:
> I would guess that the encoding of the `sample.txt` file depends on your terminal, so the first step would be to figure out your terminal text encoding, and then on your python script when trying to [...]

You are absolutely right! I tried the steps with a different terminal and the encoding shows different in the editor. Thank you very much!