mu-editor/mu #2008: Inconsistency between UI and status bar language

CSharperMantle, 2022-01-26:
### What were you trying to do?

Switching language, expecting the global display language of Mu to change.

### What steps did you take to trigger the issue?

1. Start Mu Editor. Then
2. Go to "Mu Configuration" on the bottom-right corner. Then
3. Go to "Select Language" tab and choose a language different from system language. Then
4. Press "OK" and restart the app. Then
5. Observe that the UI language (tags for buttons, etc.) has changed but the status bar still remains the text in the original language. At the same time,
6. The hover-popup label for "Mu Configuration" is in the wrong language.

### What did you expect to happen?

Language should be changed globally and consistently.

### What actually happened?

The UI language (tags for buttons, etc.) can be changed via the "Select Language" tab but the status bar is still left unchanged. So do some popup labels.

Proof:

[image]

Note the difference of the language between the text editor, the status bar and the buttons.

### Operating System Version

Microsoft Windows 10, Version 1511

### Mu Version

Mu 1.1.0b7

### Other Info

Currently none. :)

### Editor Log

2022-01-26 18:29:32,014 - root:269(run) INFO: 

-----------------

Starting Mu 1.1.0b7
2022-01-26 18:29:32,014 - root:270(run) INFO: uname_result(system='Windows', node='[PC NAME REDACTED]', release='10', version='10.0.10586', machine='AMD64', processor='Intel64 Family 6 Model 69 Stepping 1, GenuineIntel')
2022-01-26 18:29:32,061 - root:271(run) INFO: Platform: Windows-10-10.0.10586-SP0
2022-01-26 18:29:32,061 - root:272(run) INFO: Python path: ['D:\\Documents\\Downloads', 'C:\\Users\\[USERNAME REDACTED]]\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\[USERNAME REDACTED]]\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\[USERNAME REDACTED]]\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\[USERNAME REDACTED]]\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\[USERNAME REDACTED]]\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\[USERNAME REDACTED]]\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-57.4.0-py3.8.egg', 'C:\\Users\\[USERNAME REDACTED]]\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-21.1.3-py3.8.egg', 'C:\\Users\\[USERNAME REDACTED]]\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\[USERNAME REDACTED]]\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\[USERNAME REDACTED]]\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\[USERNAME REDACTED]]\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2022-01-26 18:29:32,123 - root:273(run) INFO: Language code: zh_CN
2022-01-26 18:29:32,123 - mu.settings:220(load) WARNING: No settings file found at C:\Users\[USERNAME REDACTED]]\AppData\Local\python\mu\settings.json; skipping
2022-01-26 18:29:32,123 - mu.settings:220(load) WARNING: No settings file found at C:\Users\[USERNAME REDACTED]]\AppData\Local\python\mu\session.json; skipping
2022-01-26 18:29:43,139 - mu.virtual_environment:585(ensure_and_create) INFO: Added log handler.
2022-01-26 18:30:02,843 - mu.virtual_environment:596(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2022-01-26 18:30:02,843 - mu.virtual_environment:781(create_venv) INFO: Creating virtualenv: C:\Users\[USERNAME REDACTED]]\AppData\Local\python\mu\mu_venv-38-20220126-182846
2022-01-26 18:30:02,856 - mu.virtual_environment:782(create_venv) INFO: Virtualenv name: mu_venv-38-20220126-182846
2022-01-26 18:30:02,860 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\[USERNAME REDACTED]]\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-I', '-m', 'virtualenv', '-p', 'C:\\Users\\[USERNAME REDACTED]]\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-q', 'C:\\Users\\[USERNAME REDACTED]]\\AppData\\Local\\python\\mu\\mu_venv-38-20220126-182846') with kwargs {'env': {'ALLUSERSPROFILE': 'C:\\ProgramData', 'APPDATA': 'C:\\Users\\[USERNAME REDACTED]]\\AppData\\Roaming', 'COMMONPROGRAMFILES': 'C:\\Program Files\\Common Files', 'COMMONPROGRAMFILES(X86)': 'C:\\Program Files (x86)\\Common Files', 'COMMONPROGRAMW6432': 'C:\\Program Files\\Common Files', 'COMPUTERNAME': 'PC-201401181607', 'COMSPEC': 'C:\\WINDOWS\\system32\\cmd.exe', 'DEVMGR_SHOW_DETAILS': '1', 'FPS_BROWSER_APP_PROFILE_STRING': 'Internet Explorer', 'FPS_BROWSER_USER_PROFILE_STRING': 'Default', 'FP_NO_HOST_CHECK': 'NO', 'HOMEDRIVE': 'C:', 'HOMEPATH': '\\Users\\Administrator', 'IOJS_ORG_MIRROR': 'https://npm.taobao.org/mirrors/iojs/', 'LOCALAPPDATA': 'C:\\Users\\[USERNAME REDACTED]]\\AppData\\Local', 'LOGONSERVER': '\\\\MicrosoftAccount', 'NODEJS_ORG_MIRROR': 'https://npm.taobao.org/mirrors/node/', 'NUMBER_OF_PROCESSORS': '4', 'OS': 'Windows_NT', 'PATH': 'C:\\Users\\[USERNAME REDACTED]]\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\PyQt5\\Qt\\bin;C:\\Program Files (x86)\\Common Files\\Oracle\\Java\\javapath;C:\\WINDOWS\\system32;C:\\WINDOWS;C:\\WINDOWS\\System32\\wbem;C:\\WINDOWS\\System32\\WindowsPowerShell\\v1.0\\;E:\\Programs\\Python\\Python37\\Scripts;F:\\Workspace\\opencv-4.1.2\\build\\install\\x86\\mingw\\bin;F:\\Workspace\\opencv-4.1.2\\build\\install\\x86\\mingw\\lib;C:\\Program Files (x86)\\GnuPG\\bin;C:\\Program Files\\PuTTY\\;E:\\Git\\cmd;C:\\Program Files\\dotnet\\;D:\\Polyspace\\R2021a\\runtime\\win64;D:\\Polyspace\\R2021a\\bin;D:\\Polyspace\\R2021a\\polyspace\\bin;C:\\Program Files\\PowerShell\\7\\;C:\\Program Files\\nodejs\\;C:\\Users\\[USERNAME REDACTED]]\\.cargo\\bin;E:\\rust\\.cargo\\bin;E:\\Programs\\Python\\Python37\\;E:\\Microsoft VS Code\\bin;P:\\x86_64-8.1.0-release-posix-seh-rt_v6-rev0\\mingw64\\bin;F:\\CMake\\bin;C:\\Program Files (x86)\\Nmap;C:\\Program Files\\Bandizip\\;F:\\NetHack-3.6.6_Released.x86;F:\\Workspace\\opencv-4.1.2\\build\\install;C:\\Users\\[USERNAME REDACTED]]\\.dotnet\\tools;P:\\geckodriver-v0.29.0-win64;C:\\Users\\[USERNAME REDACTED]]\\AppData\\Roaming\\npm;D:\\Firefox Developer Edition;C:\\Program Files (x86)\\LLVM\\bin;C:\\Program Files (x86)\\LLVM\\lib\\cmake\\llvm;C:\\Users\\[USERNAME REDACTED]]\\AppData\\Local\\Yarn\\bin;P:\\curl-7.68.0-win32-mingw\\bin;P:\\LLVM\\bin;P:\\msixmgr\\;P:\\doxygen\\bin\\', 'PATHEXT': '.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC', 'POWERSHELL_DISTRIBUTION_CHANNEL': 'MSI:Windows 10 Pro', 'PROCESSOR_ARCHITECTURE': 'AMD64', 'PROCESSOR_IDENTIFIER': 'Intel64 Family 6 Model 69 Stepping 1, GenuineIntel', 'PROCESSOR_LEVEL': '6', 'PROCESSOR_REVISION': '4501', 'PROGRAMDATA': 'C:\\ProgramData', 'PROGRAMFILES': 'C:\\Program Files', 'PROGRAMFILES(X86)': 'C:\\Program Files (x86)', 'PROGRAMW6432': 'C:\\Program Files', 'PROMPT': '$P$G', 'PSMODULEPATH': 'C:\\WINDOWS\\system32\\WindowsPowerShell\\v1.0\\Modules\\', 'PUBLIC': 'C:\\Users\\Public', 'RUSTUP_DIST_SERVER': 'https://opentuna.cn/rustup', 'SESSIONNAME': 'Console', 'SYSTEMDRIVE': 'C:', 'SYSTEMROOT': 'C:\\WINDOWS', 'TEMP': 'C:\\Users\\[USERNAME REDACTED]]\\AppData\\Local\\Temp', 'TMP': 'C:\\Users\\[USERNAME REDACTED]]\\AppData\\Local\\Temp', 'USERDOMAIN': 'PC-201401181607', 'USERDOMAIN_ROAMINGPROFILE': 'PC-201401181607', 'USERNAME': 'Administrator', 'USERPROFILE': 'C:\\Users\\Administrator', 'VBOX_MSI_INSTALL_PATH': 'P:\\Oracle\\VirtualBox\\', 'WINDIR': 'C:\\WINDOWS', 'WINDOWS_TRACING_FLAGS': '3', 'QT_API': 'pyqt5', 'QT_AUTO_SCREEN_SCALE_FACTOR': '1', 'QT_MAC_WANTS_LAYER': '1'}}
2022-01-26 18:30:37,103 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2022-01-26 18:30:37,103 - mu.virtual_environment:797(create_venv) INFO: Created virtual environment using C:\Users\[USERNAME REDACTED]]\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\[USERNAME REDACTED]]\AppData\Local\python\mu\mu_venv-38-20220126-182846
2022-01-26 18:30:37,108 - mu.virtual_environment:873(install_baseline_packages) INFO: Installing baseline packages.
2022-01-26 18:30:37,109 - mu.virtual_environment:880(install_baseline_packages) INFO: Expecting zipped wheels at C:\Users\[USERNAME REDACTED]]\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\wheels\1.1.0b7.zip
2022-01-26 18:30:43,388 - mu.virtual_environment:855(install_from_zipped_wheels) INFO: About to install from wheel: backcall-0.2.0-py2.py3-none-any.whl
[... cut ...]
pickleshare       0.7.5
pip               21.3.1
prompt-toolkit    3.0.24
pycparser         2.21
pygame            2.1.0
Pygments          2.10.0
pyserial          3.5
python-dateutil   2.8.2
pywin32           303
pyzmq             22.3.0
reedsolo          1.5.4
setuptools        60.0.3
six               1.16.0
tornado           6.1
traitlets         5.1.1
wcwidth           0.2.5
Werkzeug          2.0.2
wheel             0.37.0
2022-01-26 18:56:40,874 - mu.virtual_environment:931(installed_packages) INFO: []

CSharperMantle, 2022-01-26:
Please let me know if I can do any help with the issue. I have intermediate familiarity with Python and some other languages. :wink:

xbecas, 2022-01-30:
Same situation, but starting from Portuguese.
[image]

CSharperMantle, 2022-01-31:
PR filed as a solution.

xbecas, 2022-03-05:
Related to #1849