mu-editor/mu #2012: Crash Report

koussayghanmy, 2022-01-28:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0b7
Locale: en_US
Platform: Windows 10 10.0.22000 AMD64

```
alEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\KOUSSAY\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\KOUSSAY\AppData\Local\python\mu\mu_venv-38-20220128-121534
RuntimeError: failed to build image pip because:
Traceback (most recent call last):
  File "C:\Users\KOUSSAY\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\via_app_data.py", line 57, in _install
    installer.install(creator.interpreter.version_info)
  File "C:\Users\KOUSSAY\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\pip_install\base.py", line 35, in install
    self._uninstall_previous_version()
  File "C:\Users\KOUSSAY\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\pip_install\base.py", line 151, in _uninstall_previous_version
    dist_name = self._dist_info.stem.split("-")[0]
  File "C:\Users\KOUSSAY\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\pip_install\base.py", line 115, in _dist_info
    raise RuntimeError(msg)  # pragma: no cover
RuntimeError: no .dist-info at C:\Users\KOUSSAY\AppData\Local\pypa\virtualenv\wheel\3.8\image\1\CopyPipInstall\pip-21.3.1-py3-none-any, has pip

```2022-01-28 12:13:20,523 - root:269(run) INFO: 

-----------------

Starting Mu 1.1.0b7
2022-01-28 12:13:20,524 - root:270(run) INFO: uname_result(system='Windows', node='OVERLORD', release='10', version='10.0.22000', machine='AMD64', processor='Intel64 Family 6 Model 78 Stepping 3, GenuineIntel')
2022-01-28 12:13:20,691 - root:271(run) INFO: Platform: Windows-10-10.0.22000-SP0
2022-01-28 12:13:20,691 - root:272(run) INFO: Python path: ['C:\\Users\\KOUSSAY\\Downloads', 'C:\\Users\\KOUSSAY\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\KOUSSAY\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\KOUSSAY\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\KOUSSAY\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\KOUSSAY\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\KOUSSAY\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-57.4.0-py3.8.egg', 'C:\\Users\\KOUSSAY\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-21.1.3-py3.8.egg', 'C:\\Users\\KOUSSAY\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\KOUSSAY\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\KOUSSAY\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin', 'C:\\Users\\KOUSSAY\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\IPython\\extensions']
2022-01-28 12:13:20,828 - root:273(run) INFO: Language code: en_US
2022-01-28 12:13:20,833 - mu.settings:220(load) WARNING: No settings file found at C:\Users\KOUSSAY\AppData\Local\python\mu\settings.json; skipping
2022-01-28 12:13:20,837 - mu.settings:220(load) WARNING: No settings file found at C:\Users\KOUSSAY\AppData\Local\python\mu\session.json; skipping
2022-01-28 12:13:26,133 - mu.virtual_environment:585(ensure_and_create) INFO: Added log handler.
2022-01-28 12:13:29,578 - mu.virtual_environment:596(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2022-01-28 12:13:29,578 - mu.virtual_environment:781(create_venv) INFO: Creating virtualenv: C:\Users\KOUSSAY\AppData\Local\python\mu\mu_venv-38-20220128-121240
2022-01-28 12:13:29,643 - mu.virtual_environment:782(create_venv) INFO: Virtualenv name: mu_venv-38-20220128-121240
2022-01-28 12:13:29,644 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\KOUSSAY\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-I', '-m', 'virtualenv', '-p', 'C:\\Users\\KOUSSAY\\AppData\\Local\\Programs\\Mu Editor\\Python\\pythonw.exe', '-q', 'C:\\Users\\KOUSSAY\\AppData\\Local\\python\\mu\\mu_venv-38-20220128-121240') with kwargs {'env': {'ALLUSERSPROFILE': 'C:\\ProgramData', 'APPDATA': 'C:\\Users\\KOUSSAY\\AppData\\Roaming', 'COMMONPROGRAMFILES': 'C:\\Program Files\\Common Files', 'COMMONPROGRAMFILES(X86)': 'C:\\Program Files (x86)\\Common Files', 'COMMONPROGRAMW6432': 'C:\\Program Files\\Common Files', 'COMPUTERNAME': 'OVERLORD', 'COMSPEC': 'C:\\Windows\\system32\\cmd.exe', 'DRIVERDATA': 'C:\\Windows\\System32\\Drivers\\DriverData', 'HOMEDRIVE': 'C:', 'HOMEPATH': '\\Users\\KOUSSAY', 'LOCALAPPDATA': 'C:\\Users\\KOUSSAY\\AppData\\Local', 'LOGONSERVER': '\\\\OVERLORD', 'NUMBER_OF_PROCESSORS': '4', 'ONEDRIVE': 'C:\\Users\\KOUSSAY\\OneDrive', 'ONEDRIVECONSUMER': 'C:\\Users\\KOUSSAY\\OneDrive', 'OS': 'Windows_NT', 'PATH': 'C:\\Users\\KOUSSAY\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\PyQt5\\Qt\\bin;C:\\Program Files (x86)\\NVIDIA Corporation\\PhysX\\Common;C:\\Windows\\system32;C:\\Windows;C:\\Windows\\System32\\Wbem;C:\\Windows\\System32\\WindowsPowerShell\\v1.0\\;C:\\Windows\\System32\\OpenSSH\\;C:\\Users\\KOUSSAY\\AppData\\Local\\Programs\\Python\\Python310\\Scripts\\;C:\\Users\\KOUSSAY\\AppData\\Local\\Programs\\Python\\Python310\\;C:\\Users\\KOUSSAY\\AppData\\Local\\Microsoft\\WindowsApps;', 'PATHEXT': '.COM;.EXE;.BAT;.CMD;.VBS;.VBE;.JS;.JSE;.WSF;.WSH;.MSC', 'PROCESSOR_ARCHITECTURE': 'AMD64', 'PROCESSOR_IDENTIFIER': 'Intel64 Family 6 Model 78 Stepping 3, GenuineIntel', 'PROCESSOR_LEVEL': '6', 'PROCESSOR_REVISION': '4e03', 'PROGRAMDATA': 'C:\\ProgramData', 'PROGRAMFILES': 'C:\\Program Files', 'PROGRAMFILES(X86)': 'C:\\Program Files (x86)', 'PROGRAMW6432': 'C:\\Program Files', 'PSMODULEPATH': 'C:\\Program Files\\WindowsPowerShell\\Modules;C:\\Windows\\system32\\WindowsPowerShell\\v1.0\\Modules', 'PUBLIC': 'C:\\Users\\Public', 'SESSIONNAME': 'Console', 'SYSTEMDRIVE': 'C:', 'SYSTEMROOT': 'C:\\Windows', 'TEMP': 'C:\\Users\\KOUSSAY\\AppData\\Local\\Temp', 'TMP': 'C:\\Users\\KOUSSAY\\AppData\\Local\\Temp', 'USERDOMAIN': 'OVERLORD', 'USERDOMAIN_ROAMINGPROFILE': 'OVERLORD', 'USERNAME': 'KOUSSAY', 'USERPROFILE': 'C:\\Users\\KOUSSAY', 'WINDIR': 'C:\\Windows', 'QT_API': 'pyqt5', 'QT_AUTO_SCREEN_SCALE_FACTOR': '1', 'QT_MAC_WANTS_LAYER': '1'}}
2022-01-28 12:13:43,401 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 1; output: RuntimeError: failed to build image wheel, setuptools, pip because:
Traceback (most recent call last):
  File "C:\Users\KOUSSAY\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\via_app_data.py", line 56, in _install
    installer.build_image()
  File "C:\Users\KOUSSAY\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\pip_install\base.py", line 52, in build_image
    zip_ref.extractall(str(self._image_dir))
  File "C:\Users\KOUSSAY\AppData\Local\Programs\Mu Editor\Python\lib\zipfile.py", line 1647, in extractall
    self._extract_member(zipinfo, path, pwd)
  File "C:\Users\KOUSSAY\AppData\Local\Programs\Mu Editor\Python\lib\zipfile.py", line 1701, in _extract_member
    open(targetpath, "wb") as target:
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\KOUSSAY\\AppData\\Local\\pypa\\virtualenv\\wheel\\3.8\\image\\1\\CopyPipInstall\\wheel-0.37.0-py2.py3-none-any\\wheel-0.37.0.dist-info\\LICENSE.txt'
Traceback (most recent call last):
  File "C:\Users\KOUSSAY\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\via_app_data.py", line 56, in _install
    installer.build_image()
  File "C:\Users\KOUSSAY\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\pip_install\base.py", line 52, in build_image
    zip_ref.extractall(str(self._image_dir))
  File "C:\Users\KOUSSAY\AppData\Local\Programs\Mu Editor\Python\lib\zipfile.py", line 1647, in extractall
    self._extract_member(zipinfo, path, pwd)
  File "C:\Users\KOUSSAY\AppData\Local\Programs\Mu Editor\Python\lib\zipfile.py", line 1701, in _extract_member
    open(targetpath, "wb") as target:
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\KOUSSAY\\AppData\\Local\\pypa\\virtualenv\\wheel\\3.8\\image\\1\\CopyPipInstall\\setuptools-58.3.0-py3-none-any\\setuptools-58.3.0.dist-info\\entry_points.txt'
Traceback (most recent call last):
[... cut ...]
  File "C:\Users\KOUSSAY\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\virtualenv\seed\embed\via_app_data\pip_install\base.py", line 115, in _dist_info
    raise RuntimeError(msg)  # pragma: no cover
RuntimeError: no .dist-info at C:\Users\KOUSSAY\AppData\Local\pypa\virtualenv\wheel\3.8\image\1\CopyPipInstall\pip-21.3.1-py3-none-any, has pip
2022-01-28 12:15:42,926 - mu.settings:169(save) DEBUG: Saving to C:\Users\KOUSSAY\AppData\Local\python\mu\session.json
2022-01-28 12:15:42,927 - mu.settings:169(save) DEBUG: Saving to C:\Users\KOUSSAY\AppData\Local\python\mu\venv.json

carlosperate, 2022-01-28:
Thank you for the report and the log @koussayghanmy !
Could you run this in a command prompt and paste the output of the command here?

```
C:\Users\KOUSSAY\AppData\Local\Programs\Mu Editor\Python\pythonw.exe -m pip list --verbose
```

carlosperate, 2022-01-28:
If you are reading this via email I've made a small mistake in the command I've sent in my previous response, and I've updated the original message to correct it (updates don't appear on Github emails).

The right command is:
```
C:\Users\KOUSSAY\AppData\Local\Programs\Mu Editor\Python\python.exe -m pip list --verbose
```

carlosperate, 2022-01-28:
If you are reading this via email I've made a small mistake in the command I've sent in my previous response, and I've updated the original message to correct it (updates don't appear on Github emails).

The right command is:
```
"C:\Users\KOUSSAY\AppData\Local\Programs\Mu Editor\Python\python.exe" -m pip list --verbose
```