mu-editor/mu #2020: Crash Report

Salmeron52, 2022-02-01:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0b7
Locale: es_ES
Platform: Windows 10 10.0.22000 AMD64

```
Traceback (most recent call last):
  File "C:\Users\manue\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\manue\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\manue\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\manue\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\manue\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 349, in run
    editor.setup(setup_modes(editor, editor_window))
  File "C:\Users\manue\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 809, in setup
    os.makedirs(wd)
  File "C:\Users\manue\AppData\Local\Programs\Mu Editor\Python\lib\os.py", line 223, in makedirs
    mkdir(name, mode)
FileExistsError: [WinError 183] No se puede crear un archivo que ya existe: 'C:\\Users\\manue\\mu_code'

```

carlosperate, 2022-02-18:
Thanks for the report @Salmeron52!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\manue\AppData\Local\python\mu\Logs\mu.log`

I suspect maybe 2 instances of Mu are running in parallel and racing each other at creating these directories.

If you try to run Mu again, and ensure it is only running once, do you still get this problem?

carlosperate, 2022-05-31:
@Salmeron52 did opening Mu again work or are you still experiencing this issue?

Funcion Mu al abrirlo de nuevo, o todavia tienes este problema?