mu-editor/mu #2022: Crash Report

rivale, 2022-02-02:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0b7
Locale: en_GB
Platform: Linux 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/mu/app.py", line 159, in run
    raise ex
  File "/usr/local/lib/python3.7/dist-packages/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/usr/local/lib/python3.7/dist-packages/mu/virtual_environment.py", line 621, in ensure_and_create
    self.recreate()
  File "/usr/local/lib/python3.7/dist-packages/mu/virtual_environment.py", line 566, in recreate
    self.create()
  File "/usr/local/lib/python3.7/dist-packages/mu/virtual_environment.py", line 772, in create
    self.install_baseline_packages()
  File "/usr/local/lib/python3.7/dist-packages/mu/virtual_environment.py", line 883, in install_baseline_packages
    wheels.download(zipped_wheels_filepath, logger)
  File "/usr/local/lib/python3.7/dist-packages/mu/wheels/__init__.py", line 176, in download
    zip_wheels(zip_filepath, temp_dirpath, logger)
  File "/usr/local/lib/python3.7/dist-packages/mu/wheels/__init__.py", line 147, in zip_wheels
    with zipfile.ZipFile(zip_filepath, "w") as z:
  File "/usr/lib/python3.7/zipfile.py", line 1204, in __init__
    self.fp = io.open(file, filemode)
PermissionError: [Errno 13] Permission denied: '/usr/local/lib/python3.7/dist-packages/mu/wheels/1.1.0b7.zip'

```

rivale, 2022-02-02:
using buster on raspberry pi 4.

rivale, 2022-02-19:
On Fri, 18 Feb 2022 08:19:04 -0800
Carlos Pereira Atencio ***@***.***> wrote:

> Thanks for the report @rivale!
> 
> If you've install Mu in the system Python then I assume you might
> have needed to use `sudo`, would that be correct? In that case Mu
> will need to be launched with `sudo` as well to create the files it
> needs for the first launch.
> 

hi
thanks for your reply.

Im using Buster os on a pi4
the fault appear at the first run.
Im running from the desktop after installing it with the raspberry pi
reccomended software, i have tried running fro the command line as sudo
enclosed log

and get the same error

thanks

carlosperate, 2022-02-21:
Hi @rivale, thanks for getting back to us.
If you've repied to this GitHub issue via email it doesn't foward the email attachments to GitHub, so we don't have access to the log.

Could you add the file via the GitHub website? -> https://github.com/mu-editor/mu/issues/2022

Thanks!

rivale, 2022-02-21:
Hi 
Sorry never did think it would not forward the attachment.

enclosed here

2022-02-19 11:53:21,646 - root:269(run) INFO: 

-----------------

Starting Mu 1.1.0b7
2022-02-19 11:53:21,656 - root:270(run) INFO: uname_result(system='Linux', node='raspberrypi', release='5.10.63-v7l+', version='#1496 SMP Wed Dec 1 15:58:56 GMT 2021', machine='armv7l', processor='')
2022-02-19 11:53:21,658 - root:271(run) INFO: Platform: Linux-5.10.63-v7l+-armv7l-with-debian-10.11
2022-02-19 11:53:21,659 - root:272(run) INFO: Python path: ['/usr/local/bin', '/usr/lib/python37.zip', '/usr/lib/python3.7', '/usr/lib/python3.7/lib-dynload', '/home/pi/.local/lib/python3.7/site-packages', '/usr/local/lib/python3.7/dist-packages', '/usr/lib/python3/dist-packages', '/usr/lib/python3/dist-packages/IPython/extensions']
2022-02-19 11:53:21,659 - root:273(run) INFO: Language code: en_GB
2022-02-19 11:53:22,635 - mu.virtual_environment:585(ensure_and_create) INFO: Added log handler.
2022-02-19 11:53:22,733 - mu.virtual_environment:596(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2022-02-19 11:53:22,734 - mu.virtual_environment:619(ensure_and_create) WARNING: Venv created by Mu version -no-version-; Current Mu is version 1.1.0b7
2022-02-19 11:53:22,745 - mu.virtual_environment:922(installed_packages) INFO: Discovering installed third party modules in venv.
2022-02-19 11:53:22,762 - mu.virtual_environment:118(run_blocking) INFO: About to run blocking /home/pi/.local/share/mu/mu_venv-37-20220211-090846/bin/pip with args ['list', '--disable-pip-version-check'] and envvars {}
2022-02-19 11:53:25,819 - mu.virtual_environment:168(wait) DEBUG: Finished: True; exitStatus 0; exitCode 0
2022-02-19 11:53:25,820 - mu.virtual_environment:249(run) DEBUG: Process output: Package    Version
---------- -------
pip        21.3.1
setuptools 60.5.0
wheel      0.37.1
2022-02-19 11:53:25,820 - mu.virtual_environment:931(installed_packages) INFO: ['pip', 'setuptools', 'wheel']
2022-02-19 11:53:25,824 - mu.virtual_environment:540(quarantine_venv) INFO: Quarantined /home/pi/.local/share/mu/mu_venv-37-20220211-090846 as /home/pi/.local/share/mu/mu_venv-37-20220211-090846.SUPERSEDED
2022-02-19 11:53:25,842 - mu.virtual_environment:460(relocate) DEBUG: Relocating to /home/pi/.local/share/mu/mu_venv-37-20220219-115325
2022-02-19 11:53:25,842 - mu.virtual_environment:477(relocate) DEBUG: Virtual environment set up mu_venv-37-20220219-115325 at /home/pi/.local/share/mu/mu_venv-37-20220219-115325
2022-02-19 11:53:25,843 - mu.virtual_environment:781(create_venv) INFO: Creating virtualenv: /home/pi/.local/share/mu/mu_venv-37-20220219-115325
2022-02-19 11:53:25,857 - mu.virtual_environment:782(create_venv) INFO: Virtualenv name: mu_venv-37-20220219-115325
2022-02-19 11:53:25,872 - mu.virtual_environment:429(run_subprocess) INFO: Running ('/usr/bin/python3', '-I', '-m', 'virtualenv', '-p', '/usr/bin/python3', '-q', '/home/pi/.local/share/mu/mu_venv-37-20220219-115325') with kwargs {'env': {'USER': 'pi', 'LANGUAGE': 'en_GB', 'XDG_SEAT': 'seat0', 'TEXTDOMAIN': 'Linux-PAM', 'XDG_SESSION_TYPE': 'x11', 'SSH_AGENT_PID': '788', 'HOME': '/home/pi', 'DESKTOP_SESSION': 'LXDE-pi', 'XDG_SEAT_PATH': '/org/freedesktop/DisplayManager/Seat0', 'DBUS_SESSION_BUS_ADDRESS': 'unix:path=/run/user/1000/bus', 'QT_QPA_PLATFORMTHEME': 'qt5ct', 'LOGNAME': 'pi', 'XDG_SESSION_CLASS': 'user', 'XDG_SESSION_ID': '1', 'SAL_USE_VCLPLUGIN': 'gtk3', 'PATH': '/home/pi/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games:/snap/bin', 'XDG_SESSION_PATH': '/org/freedesktop/DisplayManager/Session0', 'XDG_RUNTIME_DIR': '/run/user/1000', 'XDG_MENU_PREFIX': 'lxde-pi-', 'DISPLAY': ':0.0', 'LANG': 'en_GB', 'XDG_SESSION_DESKTOP': 'lightdm-xsession', 'XAUTHORITY': '/home/pi/.Xauthority', 'XDG_CONFIG_HOME': '/home/pi/.config', 'XDG_GREETER_DATA_DIR': '/var/lib/lightdm/data/pi', 'SSH_AUTH_SOCK': '/tmp/ssh-wp0moteuxDBF/agent.756', 'SHELL': '/bin/bash', 'GDMSESSION': 'lightdm-xsession', 'NO_AT_BRIDGE': '1', 'GPG_AGENT_INFO': '/run/user/1000/gnupg/S.gpg-agent:0:1', 'XDG_VTNR': '7', 'LC_ALL': 'en_GB', 'PWD': '/home/pi', 'XDG_DATA_DIRS': '/usr/share/fkms:/usr/local/share:/usr/share/raspi-ui-overrides:/usr/share:/usr/share/gdm:/var/lib/menu-xdg', 'XDG_CURRENT_DESKTOP': 'LXDE', '_LXSESSION_PID': '756', 'XDG_CONFIG_DIRS': '/etc/xdg', 'DESKTOP_STARTUP_ID': 'lxpanel-835-raspberrypi-mu-editor-2_TIME387102', 'QT_API': 'pyqt5', 'QT_AUTO_SCREEN_SCALE_FACTOR': '1', 'QT_MAC_WANTS_LAYER': '1'}}
2022-02-19 11:53:28,489 - mu.virtual_environment:444(run_subprocess) DEBUG: Process returned 0; output: 
2022-02-19 11:53:28,490 - mu.virtual_environment:800(create_venv) INFO: Created virtual environment using /usr/bin/python3 at /home/pi/.local/share/mu/mu_venv-37-20220219-115325
2022-02-19 11:53:28,512 - mu.virtual_environment:873(install_baseline_packages) INFO: Installing baseline packages.
2022-02-19 11:53:28,512 - mu.virtual_environment:880(install_baseline_packages) INFO: Expecting zipped wheels at /usr/local/lib/python3.7/dist-packages/mu/wheels/1.1.0b7.zip
2022-02-19 11:53:28,523 - mu.virtual_environment:882(install_baseline_packages) WARNING: No zipped wheels found; downloading...
2022-02-19 11:53:28,524 - mu.virtual_environment:165(download) INFO: Downloading wheels to /usr/local/lib/python3.7/dist-packages/mu/wheels/1.1.0b7.zip
[... cut ...]
  File "/usr/local/lib/python3.7/dist-packages/mu/wheels/__init__.py", line 147, in zip_wheels
    with zipfile.ZipFile(zip_filepath, "w") as z:
  File "/usr/lib/python3.7/zipfile.py", line 1204, in __init__
    self.fp = io.open(file, filemode)
PermissionError: [Errno 13] Permission denied: '/usr/local/lib/python3.7/dist-packages/mu/wheels/1.1.0b7.zip'
2022-02-19 12:03:06,057 - mu.settings:169(save) DEBUG: Saving to /home/pi/.local/share/mu/session.json
2022-02-19 12:03:06,059 - mu.settings:169(save) DEBUG: Saving to /home/pi/.local/share/mu/venv.json

carlosperate, 2022-02-23:
Hi @rivale,

Are you sure mu was launched with `sudo` when that log was generated? The main error is not having permissions to right a file into the distribution site packages.

To be fair, it's not recommended to install Mu in the system Python, as that can affect packages used by other applications or the OS.

Could you try uninstalling mu using pip and then installing with the `--user` flag? `pip install mu-editor --user`.

rivale, 2022-02-23:
Hi 
 Thanks for your reply.

I dont know what is going on, but i have installed bulleye os and mu is installing and working properly v1.03, but it does not seem to work on v2 of the microbit.

I think my version on of buster os  maybe corrupted, will try to reload it and let you know.

Thanks for your help.

 #Valerio

closed 2022-02-23 by rivale