mu-editor/mu #2041: Crash Report

papayapeach, 2022-02-17:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0b7
Locale: en_US
Platform: Windows 10 10.0.22000 AMD64

```
e_and_create
    self.create()
  File "C:\Users\pamac\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 772, in create
    self.install_baseline_packages()
  File "C:\Users\pamac\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 885, in install_baseline_packages
    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "C:\Users\pamac\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 860, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "C:\Users\pamac\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 270, in install
    return self.run(
  File "C:\Users\pamac\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "C:\Users\pamac\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "C:\Users\pamac\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 177, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process did not terminate normally:

```

carlosperate, 2022-02-18:
Thanks for the report @papayapeach!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\pamac\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2022-06-06:
Hi @papayapeach, are you still experiencing this issue? Would you be able to try with the latest Mu release and attach the log here to help us debug it? It shoul dbe located in: `C:\Users\pamac\AppData\Local\python\mu\Logs\mu.log`