mu-editor/mu #2042: Crash Report

josh09foord, 2022-02-17:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0b7
Locale: en_US
Platform: Windows 10 10.0.22000 AMD64

```
le "C:\Users\Josh0\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 339, in installed
    next(iterlines)
StopIteration

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Josh0\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 159, in run
    raise ex
  File "C:\Users\Josh0\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Josh0\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "C:\Users\Josh0\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 773, in create
    self.register_baseline_packages()
  File "C:\Users\Josh0\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 892, in register_baseline_packages
    packages = list(self.pip.installed())
  File "C:\Users\Josh0\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 345, in installed
    raise VirtualEnvironmentError("Unable to parse installed packages")
mu.virtual_environment.VirtualEnvironmentError: Unable to parse installed packages

```C:\Users\Josh0\AppData\Local\python\mu\Logs\mu.log

carlosperate, 2022-02-18:
Thanks for the report @josh09foord !
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\Josh0\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2022-05-31:
Hi @josh09foord
Could you try to add your log again to this GitHub issue? file should be in `C:\Users\Josh0\AppData\Local\python\mu\Logs\mu.log`