mu-editor/mu #2044: Crash Report

AlistairSYm, 2022-02-18:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0b7
Locale: en_GB
Platform: Windows 10 10.0.19043 AMD64

```
\virtual_environment.py", line 772, in create
    self.install_baseline_packages()
  File "C:\Users\Alistair\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 885, in install_baseline_packages
    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "C:\Users\Alistair\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 860, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "C:\Users\Alistair\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 270, in install
    return self.run(
  File "C:\Users\Alistair\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "C:\Users\Alistair\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "C:\Users\Alistair\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 177, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process did not terminate normally:
Processing c:\users\alistair\appdata\local\temp\tmpt1w88qm6\asttokens-2.0.5-py2.py3-none-any.whl

```

carlosperate, 2022-02-18:
Thanks for the report @AlistairSYm!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\Alistair\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2022-05-31:
Hi @AlistairSYm, would you be able to attach the log file to this GitHub issue? File should be located in `C:\Users\Alistair\AppData\Local\python\mu\Logs\mu.log`.