mu-editor/mu #2046: Crash Report

lairdm, 2022-02-18:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0b7
Locale: en_CA
Platform: Linux 5.4.0-96-generic #109-Ubuntu SMP Wed Jan 12 16:49:16 UTC 2022 x86_64

```
 File "/home/lairdm/pyportal/ENV/lib/python3.8/site-packages/mu/app.py", line 159, in run
    raise ex
  File "/home/lairdm/pyportal/ENV/lib/python3.8/site-packages/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/lairdm/pyportal/ENV/lib/python3.8/site-packages/mu/virtual_environment.py", line 627, in ensure_and_create
    self.ensure()
  File "/home/lairdm/pyportal/ENV/lib/python3.8/site-packages/mu/virtual_environment.py", line 667, in ensure
    self.ensure_key_modules()
  File "/home/lairdm/pyportal/ENV/lib/python3.8/site-packages/mu/virtual_environment.py", line 750, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: flask
STDERR: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/lairdm/.local/share/mu/mu_venv-38-20220218-154557/lib/python3.8/site-packages/flask/__init__.py", line 19, in <module>
    from . import json
  File "/home/lairdm/.local/share/mu/mu_venv-38-20220218-154557/lib/python3.8/site-packages/flask/json/__init__.py", line 15, in <module>
    from itsdangerous import json as _json
ImportError: cannot import name 'json' from 'itsdangerous' (/home/lairdm/.local/share/mu/mu_venv-38-20220218-154557/lib/python3.8/site-packages/itsdangerous/__init__.py)

```

lairdm, 2022-02-18:
Env: Python 3.8.10
Running in a venv

lairdm, 2022-02-19:
Ok, this thread seems to indicate there's a breaking incompability between Flask and itsdangerous. Since mu-editor appears to build it's own venv I don't think I can try and force this through my local venv very easily. It seems the options are:

Either update Flask to 1.1.4, or fix itsdangerous version to 2.0.1

https://serverfault.com/questions/1094062/from-itsdangerous-import-json-as-json-importerror-cannot-import-name-json-fr

carlosperate, 2022-02-21:
Thanks @lairdm for the report and the analysic, between the two options it's probably best to update Flask, so I've created PR https://github.com/mu-editor/mu/pull/2051.

Until we can release a new version to PyPI you can probably apply the patch manually and then delete the `mu/wheels/1.1.0b7.zip` to force Mu to download the venv dependencies again.

lairdm, 2022-02-21:
Thanks for the prompt fix!

carlosperate, 2022-02-22:
A new version with this fix has been released to PyPI 🎉 https://pypi.org/project/mu-editor/1.1.1/

You should be install it via pip and it should now work :)

Thanks for the report and all the debugging you've done!

closed 2022-02-22 by carlosperate