mu-editor/mu #2047: Crash Report

Naveenkilaru, 2022-02-19:
[Uploading mu.log…]()
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0b7
Locale: en_US
Platform: Windows 10 10.0.19043 AMD64

```
Traceback (most recent call last):
  File "C:\Users\unic\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\unic\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\unic\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\unic\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\unic\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 349, in run
    editor.setup(setup_modes(editor, editor_window))
  File "C:\Users\unic\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 809, in setup
    os.makedirs(wd)
  File "C:\Users\unic\AppData\Local\Programs\Mu Editor\Python\lib\os.py", line 223, in makedirs
    mkdir(name, mode)
FileExistsError: [WinError 183] Cannot create a file when that file already exists: 'C:\\Users\\unic\\mu_code'

```

Naveenkilaru, 2022-02-19:
Please suggest how to proceed further to install MU

carlosperate, 2022-02-21:
Thanks for the @Naveenkilaru, it looks like you've submitted the issue before the log had time to upload (that's why it says `Uploading mu.log…` at the top).
Would you be able to try again? Thanks!

carlosperate, 2022-05-31:
Hi @Naveenkilaru 

Based on the error I think maybe there were 2 instances of Mu running in parallel and racing each other at creating these directories, which would be this issue: https://github.com/mu-editor/mu/issues/1755

If you try to run Mu again, and ensure it is only running once, do you still get this problem?