mu-editor/mu #2050: Crash Report

getimmer, 2022-02-21:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0b7
Locale: en_US
Platform: Linux 5.4.0-99-generic #112-Ubuntu SMP Thu Feb 3 13:50:55 UTC 2022 x86_64

```
Traceback (most recent call last):
  File "/home/ge/.local/lib/python3.8/site-packages/mu/app.py", line 159, in run
    raise ex
  File "/home/ge/.local/lib/python3.8/site-packages/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/ge/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 627, in ensure_and_create
    self.ensure()
  File "/home/ge/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 667, in ensure
    self.ensure_key_modules()
  File "/home/ge/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 750, in ensure_key_modules
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to import: flask
STDERR: Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/home/ge/.local/share/mu/mu_venv-38-20220221-125051/lib/python3.8/site-packages/flask/__init__.py", line 19, in <module>
    from . import json
  File "/home/ge/.local/share/mu/mu_venv-38-20220221-125051/lib/python3.8/site-packages/flask/json/__init__.py", line 15, in <module>
    from itsdangerous import json as _json
ImportError: cannot import name 'json' from 'itsdangerous' (/home/ge/.local/share/mu/mu_venv-38-20220221-125051/lib/python3.8/site-packages/itsdangerous/__init__.py)

```

carlosperate, 2022-02-21:
Thanks for the report @getimmer, we've identified the issue and created PR https://github.com/mu-editor/mu/pull/2051 to fix it.

Until we release a new version to PyPI you can probably apply the patch manually and then delete the `mu/wheels/1.1.0b7.zip` to force Mu to download the venv dependencies again.

carlosperate, 2022-02-22:
A new version with this fix has been released to PyPI 🎉 https://pypi.org/project/mu-editor/1.1.1/

You can be install it via pip and it should now work :)

closed 2022-02-22 by carlosperate