mu-editor/mu #2060: We need a new movie to celebrate the latest stable release of Mu.

ntoll, 2022-02-22:
### What were you trying to do?

I want to celebrate and acknowledge all the contributions from the many collaborators who have helped make Mu over the years.

### What steps did you take to trigger the issue?

I started writing the Mu editor on a rainy Sunday afternoon (first commit to GitHub the following day) and it kind-of got out of hand. :slightly_smiling_face: :cow: :snake: 

### What did you expect to happen?

Not much... a few bug reports and perhaps a "thank you".

### What actually happened?

* Well over a million separate downloads to date (when I last looked in the autumn 2021 it was around 1.2 million). 
* Installed all over the world, including the in Antarctic(!), top universities in the US, and junior schools in [rural China](https://www.youtube.com/watch?v=w3hQglx5Kj0).
* Hundreds of [beginner workshops](https://www.youtube.com/watch?v=SljLglsL_PI). 
* A wonderful flourishing community of contributors who never cease to amaze me at their generosity of spirit.
* [A book](https://techbookfest.org/event/tbf07/circle/5645284751179776) about using Mu published in Japanese..! 
* A personal sense of accomplishment.
* Something to celebrate (hence this ticket).

Not bad for around 5000 lines of Python. :slightly_smiling_face: 

### Operating System Version

Windows / OSX / Linux

### Mu Version

1.1.1

### Other Info

```
make video
```

:film_projector: 

We also need some funky music... oh, and the logo could do with some love. Best ask @ambv :notes:  and @hawkz :art: 

Should feature these lovely people:

@carlosperate, @tjguk, @tmontes, @dybber, @ZanderBrown, @devdanzin, @xbecas, @ladyada, @inglesp, @askvictor, @keith-packard, @bwshockley, @kyoungmeyer, @NDevox, @stestagg, @hulkco, @grovz, @martinohanlon, @kushaldas, @knowledgejunkie, @KKawase0104, @tibs, @jg00dman, @dhalbert, @microbit-carlos, @ambv, @WayneKeenan, @yonghuming, @gquintana, @tim-mccurrach, @fmorton, @willingc, @JohnGuan, @Rybec, @MinoruInachi, @XECDesign, @espoelstra, @rffontenelle, @malbarbo, @plummet555, @inventhouse, @cefn, @ShawnHymel, @hroncok, @alistair-broomhead, @tdicola, @ajs256, @bigeyex, @microbit-mark, @asb, @doanminhdang, @bletvaska, @cclauss, @lordmauve, @skerr92, @illume, @jezdean, @reneraab, @templed, @wu6692776, @CarlFK, @Craig88, @Tresonic, @romilly, @UnexpectedMaker, @camillem, @asottile, @pelikhan, @tomviner, @wayoda, @ogoletti, @ohiofi, @fk-lx, @fkorling, @voidspace, @amos1969, @LukeSlev, @baggio63446333, @ecasadod, @seantibor, @dfarning, @steamfoundry, @yeyeto2788, @CSharperMantle, @deshipu and @ntoll 

### Editor Log

```
commit 98f4ea0ba1e5fe44a6378bece7fa92db0a50354f
Author: Nicholas H.Tollervey <ntoll@ntoll.org>
Date:   Mon Dec 7 20:05:51 2015 +0000

    First commit

commit 29426bc5f88b16290e8a6337ff1bdeb0f12b4f81
Author: Nicholas H.Tollervey <ntoll@ntoll.org>
Date:   Mon Dec 7 20:22:39 2015 +0000

    Add license and initial core code.

commit 9b6705571107d704d576269ae1723d589c330a16
Author: Nicholas H.Tollervey <ntoll@ntoll.org>
Date:   Tue Dec 8 07:07:13 2015 +0000

    It works! (Sort of)
```

ntoll, 2022-02-22:
cc contributors to Mu who were truncated by Markdown:

@asb, @doanminhdang, @bletvaska, @cclauss, @lordmauve, @skerr92, @illume, @jezdean, @reneraab, @templed, @wu6692776, @CarlFK, @Craig88, @Tresonic, @romilly, @UnexpectedMaker, @camillem, @asottile, @pelikhan, @tomviner, @wayoda, @ogoletti, @ohiofi, @fk-lx, @fkorling, @voidspace, @amos1969, @LukeSlev, @baggio63446333, @ecasadod, @seantibor, @dfarning, @steamfoundry, @yeyeto2788, @CSharperMantle, @deshipu and @ntoll

ntoll, 2022-02-22:
Fixed by: 

https://www.youtube.com/watch?v=nOMCS_8Bm-E 

:cow: :heart: :snake: :tada: 

(and thank you all)

closed 2022-02-22 by ntoll

hroncok, 2022-02-22:
Should it be called *The Making Of Mu 1.1.1* instead of *1.1.0*?

ntoll, 2022-02-22:
Yeah.... late breaking bug put the kibosh on that plan so we ended up just going for 1.1.1. :stuck_out_tongue:

xbecas, 2022-02-22:
**A-M-A-Z-I-N-G   V-I-D-E-O ! ! !**

ambv, 2022-02-22:
Brilliant. Honored to be part of this!

carlosperate, 2022-02-22:
Awesome! Specially love the head bopping Mu logo 😄 🐮

ntoll, 2022-02-22:
The bopping head was all @hawkz doing (thank you Steve).

illume, 2022-02-22:
1️⃣.🎵.🇺
✍️.1️⃣.🎨 
Ⓜ️.🐍.1️⃣

WayneKeenan, 2022-02-22:
This is such a nice thought. I'm happy to have had a little bit part in it.  Nice funky toon too.

Rybec, 2022-02-22:
First, thanks for doing this.  It's awesome.  I really appreciate how this brings attribution that is generally left in the background to the forefront, in a format that is easier for normal people to appreciate.  It's kind of cool to show my kids that my name is in a YouTube video and explain to them what I did to deserve it.  It's a great opportunity to introduce them to the idea of open source software in a more intimate way than just describing how it works.  And the visuals really help.  I can show them the PNG section, and then explain how I created this cluster of SVG elements where my name is hovering around.  I hope this will help motivate them to share their skills and talents in similar ways.

Second, I think this video is a really valuable visualization of how open source projects work.  In companies (especially big ones), the parts of a project tend to be siloed, such that each part has its own team, and people only work on things assigned ot their team.  In this video there are people coming and going, there are people who just work in one area, but there are also people to jump around a lot doing what needs done regardless of where it is.  This visually demonstrates how organic open source software development is, and I think that is both valuable and awesome.

Lastly, just to add my own story to this about how and why I contributed:

I don't actually use Mu much.  I'm a professional programmer, so I have my own preferences (Vim + command line), and Mu has some non-configurable defaults I don't like (auto-tab mainly; I learned to program before this was a common feature, and I've developed what is almost a reflex for tabbing where appriopriate, so auto-tab basically just adds more tabs on top of my own, which is horrible in Python; it's the first feature I disable in any coding editor I use).  That said, I _do_ appreciate the project, and I have a great deal of respect for open source programmers, whether I use their software or not.  So, when I saw that the Mu project needed SVG icons, my internal response was, "I could do this, and I don't think it would even be that hard."  I've been using Inkscape for almost two decades, for making simple graphics, for creating extrudable cross sections for 3D printing, and for drafting 2D designs for laser cutting.  I'm not a graphic designer, but I have enough SVG experience that I figured I could handle it.  And as Mu is a well respected and generally high quality open source project, I felt a desire to contribute.  The rest is in pull request #1603.  I made the icons, worked with some other wonderful people in this community to work out the bugs and get everything right, and ultimately my work became part of Mu.  And I get satisfaction from knowing I contributed to such a value a project!  I hope Mu continues to evolve and improve to better meet the needs of those who _do_ use it!

Anyhow, this is an excellent project, and I'm happy I was able to contribute something to it, even if it is fairly small and narrow.  Thank you for making this video, and I'm honored that you feel my contribution is sufficient to include me.

ntoll, 2022-02-23:
@Rybec,

Thank you so much for your thoughtful and kind reply.

I think you give a great summary of the video and, if you're interested, I use Gource (https://gource.io/) under the hood (see the Makefile for how: https://github.com/mu-editor/mu/blob/master/Makefile#L124).

I'm also really happy to hear that you found Mu such a positive experience. All the core contributors work very hard to ensure folks feel welcome, empowered and that work is appreciated and celebrated. This is important. Several of us have had (how can I put this with candour?), not-so-good experiences of the Python community, and for me personally, Mu is consciously an antidote to those sorts of things.

N.

inglesp, 2022-02-27:
Well that was mesmerising to watch.

Congratulations @ntoll on the vision for the project, and the drive to see it through.

willingc, 2022-02-27:
Way to go @ntoll. Brilliant editor.

ntoll, 2022-02-28:
Thank you @inglesp and @willingc (I hope you're both well).

CSharperMantle, 2022-03-01:
So glad to see my name show up here! 🥳