mu-editor/mu #2072: Crash Report

nickcode55, 2022-02-25:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.1
Locale: en_CY
Platform: Windows 10 10.0.19043 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Alexander\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 194, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "C:\Users\Alexander\AppData\Local\Programs\Mu Editor\Python\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Alexander\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 9, in <module>
    main()
  File "C:\Users\Alexander\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\__main__.py", line 5, in main
    run()
  File "C:\Users\Alexander\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 349, in run
    editor.setup(setup_modes(editor, editor_window))
  File "C:\Users\Alexander\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\logic.py", line 809, in setup
    os.makedirs(wd)
  File "C:\Users\Alexander\AppData\Local\Programs\Mu Editor\Python\lib\os.py", line 223, in makedirs
    mkdir(name, mode)
FileExistsError: [WinError 183] Cannot create a file when that file already exists: 'C:\\Users\\Alexander\\mu_code'

```

carlosperate, 2022-02-28:
Thanks for the report @nickcode55!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\Alexander\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2022-05-31:
Hi @nickcode55,

Based on the error I think maybe there were 2 instances of Mu running in parallel and racing each other at creating these directories, which would be this issue: https://github.com/mu-editor/mu/issues/1755

If you try to run Mu again, and ensure it is only running once, do you still get this problem?