mu-editor/mu #2075: Crash Report

arouldassou, 2022-02-26:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.1
Locale: fr_FR
Platform: Linux 5.10.63-v7+ #1496 SMP Wed Dec 1 15:58:11 GMT 2021 armv7l

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/mu/app.py", line 159, in run
    raise ex
  File "/usr/local/lib/python3.7/dist-packages/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/usr/local/lib/python3.7/dist-packages/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/usr/local/lib/python3.7/dist-packages/mu/virtual_environment.py", line 772, in create
    self.install_baseline_packages()
  File "/usr/local/lib/python3.7/dist-packages/mu/virtual_environment.py", line 883, in install_baseline_packages
    wheels.download(zipped_wheels_filepath, logger)
  File "/usr/local/lib/python3.7/dist-packages/mu/wheels/__init__.py", line 178, in download
    zip_wheels(zip_filepath, temp_dirpath, logger)
  File "/usr/local/lib/python3.7/dist-packages/mu/wheels/__init__.py", line 149, in zip_wheels
    with zipfile.ZipFile(zip_filepath, "w") as z:
  File "/usr/lib/python3.7/zipfile.py", line 1204, in __init__
    self.fp = io.open(file, filemode)
PermissionError: [Errno 13] Permission non accordÃ©e: '/usr/local/lib/python3.7/dist-packages/mu/wheels/1.1.1.zip'

```

arouldassou, 2022-02-26:
Hello
Can you help me.It is very important for my work
Thanks you very much

carlosperate, 2022-02-28:
Thanks for the report @arouldassou.
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `~/.cache/mu/`

carlosperate, 2022-02-28:
Also, is it possible you've installed Mu in the system Python using `sudo`? In that case you might need `sudo` as well to run it.

It's recommended to install Mu in a virtual environment, or to use something like [pipx](https://pypa.github.io/pipx/).