mu-editor/mu #2076: Crash Report

kevingtonbeare, 2022-02-26:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.1
Locale: en_AU
Platform: Linux 5.4.0-100-generic #113-Ubuntu SMP Thu Feb 3 18:43:29 UTC 2022 x86_64

```
Traceback (most recent call last):
  File "/usr/local/lib/python3.8/dist-packages/mu/app.py", line 159, in run
    raise ex
  File "/usr/local/lib/python3.8/dist-packages/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 772, in create
    self.install_baseline_packages()
  File "/usr/local/lib/python3.8/dist-packages/mu/virtual_environment.py", line 883, in install_baseline_packages
    wheels.download(zipped_wheels_filepath, logger)
  File "/usr/local/lib/python3.8/dist-packages/mu/wheels/__init__.py", line 178, in download
    zip_wheels(zip_filepath, temp_dirpath, logger)
  File "/usr/local/lib/python3.8/dist-packages/mu/wheels/__init__.py", line 149, in zip_wheels
    with zipfile.ZipFile(zip_filepath, "w") as z:
  File "/usr/lib/python3.8/zipfile.py", line 1251, in __init__
    self.fp = io.open(file, filemode)
PermissionError: [Errno 13] Permission denied: '/usr/local/lib/python3.8/dist-packages/mu/wheels/1.1.1.zip'

``
[mu.log](https://github.com/mu-editor/mu/files/8146027/mu.log)
`

carlosperate, 2022-03-01:
Hi @kevingtonbeare,

Is it possible you've installed Mu in the system Python using `sudo`? In that case you'll need `sudo` as well to launch it.

It is generally recommended to install Mu in a virtual environment, or something like [pipx](https://pypa.github.io/pipx/) (which I quite like and use myself).

kevingtonbeare, 2022-03-01:
thanks fur the advice. will try it.
k