mu-editor/mu #2077: Crash Report

ElsayedAymanElzahy, 2022-02-26:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.0b7
Locale: en_US
Platform: Linux 5.13.0-30-generic #33~20.04.1-Ubuntu SMP Mon Feb 7 14:25:10 UTC 2022 x86_64

```
Traceback (most recent call last):
  File "/home/sayed/.local/lib/python3.8/site-packages/mu/app.py", line 159, in run
    raise ex
  File "/home/sayed/.local/lib/python3.8/site-packages/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/sayed/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/home/sayed/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 771, in create
    self.create_venv()
  File "/home/sayed/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 803, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using /usr/bin/python3 at /home/sayed/.local/share/mu/mu_venv-38-20220226-134642
STDERR: /usr/bin/python3: No module named virtualenv

```
Log File // Start //

2022-02-02 13:23:24,576 - root:112(run) INFO: 

-----------------

Starting Mu 1.0.2
2022-02-02 13:23:24,576 - root:113(run) INFO: uname_result(system='Linux', node='sayed', release='5.13.0-28-generic', version='#31~20.04.1-Ubuntu SMP Wed Jan 19 14:08:10 UTC 2022', machine='x86_64', processor='x86_64')
2022-02-02 13:23:24,576 - root:114(run) INFO: Python path: ['/usr/share/mu-editor', '/usr/lib/python38.zip', '/usr/lib/python3.8', '/usr/lib/python3.8/lib-dynload', '/home/sayed/.local/lib/python3.8/site-packages', '/usr/local/lib/python3.8/dist-packages', '/usr/lib/python3/dist-packages']
2022-02-02 13:23:24,576 - root:115(run) INFO: Language code: en_US
2022-02-02 13:23:24,764 - mu.logic:543(__init__) INFO: Setting up editor.
2022-02-02 13:23:24,765 - mu.logic:560(__init__) DEBUG: Creating directory: /home/sayed/.local/share/mu
2022-02-02 13:23:24,765 - mu.logic:315(get_admin_file_path) DEBUG: Creating admin file: /home/sayed/.local/share/mu/settings.json
2022-02-02 13:23:24,765 - mu.logic:562(__init__) INFO: Settings path: /home/sayed/.local/share/mu/settings.json
2022-02-02 13:23:24,766 - mu.logic:315(get_admin_file_path) DEBUG: Creating admin file: /home/sayed/.local/share/mu/session.json
2022-02-02 13:23:24,766 - mu.logic:563(__init__) INFO: Session path: /home/sayed/.local/share/mu/session.json
2022-02-02 13:23:24,766 - mu.logic:564(__init__) INFO: Log directory: /home/sayed/.cache/mu/log
2022-02-02 13:23:24,766 - mu.logic:565(__init__) INFO: Data directory: /home/sayed/.local/share/mu
2022-02-02 13:23:24,796 - mu.logic:578(setup) INFO: Available modes: python, adafruit, microbit, debugger, pygamezero
2022-02-02 13:23:24,797 - mu.logic:582(setup) DEBUG: Creating directory: /home/sayed/mu_code
2022-02-02 13:23:24,797 - mu.logic:590(setup) DEBUG: Creating directory: /home/sayed/mu_code/images
2022-02-02 13:23:24,797 - mu.logic:597(setup) DEBUG: Creating directory: /home/sayed/mu_code/fonts
2022-02-02 13:23:24,798 - mu.logic:600(setup) DEBUG: Creating directory: /home/sayed/mu_code/sounds
2022-02-02 13:23:24,798 - mu.logic:603(setup) DEBUG: Creating directory: /home/sayed/mu_code/music
2022-02-02 13:23:24,973 - mu.logic:1147(change_mode) INFO: Workspace directory: /home/sayed/mu_code
2022-02-02 13:23:24,974 - mu.logic:625(restore_session) INFO: Restoring session from: /home/sayed/.local/share/mu/session.json
2022-02-02 13:23:24,974 - mu.logic:626(restore_session) DEBUG: {}
2022-02-02 13:23:24,974 - mu.logic:1104(select_mode) INFO: Showing available modes: ['python', 'adafruit', 'microbit', 'debugger', 'pygamezero']
2022-02-02 13:23:33,119 - mu.logic:678(restore_session) INFO: Starting with blank file.
2022-02-02 13:23:33,144 - mu.logic:1147(change_mode) INFO: Workspace directory: /home/sayed/mu_code
2022-02-02 13:23:36,536 - mu.logic:1051(quit) DEBUG: Session: {'theme': 'day', 'mode': 'python', 'paths': [], 'envars': [], 'minify': False, 'microbit_runtime': '', 'zoom_level': 2}
2022-02-02 13:23:36,536 - mu.logic:1052(quit) DEBUG: Saving session to: /home/sayed/.local/share/mu/session.json
2022-02-02 13:23:36,537 - mu.logic:1066(quit) INFO: Quitting.

// The End //

ElsayedAymanElzahy, 2022-02-26:
[mu.log](https://github.com/mu-editor/mu/files/8146517/mu.log)

carlosperate, 2022-02-28:
Thanks for the report @ElsayedAymanElzahy.

I believe this issue might be https://github.com/mu-editor/mu/issues/1954#issuecomment-1001258055.

Possible workarounds: 
- Install the python3-virtualenv package using apt (or your OS package manager)
- Install Mu in a virtual environment, as that is the recommended way and it should work much better.
- I would also personally recommend to use pipx, as it's great for installing Python-based applications: https://pypa.github.io/pipx/
    - pipx will automatically install Python applications in their own isolated virtual environment, and add the executable to the PATH so that it can be easily launched like any other app.