mu-editor/mu #2079: Crash Report

butlerpaul, 2022-02-26:
[mu.log](https://github.com/mu-editor/mu/files/8147119/mu.log)
[mu.log](https://github.com/mu-editor/mu/files/8147120/mu.log)
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.1
Locale: en_US
Platform: Linux 5.13.0-27-generic #29~20.04.1-Ubuntu SMP Fri Jan 14 00:32:30 UTC 2022 x86_64

```
e-packages/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/home/pbutler/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 772, in create
    self.install_baseline_packages()
  File "/home/pbutler/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 885, in install_baseline_packages
    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "/home/pbutler/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 860, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "/home/pbutler/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 270, in install
    return self.run(
  File "/home/pbutler/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "/home/pbutler/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "/home/pbutler/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 1:
ERROR: cryptography-36.0.1-cp36-abi3-manylinux_2_24_x86_64.whl is not a supported wheel on this platform.

```

carlosperate, 2022-03-01:
Thanks for the report @butlerpaul!

Could you run these commands on a terminal and paste the results here? Thanks!

```
/home/pbutler/.local/share/mu/mu_venv-38-20220226-113044.FAILED/bin/pip --version
```

```
/usr/bin/python --version
```

```
/usr/bin/python -m pip --version
```

butlerpaul, 2022-03-01:
Thanks for the help!

I suspected the pip version might be part of the problem so I tried to
manually update pip, but it seems to create a new install folder each time
I try to run it.

Also discovered the scripts rename the folder to .FAILED after startup
fails and the pip script in the renamed folder is pointing back to the
original folder name. So I created a symlink to make it work.

***@***.*** mu]$ ln -s mu_venv-38-20220301-114556.FAILED
mu_venv-38-20220301-114556

***@***.*** ~]$
/home/pbutler/.local/share/mu/mu_venv-38-20220301-114556/bin/pip --version
pip 20.0.2 from
/home/pbutler/.local/share/mu/mu_venv-38-20220301-114556/lib/python3.8/site-packages/pip
(python 3.8)

***@***.*** ~]$ /usr/bin/python --version
Python 3.8.10

***@***.*** ~]$ /usr/bin/python -m pip --version
pip 22.0.3 from /home/pbutler/.local/lib/python3.8/site-packages/pip
(python 3.8)

-paul

Paul Butler
***@***.***
Mobile: 214-250-6159

carlosperate, 2022-03-01:
Thanks for the quick reply!

Base on the pip versions I think your encountering this issue: https://github.com/mu-editor/mu/issues/1826

Basically the system pip is newer than the pip inside the virtual environment that Mu creates. For this reason the newer pip is able to download a wheel in a format not supported by the virtual environment pip and that's why it throws that error.

The easiest workaround would be to (temporally) downgrade the system pip to version 20.0.2 (the same version used inside the virtual environment), that way the system pip would download a different wheel which the venv can install.
Of course, after Mu run correctly you can update your system pip to the newer version.

One thing to keep in mind is that Mu has already downloaded the wheels and zipped them up. So to be able to force Mu to download the wheels again there are two options:
- Uninstall and reinstall Mu
- Delete the zip file yourself, file should be `/home/pbutler/.local/lib/python3.8/site-packages/mu/wheels/1.1.1.zip`

butlerpaul, 2022-03-01:
Thanks! Workaround solved the problem.

-paul

Paul Butler
***@***.***
Mobile: 214-250-6159

carlosperate, 2022-03-01:
Thanks for the update, I'm glad it worked!

As we are tracking this issue in https://github.com/mu-editor/mu/issues/1826 I'll close this one 👍

closed 2022-03-01 by carlosperate