mu-editor/mu #2086: Pygame Zero mode - window starting at screen center

xbecas, 2022-03-01:
### What were you trying to do?

Run Pygame Zero mode 

### What steps did you take to trigger the issue?

Simply running a Pygame Zero script - any script.

### What did you expect to happen?

The Pygame Zero window should be centered / at last position

### What actually happened?

The top-left corner of the Pygame Zero windows starts at the center of the screen.

Check image below:
[image]

### Operating System Version

W10 21H2

### Mu Version

1.1.1

### Other Info

FullHD screen - tested at different DPI scales, namely 100% and 125%.
No error reported from Mu, just the strange window positioning in Pygame Zero.

_(%USERPROFILE% replaced in log to mask username)_

### Editor Log

2022-03-01 23:44:45,447 - root:269(run) INFO: 

-----------------

Starting Mu 1.1.1
2022-03-01 23:44:45,447 - root:270(run) INFO: uname_result(system='Windows', node='P8826', release='10', version='10.0.19044', machine='AMD64', processor='Intel64 Family 6 Model 140 Stepping 1, GenuineIntel')
2022-03-01 23:44:45,473 - root:271(run) INFO: Platform: Windows-10-10.0.19044-SP0
2022-03-01 23:44:45,473 - root:272(run) INFO: Python path: ['C:\\Users\\vasco.f.mora\\AppData\\Local\\Programs\\Mu Editor', 'C:\\Users\\vasco.f.mora\\AppData\\Local\\Programs\\Mu Editor\\Python\\python38.zip', 'C:\\Users\\vasco.f.mora\\AppData\\Local\\Programs\\Mu Editor\\Python\\DLLs', 'C:\\Users\\vasco.f.mora\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib', 'C:\\Users\\vasco.f.mora\\AppData\\Local\\Programs\\Mu Editor\\Python', 'C:\\Users\\vasco.f.mora\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages', 'C:\\Users\\vasco.f.mora\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\setuptools-57.4.0-py3.8.egg', 'C:\\Users\\vasco.f.mora\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\pip-21.1.3-py3.8.egg', 'C:\\Users\\vasco.f.mora\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32', 'C:\\Users\\vasco.f.mora\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\win32\\lib', 'C:\\Users\\vasco.f.mora\\AppData\\Local\\Programs\\Mu Editor\\Python\\lib\\site-packages\\Pythonwin']
2022-03-01 23:44:45,473 - root:273(run) INFO: Language code: pt_PT
2022-03-01 23:44:45,473 - mu.settings:220(load) WARNING: No settings file found at %USERPROFILE%\AppData\Local\python\mu\settings.json; skipping
2022-03-01 23:44:45,524 - mu.virtual_environment:585(ensure_and_create) INFO: Added log handler.
2022-03-01 23:44:45,679 - mu.virtual_environment:596(ensure_and_create) DEBUG: Checking virtual environment; attempt #1.
2022-03-01 23:44:45,679 - mu.virtual_environment:685(ensure_path) INFO: Virtual Environment found at: %USERPROFILE%\AppData\Local\python\mu\mu_venv-38-20220222-180624
2022-03-01 23:44:45,679 - mu.virtual_environment:698(ensure_interpreter) INFO: Interpreter found at: %USERPROFILE%\AppData\Local\python\mu\mu_venv-38-20220222-180624\scripts\python.exe
2022-03-01 23:44:45,679 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\vasco.f.mora\\AppData\\Local\\python\\mu\\mu_venv-38-20220222-180624\\scripts\\python.exe', '-c', 'import sys; print("%s%s" % sys.version_info[:2])') with kwargs {'shell': True}
2022-03-01 23:44:45,763 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 38
2022-03-01 23:44:45,763 - mu.virtual_environment:730(ensure_interpreter_version) INFO: Both interpreters at version 38
2022-03-01 23:44:45,763 - mu.virtual_environment:759(ensure_pip) INFO: Pip found at: %USERPROFILE%\AppData\Local\python\mu\mu_venv-38-20220222-180624\scripts\pip.exe
2022-03-01 23:44:45,763 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: pgzero
2022-03-01 23:44:45,763 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\vasco.f.mora\\AppData\\Local\\python\\mu\\mu_venv-38-20220222-180624\\scripts\\python.exe', '-c', 'import pgzero') with kwargs {'shell': True}
2022-03-01 23:44:45,848 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2022-03-01 23:44:45,848 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: flask
2022-03-01 23:44:45,848 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\vasco.f.mora\\AppData\\Local\\python\\mu\\mu_venv-38-20220222-180624\\scripts\\python.exe', '-c', 'import flask') with kwargs {'shell': True}
2022-03-01 23:44:46,199 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2022-03-01 23:44:46,199 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: ipykernel
2022-03-01 23:44:46,199 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\vasco.f.mora\\AppData\\Local\\python\\mu\\mu_venv-38-20220222-180624\\scripts\\python.exe', '-c', 'import ipykernel') with kwargs {'shell': True}
2022-03-01 23:44:46,822 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2022-03-01 23:44:46,822 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: ipython_genutils
2022-03-01 23:44:46,822 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\vasco.f.mora\\AppData\\Local\\python\\mu\\mu_venv-38-20220222-180624\\scripts\\python.exe', '-c', 'import ipython_genutils') with kwargs {'shell': True}
2022-03-01 23:44:46,894 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2022-03-01 23:44:46,894 - mu.virtual_environment:742(ensure_key_modules) DEBUG: Verifying import of: esptool
2022-03-01 23:44:46,894 - mu.virtual_environment:429(run_subprocess) INFO: Running ('C:\\Users\\vasco.f.mora\\AppData\\Local\\python\\mu\\mu_venv-38-20220222-180624\\scripts\\python.exe', '-c', 'import esptool') with kwargs {'shell': True}
2022-03-01 23:44:47,018 - mu.virtual_environment:441(run_subprocess) DEBUG: Process returned 0; output: 
2022-03-01 23:44:47,018 - mu.virtual_environment:628(ensure_and_create) INFO: Valid virtual environment found at %USERPROFILE%\AppData\Local\python\mu\mu_venv-38-20220222-180624
[... 164 lines cut ...]
Pygments          2.11.2
pyserial          3.5
python-dateutil   2.8.2
pywin32           303
pyzmq             22.3.0
reedsolo          1.5.4
setuptools        60.9.3
six               1.16.0
stack-data        0.2.0
tomli             2.0.1
tornado           6.1
traitlets         5.1.1
typing_extensions 4.1.1
wcwidth           0.2.5
Werkzeug          2.0.3
wheel             0.37.1
2022-03-01 23:45:05,366 - mu.virtual_environment:931(installed_packages) INFO: []
2022-03-01 23:45:07,838 - mu.logic:1479(sync_package_state) INFO: Synchronize package states...
2022-03-01 23:45:07,838 - mu.logic:1480(sync_package_state) INFO: Old: set()
2022-03-01 23:45:07,838 - mu.logic:1481(sync_package_state) INFO: New: set()
2022-03-01 23:45:09,469 - mu.settings:169(save) DEBUG: Saving to %USERPROFILE%\AppData\Local\python\mu\session.json
2022-03-01 23:45:09,469 - mu.logic:1407(quit) INFO: Quitting.

2022-03-01 23:45:09,469 - mu.settings:169(save) DEBUG: Saving to %USERPROFILE%\AppData\Local\python\mu\session.json
2022-03-01 23:45:09,469 - mu.settings:169(save) DEBUG: Saving to %USERPROFILE%\AppData\Local\python\mu\venv.json

xbecas, 2022-03-02:
Found related issue on pgzero (@ntoll comment included!)
https://github.com/lordmauve/pgzero/issues/263

Prior to founding issue on pgzero, I found different behaviour on different pgzero/pygame:
- pgzero-1.2 pygame-1.9.6   --> window's top-left corner at screen's top-left position;
- pgzero-1.2.1 pygame-2.1.2 --> window's top-left corner at screen's center.

Release history:
- pgzero 1.2.1 on 2021-03-02: https://pypi.org/project/pgzero/#history
- pygame 2.1.2 on 2021-12-27: https://pypi.org/project/pygame/#history

Tested pygame down to 2.0.0 (released 2020-10-28), since "pgzero 1.2.1 requires pygame~=2.0"

xbecas, 2022-03-02:
Yes, apparently. In this case we close the issue in our side, right?

closed 2022-03-02 by xbecas

carlosperate, 2022-03-02:
Yeah, I think there isn't much for us to do, but it's good to have this in the tracker for people to find, and to be able to link to it if it comes up in a different issue.