mu-editor/mu #2091: Mocking super() in tests breaks with setuptools 60.6+

hroncok, 2022-03-02:
### What were you trying to do?

Run mu tests with latest setuptools installed

### What steps did you take to trigger the issue?

We are rebuilding an older version of mu in Fedora with setuptools 60.9 as we are rebuilding everything. I suppose the newer version of mu has the same problem as the test has never changed.

I don't have an easy reproducer yet. I assume run the tests with setuptools 60.9 should be enough.

The CI here appears to run with an older setuptools version.

I can submit a PR that should try to update it and we can see if it triggers the same failure.

### What did you expect to happen?

The tests should pass

### What actually happened?

The tests fail with:

```
=================================== FAILURES ===================================
__________________________ test_ButtonBar_add_action ___________________________

    def test_ButtonBar_add_action():
        """
        Check the appropriately referenced QAction is created by a call to
        addAction.
        """
        bb = mu.interface.main.ButtonBar(None)
        with mock.patch('builtins.super') as mock_s:
>           bb.addAction('save', 'Save', 'save stuff')

tests/interface/test_main.py:119: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
mu/interface/main.py:121: in addAction
    action = QAction(load_icon(name), display_name, self,
mu/resources/__init__.py:37: in load_icon
    return QIcon(path(name))
mu/resources/__init__.py:32: in path
    return resource_filename(__name__, resource_dir + name)
/usr/lib/python3.10/site-packages/pkg_resources/__init__.py:1141: in resource_filename
    return get_provider(package_or_requirement).get_resource_filename(
/usr/lib/python3.10/site-packages/pkg_resources/__init__.py:360: in get_provider
    return _find_adapter(_provider_factories, loader)(module)
/usr/lib/python3.10/site-packages/pkg_resources/__init__.py:1591: in __init__
    self._setup_prefix()
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

self = <pkg_resources.DefaultProvider object at 0x7f572f61fcd0>

    def _setup_prefix(self):
        # Assume that metadata may be nested inside a "basket"
        # of multiple eggs and use module_path instead of .archive.
>       eggs = filter(_is_egg_path, _parents(self.module_path))
E       AttributeError: 'DefaultProvider' object has no attribute 'module_path'

/usr/lib/python3.10/site-packages/pkg_resources/__init__.py:1596: AttributeError
```

This is because you mock super and pkg_resources now use it as well:

https://github.com/pypa/setuptools/commit/a7c6d64f76091d1cfd8297ba813fe4e901d00ce4

I suppose mocking super is dangerous :biohazard: 

### Operating System Version

Fedora Linux 37

### Mu Version

1.0.3 but the same tests still exist

hroncok, 2022-03-02:
Yes, the CI shows the failure.

xbecas, 2022-03-11:
Hi @hroncok. I had the same problem, failing `test_ButtonBar_add_action`.

Your suggestion of "running the tests with setuptools 60.9 should be enough" was a major clue, but I had 60.9.3 installed and it failed anyway.
"Fixed" it by downgrading setuptools from  60.9.3 to 56.0.0, using `pip install setuptools==56.0.0`. Now all tests, passed on my machine (W10 64bit, Python 3.8.10 64bit)!

Regarding the upcoming setuptools versions more research will be necessary!
Thanks!

carlosperate, 2022-05-20:
Thanks @hroncok, I've created PR https://github.com/mu-editor/mu/pull/2238, which I think should resolve the issue.

closed 2022-05-21 by carlosperate