mu-editor/mu #2100: Crash Report

rick0142, 2022-03-05:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.1
Locale: en_US
Platform: Windows 10 10.0.19043 AMD64

```
Traceback (most recent call last):
  File "C:\Users\JNR\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 159, in run
    raise ex
  File "C:\Users\JNR\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\JNR\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "C:\Users\JNR\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 771, in create
    self.create_venv()
  File "C:\Users\JNR\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 803, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using C:\Users\JNR\AppData\Local\Programs\Mu Editor\Python\pythonw.exe at C:\Users\JNR\AppData\Local\python\mu\mu_venv-38-20220305-171420
FileExistsError: [WinError 183] Can
[mu.log](https://github.com/mu-editor/mu/files/8191597/mu.log)
[mu.log](https://github.com/mu-editor/mu/files/8191599/mu.log)
not create a file when that file already exists: 'C:\\Users\\JNR\\AppData\\Local\\python\\mu\\mu_venv-38-20220305-171420\\Lib\\site-packages'
[mu.log](https://github.com/mu-editor/mu/files/8191593/mu.log)

``
[mu.log](https://github.com/mu-editor/mu/files/8191602/mu.log)
`

carlosperate, 2022-03-06:
Hi @rick0142, thanks for the report.

None of the log files attached include the error listed in the crash report. Are you able to open the Mu editor correctly?

rick0142, 2022-03-06:
Hi Carlos, 

Thanks for your response.

I got it working with a shutdown/restart on the Win 10 box.  I should have known better, when I was installing the 1.1.1 Mu release, it warned me that a process (python) was currently in use, but it gave the option to cancel and stop the install, or do a restart after the install was complete.  I tried the latter, after the install completed, I restarted the box opened Mu and that is when the error came up.  Of course I tried restarting Mu again, to no avail.

I then shutdown and rebooted the Win 10 box.  That solved the problem, Mu came right up and I tested it with my little esp32  Micropython project.

Again Thanks for you response.
..rick..

carlosperate, 2022-03-07:
Thanks for the updated, glad you got it working!

closed 2022-03-07 by carlosperate