mu-editor/mu #2101: Crash Report

Szwagier420, 2022-03-07:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.1
Locale: pl_PL
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Rogers\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 159, in run
    raise ex
  File "C:\Users\Rogers\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Rogers\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 627, in ensure_and_create
    self.ensure()
  File "C:\Users\Rogers\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 665, in ensure
    self.ensure_interpreter_version()
  File "C:\Users\Rogers\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 732, in ensure_interpreter_version
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Mu interpreter at version 38; venv interpreter at version .

```

carlosperate, 2022-03-07:
Thanks for the report @Szwagier420!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\Rogers\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2022-03-07:
Thanks for the log!

Could you run these command in a command prompt and paste the result here? This is the step causing an issue, but I cannot see what the problem might be from the log:

```
C:\Users\Rogers\AppData\Local\python\mu\mu_venv-38-20220307-081802.FAILED\scripts\python.exe -c 'import sys; print("%s%s" % sys.version_info[:2])'
```

```
C:\Users\Rogers\AppData\Local\python\mu\mu_venv-38-20220307-081802.FAILED\scripts\python.exe -c 'import sys; print(sys.version_info)'
```