mu-editor/mu #2102: Crash Report

djibb, 2022-03-08:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.1
Locale: fr_FR
Platform: Windows 10 10.0.18363 AMD64

```
Traceback (most recent call last):
  File "C:\Users\jeanbaptist.butet\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 159, in run
    raise ex
  File "C:\Users\jeanbaptist.butet\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\jeanbaptist.butet\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 627, in ensure_and_create
    self.ensure()
  File "C:\Users\jeanbaptist.butet\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 665, in ensure
    self.ensure_interpreter_version()
  File "C:\Users\jeanbaptist.butet\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 732, in ensure_interpreter_version
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Mu interpreter at version 38; venv interpreter at version 38

STDERR: '\\TLPU005-AD.TLPU005
[mu.log](https://github.com/mu-editor/mu/files/8205021/mu.log)
.ETAB.LOCAL\professeurs\jeanbaptist.butet\Mes documents'
CMD.EXE a ï¿½tï¿½ dï¿½marrï¿½ avec le chemin d'accï¿½s comme rï¿½pertoire en
cours. Les chemins d'accï¿½s UNC ne sont pas prise en charge. Utilisation
du rï¿½pertoire Windows par dï¿½faut..

``
[mu.log](https://github.com/mu-editor/mu/files/8205024/mu.log)
`

carlosperate, 2022-03-08:
Thanks for the report @djibb!
Could you run these commands in a command prompt (cmd.exe) and paste the result here?

```
C:\Users\jeanbaptist.butet\AppData\Local\python\mu\mu_venv-38-20220308-112526.FAILED\scripts\python.exe -c 'import sys; print("%s%s" % sys.version_info[:2])'
```

```
C:\Users\jeanbaptist.butet\AppData\Local\python\mu\mu_venv-38-20220308-112526.FAILED\scripts\python.exe -c "import sys; print(sys.version_info)"
```

djibb, 2022-03-08:
first one : 38
seconde one : 
sys.version_info(major=3, minor=8, micro=11, releaselevel='final', serial=0)

carlosperate, 2022-03-08:
Thank you @djibb, this is really useful!

Could try this as well and paste the result here?

```
C:\Users\jeanbaptist.butet\AppData\Local\python\mu\mu_venv-38-20220308-112526.FAILED\scripts\python.exe -c 'import sys; print("%s" % bytes(sys.version_info[:2]))'
```