mu-editor/mu #2107: Crash Report

Sunzany, 2022-03-09:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.1
Locale: en_US
Platform: Windows 10 10.0.22000 AMD64

```
in create
    self.install_baseline_packages()
  File "C:\Users\hp\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 885, in install_baseline_packages
    self.install_from_zipped_wheels(zipped_wheels_filepath)
  File "C:\Users\hp\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 860, in install_from_zipped_wheels
    self.pip.install(wheel, deps=False, index=False)
  File "C:\Users\hp\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 270, in install
    return self.run(
  File "C:\Users\hp\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 246, in run
    result = self.process.run_blocking(
  File "C:\Users\hp\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 122, in run_blocking
    return self.wait(wait_for_s=wait_for_s)
  File "C:\Users\hp\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 186, in wait
    raise VirtualEnvironmentError(
mu.virtual_environment.VirtualEnvironmentError: Process finished but with error code 259:
Processing c:\users\hp\appdata\local\temp\tmp4v4m0kw0\cffi-1.15.0-cp38-cp38-win_amd64.whl
Installing collected packages: cffi
Successfully installed cffi-1.15.0

```

carlosperate, 2022-03-09:
Thanks for the report @Sunzany!
Would you be able to attach the Mu log to this GitHub Issue as well? It'll be really useful to us for debugging. File should be located in: `C:\Users\hp\AppData\Local\python\mu\Logs\mu.log`

carlosperate, 2022-06-06:
Hi @Sunzany, are you still experiencing this issue? Would you be able to attach the log here to help us debug it? It shoul dbe located in `C:\Users\hp\AppData\Local\python\mu\Logs\mu.log`