mu-editor/mu #2137: \r not working as intended

john-kat, 2022-03-21:
### What were you trying to do?

Trying to build a clock application.

### What steps did you take to trigger the issue?

from time import strftime
import time

def clock():
    run = True
    while run:
        string = strftime("%H:%M:%S")
        print(string, end = "\r")
        time.sleep(1)
clock()

### What did you expect to happen?

The clock should refresh each second on the existing output line

### What actually happened?

The clock refreshes on a new line each second.

### Operating System Version

Windows 10 (previously) Windows 11 now.

### Mu Version

1.1.1

### Other Info

 I wrote about this in chat previously. I have now reinstalled windows, Python and MU and the problem is not fixed.
Please see attached video

https://user-images.githubusercontent.com/72275187/159249863-6cab0605-4372-4ddb-8957-ed18eb40b8dc.mp4

.

ntoll, 2022-03-21:
Hi @john-kat this will be a bug in the shell/terminal widget we use. Thanks for pointing this out, and your code example will be a great way for us to test the fix. :+1:

john-kat, 2022-03-21:
Hi @ntoll. Fantastic! Thanks for getting back to me so quickly.  Happy to help 👍