mu-editor/mu #2142: Crash Report

knkp, 2022-03-23:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.1
Locale: en_US
Platform: Linux 5.13.0-30-generic #33~20.04.1-Ubuntu SMP Mon Feb 7 14:25:10 UTC 2022 x86_64

```
Traceback (most recent call last):
  File "/home/knkp/.local/lib/python3.8/site-packages/mu/app.py", line 159, in run
    raise ex
  File "/home/knkp/.local/lib/python3.8/site-packages/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/knkp/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/home/knkp/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 771, in create
    self.create_venv()
  File "/home/knkp/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 803, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using /usr/bin/python3 at /home/knkp/.local/share/mu/mu_venv-38-20220323-005324
STDERR: /usr/bin/python3: No module named virtualenv

```

knkp, 2022-03-23:
I fixed this by removing the '-I' from the `create_venv(self)` method.

```
def create_venv(self):
         """
         Create a new virtualenv
         """
         logger.info("Creating virtualenv: {}".format(self.path))
         logger.info("Virtualenv name: {}".format(self.name))
 
         env = dict(os.environ)
         ok, output = self.run_subprocess(
             sys.executable,
             # "-I", #  no bueno 
             "-m",
              "virtualenv",
              "-p",
              sys.executable,
              "-q",
              self.path,
              env=env,
          )
          if ok:
              logger.info(
                  "Created virtual environment using %s at %s",
                  sys.executable,
                  self.path,
              )
          else:
              raise VirtualEnvironmentCreateError(
                  "Unable to create a virtual environment using %s at %s\n%s"
                  % (sys.executable, self.path, compact(output))
              )
```

This worked for me running python3.8 on ubuntu 20.04.

No idea if it's an appropriate fix. Figured I'd at least point in the direction I'm taking.

knkp, 2022-03-23:
If interested I can find time this week to submit a pull request. Otherwise obviously the fix is pretty easy.

carlosperate, 2022-05-10:
Thanks for the report @knkp, and for finding a workaround!

I believe this issue might be https://github.com/mu-editor/mu/issues/1954#issuecomment-1001258055.

Other possible workarounds until we fix the issue are: 
- Install the `python3-virtualenv` package using apt (or your OS package manager)
- Install Mu in a virtual environment, as that is the recommended way and it should work much better.
- I would also personally recommend to use pipx, as it's great for installing Python-based applications: https://pypa.github.io/pipx/
    - pipx will automatically install Python applications in their own isolated virtual environment, and add the executable to the PATH so that it can be easily launched like any other app.

knkp, 2022-05-11:
I wasn't aware of pipx - thanks for the tip!

carlosperate, 2022-05-12:
Glad that was useful!
As you've been able to work around the issue and we are tracking progress via https://github.com/mu-editor/mu/issues/1954 I'll close this one.

Thanks again for the report & the workaround!

closed 2022-05-12 by carlosperate