mu-editor/mu #2144: Crash Report

carycode, 2022-03-24:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.1
Locale: en_US
Platform: Linux 5.13.0-35-generic #40~20.04.1-Ubuntu SMP Mon Mar 7 09:18:32 UTC 2022 x86_64

```
Traceback (most recent call last):
  File "/home/david/.local/lib/python3.8/site-packages/mu/app.py", line 159, in run
    raise ex
  File "/home/david/.local/lib/python3.8/site-packages/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/david/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/home/david/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 771, in create
    self.create_venv()
  File "/home/david/.local/lib/python3.8/site-packages/mu/virtual_environment.py", line 803, in create_venv
    raise VirtualEnvironmentCreateError(
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using /usr/bin/python3 at /home/david/.local/share/mu/mu_venv-38-20220323-204856
STDERR: /usr/bin/python3: No module named virtualenv

```
[mu.log](https://github.com/mu-editor/mu/files/8338362/mu.log)

carycode, 2022-03-24:
The fix described in
https://github.com/mu-editor/mu/issues/2142
works for me:

removing the '-I' from the create_venv(self) method
in the virtual_environment.py file.

Or in other words:

    cd ~/.local/lib/python3.8/site-packages/mu

and then editing
    
    vim virtual_environment.py

to comment out one line:

    def create_venv(self):
        """
        Create a new virtualenv
        """
        logger.info("Creating virtualenv: {}".format(self.path))
        logger.info("Virtualenv name: {}".format(self.name))

        env = dict(os.environ)
        ok, output = self.run_subprocess(
            sys.executable,
            # "-I",  # no bueno
            "-m",
            "virtualenv",
...

carlosperate, 2022-05-10:
Thanks for the report @rickcox!

As you have found, I believe this issue might be https://github.com/mu-editor/mu/issues/1954#issuecomment-1001258055.

Other possible workarounds until we fix the issue are: 
- Install the `python3-virtualenv` package using apt (or your OS package manager)
- Install Mu in a virtual environment, as that is the recommended way and it should work much better.
- I would also personally recommend to use pipx, as it's great for installing Python-based applications: https://pypa.github.io/pipx/
    - pipx will automatically install Python applications in their own isolated virtual environment, and add the executable to the PATH so that it can be easily launched like any other app.

As you were able to get your local Mu app running, I'll close this issue and we'll track the problem via https://github.com/mu-editor/mu/issues/1954

Thanks again!

closed 2022-05-10 by carlosperate