mu-editor/mu #2178: Crash Report

willywog, 2022-04-15:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.1
Locale: en_US
Platform: Linux 5.4.0-107-generic #121~18.04.1-Ubuntu SMP Thu Mar 24 17:21:33 UTC 2022 x86_64

```
Traceback (most recent call last):
  File "/home/bob/.local/lib/python3.6/site-packages/mu/app.py", line 159, in run
    raise ex
  File "/home/bob/.local/lib/python3.6/site-packages/mu/app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "/home/bob/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 603, in ensure_and_create
    self.create()
  File "/home/bob/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 771, in create
    self.create_venv()
  File "/home/bob/.local/lib/python3.6/site-packages/mu/virtual_environment.py", line 805, in create_venv
    % (sys.executable, self.path, compact(output))
mu.virtual_environment.VirtualEnvironmentCreateError: Unable to create a virtual environment using /usr/bin/python3 at /home/bob/.local/share/mu/mu_venv-36-20220415-151501
STDERR: /usr/bin/python3: No module named virtualenv

```

willywog, 2022-04-15:
Totally lost on this. Can anyone help?

willywog, 2022-04-15:
Trying to install Mu editor on ubuntu 18.04. Having this error when running the program.

carlosperate, 2022-05-10:
Thanks for the report @willywog!

I believe this issue might be https://github.com/mu-editor/mu/issues/1954#issuecomment-1001258055.

Possible workarounds: 
- Install the `python3-virtualenv` package using apt (or your OS package manager)
- Install Mu in a virtual environment, as that is the recommended way and it should work much better.
- I would also personally recommend to use pipx, as it's great for installing Python-based applications: https://pypa.github.io/pipx/
    - pipx will automatically install Python applications in their own isolated virtual environment, and add the executable to the PATH so that it can be easily launched like any other app.