mu-editor/mu #2186: Crash Report

xbecas, 2022-04-21:
**REMEMBER TO ATTACH YOUR LOG FILE**

Version: 1.1.1
Locale: pt_PT
Platform: Windows 10 10.0.19044 AMD64

```
Traceback (most recent call last):
  File "C:\Users\vasco.f.mora\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\python3.py", line 109, in start_kernel
    self.repl_kernel_manager.start_kernel()
  File "C:\Users\vasco.f.mora\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\modes\python3.py", line 42, in start_kernel
    kernel_cmd, kw = self.pre_start_kernel(**kw)
  File "C:\Users\vasco.f.mora\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\jupyter_client\manager.py", line 292, in pre_start_kernel
    kernel_cmd = self.format_kernel_cmd(extra_arguments=extra_arguments)
  File "C:\Users\vasco.f.mora\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\jupyter_client\manager.py", line 210, in format_kernel_cmd
    cmd = self.kernel_spec.argv + extra_arguments
  File "C:\Users\vasco.f.mora\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\jupyter_client\manager.py", line 114, in kernel_spec
    self._kernel_spec = self.kernel_spec_manager.get_kernel_spec(self.kernel_name)
  File "C:\Users\vasco.f.mora\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\jupyter_client\kernelspec.py", line 230, in get_kernel_spec
    raise NoSuchKernel(kernel_name)
jupyter_client.kernelspec.NoSuchKernel: No such kernel named mu_venv-38-20220302-114230

```
[mu.log](https://github.com/mu-editor/mu/files/8529850/mu.log)

xbecas, 2022-04-21:
Just opened Mu and pressed REPL button.

carlosperate, 2022-05-10:
Did you find out what was the cause of the issue @xbecas ?

Is it possible you've deleted the Mu (or all) Jupyter kernels? Maybe from a different Jupyter notebook/lab installation?

xbecas, 2022-05-10:
I don't think I have, and I have uninstalled and reinstalled Mu 1.1.1 and the problem persists.
Can you give some guidance on how to check that?

carlosperate, 2022-05-10:
Weird, if you delete the virtual environment folders from  `C:\Users\vasco.f.mora\AppData\Local\python\mu\` (they should all be something like  `mu_venv-38-.....`), and the `venv.json` file, and then open Mu, does still crash?

xbecas, 2022-05-10:
Following your suggestion I remove all the local files and folders from `C:\Users\vasco.f.mora\AppData\Local\python\mu\`. Mu started with the post-installation setup, taking a bit longer and asking for the mode at start. New `mu_venv-38-YYYYMMDD-hhmmss` folder, new `venv.json` and `session.json` as expected.

All works now - many thanks for the assistance. Don't know how I got here, but learning a bit more about how it works ;-)

closed 2022-05-10 by xbecas