mu-editor/mu #2190: "Failed to run venv interpreter"

gvanem, 2022-04-22:
First I tried installing using `pip3 install mu-editor`. This failed after a long time on some PyQt5 version incompatibilities.

The tried installing using the Windows MSI. That failed too on illegal syntax.
The generated  log from the crash:

Version: 1.1.1
Locale: nb_NO
Platform: Windows 10 10.0.19041 AMD64

```
Traceback (most recent call last):
  File "C:\Users\Gisle\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 159, in run
    raise ex
  File "C:\Users\Gisle\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\app.py", line 146, in run
    venv.ensure_and_create(self.display_text)
  File "C:\Users\Gisle\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 627, in ensure_and_create
    self.ensure()
  File "C:\Users\Gisle\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 665, in ensure
    self.ensure_interpreter_version()
  File "C:\Users\Gisle\AppData\Local\Programs\Mu Editor\Python\lib\site-packages\mu\virtual_environment.py", line 723, in ensure_interpreter_version
    raise VirtualEnvironmentEnsureError(
mu.virtual_environment.VirtualEnvironmentEnsureError: Failed to run venv interpreter c:\Users\Gisle\AppData\Local\python\mu\mu_venv-38-20220422-105747\scripts\python.exe
STDERR:   File "<string>", line 1
    import sys; print("s"  sys.version_info[:2])
                           ^
SyntaxError: invalid syntax
```

And the screen on running manually the 1st time:
[image]

I failed to find which file that ` import sys; print("s"  sys.version_info[:2])` is in. Or perhaps it's a shell issue where the `%` got lost.
A very confusing package IMHO.

gvanem, 2022-04-22:
> Or perhaps it's a shell issue ...

Yes it was. With this `mu-edit.bat`, it works:
```
@echo off
setlocal
on break quit

set VBS_FILE="c:\Users\Gisle\AppData\Local\Programs\Mu Editor\Mu Editor.vbs"
set COMSPEC=cmd.exe

echo Starting %VBS_FILE%....
cmd.exe /C c:\Windows\system32\cscript.exe //Nologo %VBS_FILE% %*
```

And yes, I do no normally use the CMD crap. I've been using [JPsoft](https://jpsoft.com)'s 4NT for almost 20 years.

carlosperate, 2022-04-22:
Thanks for the report @gvanem!
I'm not familiar with Windows shells, but we never encountered this issue before, do you know what the problem in itself might be?

The line in question is this one:
https://github.com/mu-editor/mu/blob/00edfc4464a8cf1b94b102bcfe84cd81fbe921dd/mu/virtual_environment.py#L719

So, I'm aslo a bit confused on how `"%s%s"` transformed into a single `"s"`.

carlosperate, 2022-04-22:
Ah, is the issue that the shell is running [JPsoft](https://jpsoft.com/)'s 4NT by default instead of CMD?

gvanem, 2022-04-22:
> Ah, is the issue that the shell is running 4NT ...

Seems so. The quotes are not correct for 4NT. But why does it need to the shell anyway?